SolidEva/PrawnOS #6: test HDMI

SolidHal, 2018-06-23:
on kernel 4.9, HDMI didnt show up in dmesg, but now it does on 4.17.2 with new config. Untested, but should work. 
Test this

SolidHal, 2018-07-10:
video works, no sound

SolidHal, 2018-08-14:
For sound, the patch at the bottom of this may be helpful https://github.com/Miouyouyou/RockMyy/issues/2

SolidHal, 2018-09-10:
Confirmed that video still works, just a little weird to setup in lxqt. 

Start with the monitors settings pane closed, plug in the hdmi, then open the monitor settings. Now you have to check "enable this display" under HDMI

SolidHal, 2018-09-17:
Audio doesnt work, may be fixed if 
```
&hdmi_audio {
	status = "okay";
};
```
is added to https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/arch/arm/boot/dts/rk3288-veyron-speedy.dts

to match https://chromium.googlesource.com/chromiumos/third_party/kernel/+/chromeos-3.14/arch/arm/boot/dts/rk3288-veyron.dtsi

gregordinary, 2019-09-22:
Not sure if this will be the fix, but looks like 5.4 may be getting better HDMI audio support for the rk3288:
[Add HDMI jack support on RK3288](https://lkml.org/lkml/2019/9/19/431)

gregordinary, 2020-05-17:
On the most recent PrawnOS release, output of `aplay -l` is:
```
**** List of PLAYBACK Hardware Devices ****
card 0: VEYRONI2S [VEYRON-I2S], device 0: Audio HiFi-0 [Audio HiFi-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
```
I tried patching rk3288-veyron-speedy.dts with the suggestion above but did not get a different result.

I did find a couple things of interest:

1. [This patch](https://lkml.org/lkml/2019/10/28/62) was added to kernel 5.5. 

> All boards using rk3288-veyron-analog-audio.dtsi have HDMI audio.
> Specify the support of HDMI audio on machine driver using
> rockchip,hdmi-codec property so machine driver creates HDMI audio device.

2.  [CONFIG_SND_SOC_RK3288_HDMI_ANALOG](https://cateee.net/lkddb/web-lkddb/SND_SOC_RK3288_HDMI_ANALOG.html) is currently not set in the kernel config file: 

> All boards using rk3288-veyron-analog-audio.dtsi have HDMI audio.
> Specify the support of HDMI audio on machine driver using
> rockchip,hdmi-codec property so machine driver creates HDMI audio device.

I built PrawnOS with Kernel 5.5.19 and enabled the HDMI_ANALOG mentioned above in the kernel config. 

aplay -l output on the resulting build was: 
```
**** List of PLAYBACK Hardware Devices ****
card 0: VEYRONI2S [VEYRON-I2S], device 0: Analog HiFi-0 [Analog HiFi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: VEYRONI2S [VEYRON-I2S], device 1: HDMI i2s-hifi-1 [HDMI i2s-hifi-1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
I tried doing a speaker-test with the device but could not get it to work. Although it is listed as a device via aplay, I do not see it under the Pulse Audio Volume Control. https://i.imgur.com/z7U4qpd.png

SolidHal, 2020-05-18:
Thanks for digging into this @gregordinary. Its not clear to me yet what the issue might be, or when I'll have a chance to dig into this myself

kapouer, 2021-04-05:
@gregordinary you were actually very close to getting it to work: just add to `/etc/pulse/default.pa`
```
load-module module-alsa-sink device=hw:0,1
```
and then hdmi output works and automatically switches from internal to hdmi when plugging cable.
(on c201).

gregordinary, 2022-10-23:
Revisiting this, thank you for the guidance @kapouer 

1. Built PrawnOS from source, since it's now using 5.15.6, I didn't apply the patch from 5.5 mentioned in my previous comment. I did however, enable CONFIG_SND_SOC_RK3288_HDMI_ANALOG in the config. 
2. After building and installing XFCE, I modified `/etc/pulse/default.pa` to contain `load-module module-alsa-sink device=hw:0,1`
3. Reboot after modifying the above, plugged in HDMI cable and audio is sent externally. 

While the audio worked over HDMI, it was choppy. After another reboot and playing an ogg file in VLC, it was smooth but if I selected different speakers and reverted back to HDMI, it was choppy again. I'd like to test this further.

kapouer, 2022-10-23:
See also https://github.com/alsa-project/alsa-ucm-conf/issues/208

gregordinary, 2022-10-23:
Unfortunately same behavior with the updated .conf file. 

I came across that .conf while troubleshooting the [other sound-related issue](https://github.com/SolidHal/PrawnOS/issues/305#issuecomment-1287961947). Not sure if it would behave any differently under different kernels or Debian release versions. Some chance it's also something with my chromebook, I haven't used HDMI audio much at all on this device.

kapouer, 2022-10-23:
I think choppy sound is fixed in more recent versions of linux, or pulseaudio, I don't know which. Or maybe with pipewire/wireplumber. In any case, give a few weeks and use latest releases.