SolidEva/PrawnOS #19: Strip non-libre components

SolidHal, 2018-08-09:
Build a package archive mirror of the debian archive, but without any of the nasty non-libre bits.

Examples:
PureOS (based on Debian) uses https://github.com/lkorigin/laniakea
some further information here https://puri.sm/posts/what-is-pureos-and-how-is-it-built/

Trisquel  (Based on Ubuntu) uses Package Helpers https://trisquel.info/en/wiki/package-helpers
Their process is described quite thoroughly here https://trisquel.info/en/wiki/how-trisquel-made

SolidHal, 2018-08-10:
Use the fork of firefox trisquel uses? http://mirror.fsf.org/trisquel/pool/main/f/firefox/
Then use their browser addons page?

SolidHal, 2018-08-28:
Trying to build/install https://github.com/lkorigin/laniakea
Clone https://github.com/lkorigin/laniakea
install meson `pip3 install meson` 
install prereqs: `apt install pkg-config gdc libarchive-dev libcurl-dev libcurl4-openssl-dev libczmq-dev libglib2.0-dev appstream libappstream-dev libpq-dev libvibe-core-dev gir-to-d

maybe required...:
get glibd from sid or buster https://packages.debian.org/search?lang=en&searchon=names&keywords=glibd by editing /etc/apt/sources.list

SolidHal, 2018-08-28:
as for the actual repo, this has some dak setup instructions https://salsa.debian.org/ftp-team/dak/tree/master/setup
This is the pureos version https://source.puri.sm/pureos/infra/pdak/tree/master/setup

SolidHal, 2018-08-28:
Some docs from debians management: https://www.debian.org/doc/manuals/debian-reference/ch02.en.html#_debian_package_management_internals

SolidHal, 2018-08-28:
Some packaging docs from pure os: https://tracker.pureos.net/w/development/packaging_overview/

SolidHal, 2018-09-01:
To build dak on sid:

apt install postgresql-10 postgresql-client-10 postgresql-10-debversion python-psycopg2 python-sqlalchemy python-apt gnupg dpkg-dev lintian

polarisfm, 2020-12-05:
I know this issue has been inactive for over two years, but it is worth looking at PureOS. They have perfectly functional arm64 Debian repositories, and PureOS is FSF approved. While they don't support armhf officially it could certainly be used as a a base for armhf PrawnOS packages.

My understanding is that they use Debian Bullseye as their base, so that works out well enough.

albsod, 2020-12-05:
Would it make sense to base PrawnOS entirely on PureOS? Apart from the hardware specific quirks needed for Chromebooks it's perhaps unnecessary to maintain a separate "stripped"/"purified" Debian distro when such a project already exists? PureOS doesn't have any arm32 packages, but if we provide them then perhaps they would be interested in co-hosting?

SolidHal, 2020-12-12:
basing off of PureOS would make sense, and save PrawnOs from needing to host a bunch of packages. Unfortunately it would likely be difficult to convince Purism to host a bunch of armhf packages for devices they don't support.

albsod, 2020-12-12:
If the packages are generic, I don't see why they would be opposed to it. It would mean more visibility to their distro and their brand. Even if they won't accept armhf packages, we could host that particular architecture separately.

SolidHal, 2020-12-15:
Its definitely a route I would consider, as the cost savings would be nice. There is also some value to maintaining our own PrawnOS repos rather than be dependent on Purism though.

spoelstraethan, 2021-01-13:
Debian keeps non-free in a separate target that isn't enabled by default, are you referring to additional blobby or proprietary software beyond what Debian considers non-free?