SolidEva/PrawnOS #20: Suggestions for reasonable video playback

SolidHal, 2018-08-09:
The GPU section of libreplanets hardware Workarounds gives some suggestions for better video playback:
https://libreplanet.org/wiki/Hardware/Workarounds

> Although the VPU is specifically for video, GPUs often play a role in video playback. However, if there is no GPU available (software rasterisation is used as a fallback), modern video players may have a heavy performance hit from attempting to use this “optimised” path.
> 
> One video player which can avoid this issue is mpv. By default, mpv usually requires a GPU or software rasterisation (through OpenGL). However, if it is passed with the --vo x11 option, OpenGL will be avoided for significantly improved performance. For example, instead of:
> 
> $ mpv <https://static.fsf.org/nosvn/FSF30-video/FSF_30_720p.webm>
> 
> Use:
> 
> $ mpv --vo x11 <https://static.fsf.org/nosvn/FSF30-video/FSF_30_720p.webm>
> 
> mpv will print a warning that “this legacy VO has bad performance. Consider fixing your graphics drivers, or not forcing the x11 VO.”. Ignore the message; counter-intuitively, in the absence of the GPU, this will usually improve performance.
> 
> Depending on the type of video, certain software scaling algorithms may be better. For instance, on cartoon-like videos which are predominantly flat-shaded except for edges, the area algorithm may provide a considerable performance boost with minimal loss of quality from the default algorithm. Invoke it by passing the --sws-scaler=area flag, e.g.:
> 
> $ mpv --vo x11 --sws-scaler=area <https://static.fsf.org/nosvn/FSF30-video/FSF_30_720p.webm>
> 
> You may prefer the even-faster fast-bilinear algorithm:
> 
> $ mpv --vo x11 --sws-scaler=fast-bilinear <https://static.fsf.org/nosvn/FSF30-video/FSF_30_720p.webm>
> 
> For VLC, GPU acceleration can be disabled by navigating to “Tools->Preferences->Video” and setting “Output” to “X11 video output (XCB)”, assuming X11 is in use on your system. VLC may need to be restarted before this settings are enabled. However, with both configured optimally, mpv may still be faster than VLC.

albsod, 2018-09-23:
I can confirm that the suggestion for **VLC** of using the "X11 video output (XCB)" setting is improving the performance for the FSF video mentioned above. I also figured out a way to enable this option without running the application: just replace the line `vout=any` with `vout=xcb_x11` in the file `.config/vlc/vlcrc` (just grep for `vout=`).

SolidHal, 2019-06-24:
Closing, as video playback is alright and gpu support is inbound.

closed 2019-06-24 by SolidHal