SolidEva/PrawnOS #28: Pursue FSF endorsement?

SolidHal, 2018-08-28:
Once all major bugs are fixed and the package repo is up, pursue this checklist https://libreplanet.org/wiki/Template:FSDG_Checklist

SolidHal, 2018-09-13:
Idea: Instead of copying all of debians repos, add a package that conflicts with all non-free packages, akin to "your freedom"

jonassmedegaard, 2018-09-22:
You most likely won't achieve FSF endorsement unless distributing independent of debian.org domain: Reason Debian itself is not endorsed is that non-free stuff _exist_ on same debian.org domain, even though it requires explicit action from the user to make any use of it (much like your idea above).

SolidHal, 2018-09-22:
Yeah, I've been chatting with someone interning at the FSF about just that. It'd be nice to leverage debians large set of mirrors, and not have to duplicate all of their infrastructure though :/

endian-albin, 2020-12-05:
I'd be willing to provide a Swedish mirror for PrawnOS if the required amount of storage isn't too much. How much do you think is needed per arch?

albsod, 2020-12-05:
I'd be willing to provide a Swedish mirror for PrawnOS if the required amount of storage isn't too much. How much do you think is needed per arch?

SolidHal, 2020-12-12:
@albsod likely 100s of gigabytes to mirror all of the packages from each arch would be required. The next biggest thing holding this back would be firefox, distros with firefox can't get FSF endorsement.

albsod, 2020-12-12:
Oh, that's a lot. Is there a way to find out the total size of just the compressed armhf packages in main?

gregordinary, 2020-12-12:
The armhf archive size is 275 GB:
https://www.debian.org/mirror/size

It'll need a bit more coordination to confirm we have resources, but I'd be willing to contribute some monthly amount for hosting to help achieve this.

SolidHal, 2020-12-15:
Thanks @gregordinary, thats actually less than I figured. Creating a PrawnOS mirror wouldn't be too too expensive from a storage side, but the bandwidth will likely be the pricey bit. I'm not familiar with how to estimate how much bandwith will be required so I'll have to spend some time figuring that out. 

Please share any opinions on Firefox replacements, and if FSF endorsement is "worth" exchanging Firefox for Icecat/Iceweasel/something else.

austin987, 2020-12-15:
IMO it would be nice to have, but I'd be concerned that they might object to blobby_kernel support being available, which I'd like to see remain. If they're okay with that, then I'm onboard. Not hard for someone to install firefox on their own if they really want to.

herbsmn, 2020-12-16:
Not sure if this is helpful or not but PureBrowser is the Firefox fork that PureOS, a FSF FSDG certified OS, uses. https://software.pureos.net/package/bin/amber/purebrowser

SolidHal, 2020-12-16:
@austin987 blobby_kernel support *should* be fine, distributing images built with a blobby kernel might be a different story. Not sure though. 
blobby_kernel support is useful as a debug too.

jonassmedegaard, 2020-12-16:
PureBrowser is no longer maintained: PureOS now ships with regular Debian-maintained Firefox ESR.

SolidHal, 2020-12-19:
@jonassmedegaard do you know how PureOS manages to get FSF endorsed while using standard Firefox ESR?

jonassmedegaard, 2020-12-19:
Quoting SolidHal (2020-12-19 18:01:10)
> @jonassmedegaard do you know how PureOS manages to get FSF endorsed while using standard Firefox ESR? 

Only FSF can answer that question: Neither me nor PureOS nor fans of FSF 
decides what FSF chooses to endorse.

I was previously hired by Purism to maintain a patched Firefox ESR 
rebranded as PureBrowser.

I still do work for Purism, but doing other tasks now (e.g. packaging 
web-apps for the matrix.org federated messaging network).

 - Jonas

-- 
 * Jonas Smedegaard - idealist & Internet-arkitekt
 * Tlf.: +45 40843136  Website: http://dr.jones.dk/

 [x] quote me freely  [ ] ask before reusing  [ ] keep private

raghavgururajan, 2021-01-15:
Mozilla Firefox can be replaced with [GNU Icecat](https://www.gnu.org/software/gnuzilla/).

Google-Chromium can be replaced with [Ungoogled-Chromium](https://github.com/Eloston/ungoogled-chromium).

ncorder, 2021-11-27:
@solidhat I ran a small organization called the EPuters Project.

We would most likely be able to aid in basic hosting of repositories ( up to about 500GB ) until your project matures more as long as you are seeking FSDG compliance.

Let me know if that would interest you, you can email me at admin@eputers.net .
I would also highly advise the project move away from Github to [NotABug](https://notabug.org/) or another platform.