SolidEva/PrawnOS #29: Use full sources.list

SolidHal, 2018-08-31:
We use a very basic sources.list right now, eventually convert to a full one like this:

```
deb http://ftp.us.debian.org/debian/ sid main
deb-src http://ftp.us.debian.org/debian/ sid main

deb http://security.debian.org/debian-security sid/updates main
deb-src http://security.debian.org/debian-security sid/updates main

# stretch-updates, previously known as 'volatile'
deb http://ftp.us.debian.org/debian/ sid-updates main
deb-src http://ftp.us.debian.org/debian/ sid-updates main
```

SolidHal, 2018-09-22:
I'd be curious to see what packages were installed by debian-security. I recall there was a specific xorg package that caused the whole system to slow when I was initially testing with xfce, but I don't think it was specific to the debian security repo. 
I'll do some further testing.

albsod, 2018-09-23:
I tried to reproduce my issues reported above but this time the result was different.
This is what I did:
1. added `deb http://security.debian.org/debian-security stretch/updates main` to `/etc/apt/sources.list`
2. ran `sudo apt-get update && sudo apt-get upgrade` which resulted in:
```
The following packages have been held back:
  chromium
The following packages will be upgraded:
  fuse ghostscript imagemagick-6-common libavcodec57 libavformat57 libavresample3 libavutil55 libcups2 libcupsimage2 libcurl3-gnutls
  libfuse2 libgs9 libgs9-common liblcms2-2 libmagickcore-6.q16-3 libmagickwand-6.q16-3 libpostproc54 libsmbclient libswresample2
  libswscale4 libvlc-bin libvlc5 libvlccore9 libwbclient0 linux-libc-dev openssh-client samba-libs vlc vlc-bin vlc-data vlc-l10n
  vlc-plugin-base vlc-plugin-notify vlc-plugin-qt vlc-plugin-samba vlc-plugin-skins2 vlc-plugin-video-output vlc-plugin-video-splitter
  vlc-plugin-visualization
```
3. rebooted the system but this time didn't experience any slow down
4. installed `firefox-esr` along with the `libjsoncpp1` dependency
5. tried to open Firefox, which failed.

I will continue using my c201 system with this setup and report any new issues in this thread.

SolidHal, 2018-09-27:
@albsod Thats very strange that firefox would fail to run properly after installing those updates, as none of them seem to directly connect to firefox.

SolidHal, 2018-10-03:
@albsod Nevermind, just tested this, and run into the same exact issue. Super strange. 
Found a possibly related issue on arm64 https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=906570

SolidHal, 2018-10-03:
Ah, without the stretch security updates, version 52.9.0 is installed. With the stretch security updated repo, 60.2.1. is installed.

albsod, 2018-10-04:
Yes, and the same thing happens with chromium (you get a newer version). Starting firefox or chromium from the terminal produces some error messages. I suspect that the lack of GPU acceleration is the problem.

SolidHal, 2018-11-27:
The versions of firefox-esr in the debian sid repo at the time of writing work properly.
It is currently 60.3.0esr-2, so specially bundling that into stretch should work and allows us to use the full sources.

JeremyRand, 2018-12-09:
Currently, `firefox-esr` crashes immediately on Stretch regardless of whether security updates are enabled, while it works fine on Buster regardless of whether security updates are enabled.  So, I think Firefox has ceased to be a reason to disable the security updates repo.  I haven't tried Chromium.

SolidHal, 2018-12-11:
@JeremyRand Yep, just noticed that myself. Really a bummer since stretch is supposed to be the stable release. I'm considering pivoting to just using buster all together. Otherwise I'll build in an apt exception to get firefox and chromium from the buster repo.

albsod, 2018-12-12:
It sounds like a good idea to use Buster, not the least considering that you're already using a more recent kernel version. Those who wish to use Debian Stable (including a stable Linux version) on the c201 could just install devsus instead.

JeremyRand, 2018-12-15:
I don't see a problem with supporting both Stretch and Buster, since some users probably prefer Stretch.  That said, I personally find Buster more usable, and I think it would be quite reasonable to make Buster the default.  (Whether you make an effort to keep Stretch working well is of course up to you, but if the community submits patches for Stretch, I don't see any reason to decline them.)

albsod, 2019-01-03:
Either version would be good as long as it's reliable (with fully working suspend/resume, wifi, etc.), secure and where you don't need to reinstall the whole system on a regular basis to get the benefits of new features and fixes.

SolidHal, 2019-01-04:
Ill pivot to focusing on buster, but will always welcome patches to include stretch support.

> where you don't need to reinstall the whole system on a regular basis to get the benefits of new features and fixes.

I hope for this to be the case soon, but suspend and resume needs to be stable before I can figure out a smooth update system. Anyway, now that the holidays have wrapped up I'm back and can focus on that. Thanks for your patience.

ericfont, 2019-01-04:
fyi, I'm on a fresh install (lxqt on c201 with stretch), and just reporting that both chromium and firefox have this problem of not starting.  Chromium does indeed complain that about gpu:

ERROR: sandbox_linux.cc(379)] InitializeSandbox() called with multiple threads in process gpu-process.

SolidHal, 2019-01-09:
Played around with buster. XFCE doesnt seem to work properly. Doesnt even load the login screen. @JeremyRand, did you have any problems with buster and xfce?

JeremyRand, 2019-01-09:
> Played around with buster. XFCE doesnt seem to work properly. Doesnt even load the login screen. @JeremyRand, did you have any problems with buster and xfce?

@SolidHal not any that I can recall; IIRC it was pretty straightforward, and my PrawnOS Buster installation uses XFCE.  I've actually been meaning to submit a PR that enables using non-stretch Debian versions (IIRC most of the prerequisites are already merged by now), just haven't had time yet.  That said, I haven't built a fresh PrawnOS image for a month or so, so I guess it's possible that something broke in the meantime (either in PrawnOS's scripts or in some upstream package).

SolidHal, 2019-01-09:
@JeremyRand Something must have broken in the upstream package as removing all of the PrawnOS customizations for the base xfce packages didn't fix it. Since all of stretch seems to work besides firefox, I'm going to continue focusing on stretch in order to keep everything (hopefully) more reliably stable. If you do get around to that pull request I'd still be happy to merge it, I have no problem with giving the option to use buster for those that are able to handle some potentially broken things.

Speaking of firefox, I've commited a patch that uses apt pinning to pull only firefox and the required dependencies from testing. Along with that, I fixed up sources.list to include all of the relevant stretch repos so this issue is resolved. Sound from firefox doesn't seem to work, but that may be due to the optimizations that PrawnOS makes to the config files.

closed 2019-01-09 by SolidHal