SolidEva/PrawnOS #30: Doesnt work(tm)

Motophan, 2018-09-04:
I really appreciated your comment on my issue in Devsus.
I build this kernel in debian 9 stretch with all prereqs, and am able to build the img files. 

It doesnt pack them correctly when looking at readme.md, it puts two .img files in a img file, and writes that to usb with the dd command.

However, this is a very good idea for a program and I am very interested in using this as a daily driver!

Motophan, 2018-09-04:
Also something nice would be a way to use the new developing drivers if we want to try them out. Like a 

type su
password
type this command
then this command
and some more commands

congratulations your using the gnugpu drivers!!

To update those bad boys, 
type this,
then this, 
here are some more
and this final command

congratulations your using the gnunewgnugpunew drivers!!

SolidHal, 2018-09-04:
Sorry about that, I working to correct the build process right now. I'm also refineing the installation instructions. Unfortunately the c201 uses at least 2 different brands of emmc which get given different partition maps in the factory. Certain models allow the image to be straight dd'd onto the internal storage while the other does not. I've been looking at the chromeos recovery tool to see how to rectify this. I believe the different emmcs also have different reported sizes. 

To help with debugging, could you prove me the output of `fdisk -l /dev/mmcblk#` where # is either 0 or 2 depending on what kernel you are running? 

I'll have the build instructions cleaned up before the end of tomorrow, and hopefully a prototype of an installation script that works for both models. 

As for the gpu drivers, they are currently very in development and don't do anything useful right now as far as I know. I plan to have a chat with the panfrost people once I get PrawOs cleaned up to see if anything from their project could be used. 

Thanks for the kind words! My hope for this project is to make installing linux on arm as easy as installing linux on x86 platforms!

Motophan, 2018-09-04:
/dev/mmcblk0p1  8671232 30752767 22081536 10.5G Microsoft basic data
/dev/mmcblk0p2    20480    53247    32768   16M ChromeOS kernel
/dev/mmcblk0p3  4476928  8671231  4194304    2G ChromeOS root fs
/dev/mmcblk0p4    53248    86015    32768   16M ChromeOS kernel
/dev/mmcblk0p5   282624  4476927  4194304    2G ChromeOS root fs
/dev/mmcblk0p6    16448    16448        1  512B ChromeOS kernel
/dev/mmcblk0p7    16449    16449        1  512B ChromeOS root fs
/dev/mmcblk0p8    86016   118783    32768   16M Microsoft basic data
/dev/mmcblk0p9    16450    16450        1  512B ChromeOS reserved
/dev/mmcblk0p10   16451    16451        1  512B ChromeOS reserved
/dev/mmcblk0p11      64    16447    16384    8M unknown
/dev/mmcblk0p12  249856   282623    32768   16M EFI System

This is blk0, blk2 does not exist.
I was actually only going to boot from usb, it beeps loudly indicating that booting from usb is not possible. Devsus booted from usb for me immediatally, and I was greeted by a prompt. I just could not get connected because there were not enough packages to connect to wifi (I tried for a few hours). I have a usb wifi adapter (mainline kernel drivers in 2.6) and was going to use it to download python and wicd to get connected, but this (when its ready) looks amazing! Very new kernel, no proprietary blobs!

SolidHal, 2018-09-05:
Hm, it should work just off usb. That makes me thing the build didn't finish. make sure you have all of the preqs installed, I just added a few more to the readme. 

I've updated the building process, so build the image with the script in the root of the repo. make sure to remove all .img files before building a new one.

SolidHal, 2018-09-08:
Since you mentioned you got it working in issue #31 I'm going to close this for now, let me know if you want it reopened.

closed 2018-09-08 by SolidHal

Motophan, 2018-09-08:
Well, WiFi isn't working yet. But is that a new ticket?
About the kpatches, I purchased the chip you recommend. Atheros AR9271

I will try to get the info you requested.

Motophan, 2018-09-09:
> @Motophan <https://github.com/Motophan>, are you talking about the built
> in wifi or the AR9271 usb wifi? The internal wifi can't be supported as
> there is no libre driver available for it.

Ar9271. Here is what I purchase. The site runs nonfree JavaScript so I am
copy paste.

Netgear wireless usb adapter. Model is WNA1100. N150 wireless usb adapter.
Smoke and pet free home. No damages or problems. Thanks for looking.

SolidHal, 2018-09-09:
I don't see any reason why the one you posted wouldn't work. Can you post your dmesg logs? 
With the adapter unplugged run:
```
dmesg -C
```
Then plug in the adapter, and run:
```
dmesg > log.txt
```
Please paste the contents of log.txt here.