SolidEva/PrawnOS #31: Test basic functionality

SolidHal, 2018-09-04:
The basic functionality needs to be confirmed. 

- Video
- HDMI out
- Sound over hdmi? 
- Speakers
- Headphone
- Keyboard
- Trackpad
- USB
- SD card reader
- Battery management
- Close lid sleep/hibernate/suspend
- wireless

Motophan, 2018-09-05:
Once I can get the image booted I will do this, can test daily builds. I have HDMI out, etc.

dimkr, 2018-09-08:
For some reason, with 4.18.5 and your patches I get system lockups when I run wpa_supplicant against my AR9271 dongle. I'm not sure which patch is the one that should fix dwc2. @SolidHal, can you point me to the relevant patch, so I can see if anything changed in the relevant areas in the kernel?

Motophan, 2018-09-08:
Me too btw. I was not sure I was doing it right.
SolidHal this thing looks really good. Was able to get to the desktop. Felt
much snappier than the ancient think pads.

I ordered a 100mbps USB 2.0 adapter that has been mainlined kernel driver
since 2.6 and was going to run a

Sudo apt update
Sudo apt upgrade

And just fingers cross it works.
:)

SolidHal, 2018-09-08:
Hey @dimkr! First of all, thank for all of your work on devsus, your project did a lot to inspire me to build this. 

The patch that fixes dwc2 (which is a modified version of the one from one of the devsus branches) can be found here https://github.com/SolidHal/PrawnOS/blob/master/resources/BuildResources/patches-tested/chromeos-dwc2-glue.patch. The build system only tries to apply the patches if the linux source folder is "fresh" aka was just untar'd, so removing the `[ "$FRESH" = true ] &&` from the following line in scripts/buildKernel.sh should make debugging easier. I had to do this as otherwise it prompts the user to reverse the patch or throws errors.   

```
[ "$FRESH" = true ] && for i in $RESOURCES/patches-tested/*.patch; do patch -p1 < $i; done
```

Another useful thing for debugging the kernel is `make inject_kernel` which if you have an already built PrawnOS image in the root of the checkout will take the newly built kernel image and inject it into the previously built image removing the need for the long filesystem build process. I'm working to document the whole make system as I post this. 

Regarding kernel version 4.18.5, I had been testing with a later version of 4.17 (maybe 4.17.9?) and ran into a ton of issues so for now I plan to stick with 4.17.2. Do you get the same lockups on 4.17.2 (just for curiosity's sake)?

@Motophan Thanks! I'm glad you got it booting! The kernel config may not have support for it enabled by default as it is pretty sparse, mostly for debug reasons. If you know what driver it uses, I'd more than happy to enable it in the config, then all you have to do is run `make kernel` and `make kernel_inject` (assuming you haven't deleted the prawnOS.img), write that prawnOS.img to a flash drive and go. 
Otherwise, you can do `make crossmenuconfig` which cross compiles make menuconfig, which will allow you to enable it yourself. If you do get it working, a pull request would be awesome.

dimkr, 2018-09-08:
> The patch that fixes dwc2

This patch doesn't seem to work for me. My C201 is an early 2015 2GB one and I use a generic AR9271 from AliExpress, plus the ThinkPenguin AR9271 dongle.

I haven't tried 4.17.x, but I did try 4.9.125 and I do see an improvement after backporting some ath9k_htc changes from 4.19-rc2. There *is* an issue with ath9k_htc and there was some regression around 4.15, that got fixed in 4.15+ AFAIK. Even with all available ath9k_htc fixes and the Chrome OS dwc2, I get that "failed to wake up blah blah" from ath9k_htc in the kernel log, so now I don't know if it's dwc2, ath9k_htc or something about the hardware to blame.

Can you share more details about your setup? The C201 model and revision, etc'?

dimkr, 2018-09-08:
@SolidHal, if you have a ready image, that would be great! I remember I tested the dwc2 patch against a 2017 4GB C201, so I'll test on both C201s I own. There could be a difference in build environments in addition to changes in the kernel and stuff - I remember it took me ages to find out that kernels built with GCC 4.9 don't boot and I'm very cautious in C201 related issues since then :)

SolidHal, 2018-09-09:
@dimkr Sorry it took so long to upload, here a link. https://archive.org/details/PrawnOSAlphaC201Libre2GBVERSION2

I added some information to the README about what I use for my build environment as well if you'd like to replicate that.

dimkr, 2018-09-09:
@SolidHal Thank you! Downloading, I'll try this image on the older C201. If things don't work with your image, it means the kernel (probably dwc2, ath9k_htc or both) is still broken, because this laptop works perfectly with the Chrome OS 3.14 kernel. Otherwise, if it works, I'll try it on the newer 4GB one too.

dimkr, 2018-09-12:
I've tested the image on my older 2GB C201 and the AR9271 is unstable. I'm on vacation until Sunday and then I have to back up all files on the newer 4GB C201, so it will take time until I get to test the image on that one as well.

dimkr, 2018-09-12:
@SolidHal I just connect to WiFi using `wpa_supplicant` and `dhclient`. Then I run `apt update`. It takes less than a minute until all traffic is stopped (`apt` freezes during download). With some kernels (probably those missing some `dwc2` fixes), `ifconfig`, `wpa_supplicant` and anything else that queries or changes network interface related parameters freezes, because something in the kernel hangs. On other kernels, traffic is stopped but I can still run stuff like `ifconfig`. When I unplug the dongle, I get some `ath9k_htc` and/or `dwc2` errors in `dmesg`.

SolidHal, 2018-09-12:
Weird, I'll try recreating those steps. A few more questions:
When using the PrawnOS alpha image, were you running off of usb or had you installed it to the emmc?
Had you ran /Installresources/InstallPackages.sh ?
Have you been using the same AR9271 dongle?
What AR9217 dongle(s) have you been using?  
With the prawnOS Alpha image, does the whole kernel hang or does the traffic just stop? 

And finally, when you get a chance, could you attach a dmesg log that includes when you run wpa_supplicant through when you unplug the dongle? 
Thanks! I hope we can get this figured out. I wouldn't be surprised if there's some other patch the chromeos kernel has that prawnOS doesn't yet have.

Motophan, 2018-09-12:
I will also dmesage in about 10 hrs. Going to use your iso from archive.org.
I build in a VM following your instructions, but going to try yours.

dimkr, 2018-09-12:
@SolidHal I was using your image, booted from USB without running the script. I use a generic Chinese dongle and a Thinkpenguin one. I booted after shutting down, ran `wpa_passphrase`, `wpa_supplicant`, `dhclient` and `apt`. After some time, only traffic stops: this isn't a full system lockup. With some kernels (don't remember if your image has this problem, too) `ifconfig` freezes too in this state.

Next week I'll try to flash your kernel to the eMMC to see if this could could be a power issue (the dongle shuts down due to under voltage) - maybe a (cheap) flash drive consumes too much voltage that's shared between the USB ports. Too bad I don't have a powered USB hub, since that could rule out this possibility. With the Chrome OS 3.14 kernel everything works perfectly, but there are changes in power management support in the kernel between 3.14 and 4.17, of course.

SolidHal, 2018-09-13:
I pulled out my other c201,a 2gb model, and my second AR9271 and managed to recreate this issue while booting off of usb. It seems to lock up completely, and eventually the kernel crashes. 

If I plug in the AR9271 after boot, I get errors just like in this issue https://github.com/qca/open-ath9k-htc-firmware/issues/136. Do you as well @dimkr ?

I realize now why I'm not seeing this issue on my main 4GB c201; Because I have my ath9271 soldered in to the webcam wiring. (or maybe its a 2GB vs 4GB issue, but I doubt it?) 

Things I'll test:
Running off of usb vs emmc
4gb vs 2gb models
libreboot vs stock coreboot

The issue I linked above also mentions not having this problem when using a usb hub, but doesn't mention if it is powered or unpowered. I'll test with one of each.

Its sounding more and more like a kernel issue, especially if everything works fine with the chrome os 3.14 kernel.  After testing the above things I'll start digging into the kernel. If using an unpowered usb hub fixes this, a pretty nasty usb bug is hiding somewhere.

samdima, 2018-09-13:
> If I plug in the AR9271 after boot, I get errors just like in this issue qca/open-ath9k-htc-firmware#136. Do you as well @dimkr ?

I get three kinds of errors: these you mentioned, something related to `ath` and timeout (I remember the number 500 or 5000 mentioned in the logs), USB connection/disconnection related messages and errors due to inability to transfer the ath9k_htc firmware, right after the error you see.

SolidHal, 2018-09-14:
I'm going to move this conversation over to #38 to better track this issue.

Motophan, 2018-09-14:
Absolute unit of an os. Dmsg coming soon

SolidHal, 2018-09-19:
@Motophan, the issue with the ath9k devices is now fixed! Everything needed to fix it is commited so feel free to try out a build. I'll have another compiled image uploaded soon if you don't want to build it.

Motophan, 2018-09-21:
oh you are awesome. This is the first operating system I have ever compiled
in my life.
 It was fun. Thank you for making this, I really like the idea of a libre
os with a libre devise. Thinkpad is too old, low battery life, slow.

SolidHal, 2018-09-25:
@Motophan I'm glad you like it!

SolidHal, 2018-10-04:
Closing this as all functions have been tested, other issues opened for specific problems.

closed 2018-10-04 by SolidHal