SolidEva/PrawnOS #32: Buffer I/O error / Print_req_error on dd to emmc

harshbarj, 2018-09-06:
Seems like I have one of the emmc chips that has a different layout. Everything works great up to when I dd the image to the emmc. usb created on a linux mint 18.3 64-bit with all required dependencies installed. Here is my partition information. If you need something else, just ask.
----------------Start-------------------
Disk /dev/mmcblk2: 14.7 GiB, 15762194432 bytes, 30785536 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 39EBCEA6-6B2F-3843-BE5D-779153C4E211

Device            Start      End  Sectors  Size Type
/dev/mmcblk2p1  8704000 30752767 22048768 10.5G Microsoft basic data
/dev/mmcblk2p2    20480    53247    32768   16M ChromeOS kernel
/dev/mmcblk2p3  4509696  8703999  4194304    2G ChromeOS root fs
/dev/mmcblk2p4    53248    86015    32768   16M ChromeOS kernel
/dev/mmcblk2p5   315392  4509695  4194304    2G ChromeOS root fs
/dev/mmcblk2p6    16448    16448        1  512B ChromeOS kernel
/dev/mmcblk2p7    16449    16449        1  512B ChromeOS root fs
/dev/mmcblk2p8    86016   118783    32768   16M Microsoft basic data
/dev/mmcblk2p9    16450    16450        1  512B ChromeOS reserved
/dev/mmcblk2p10   16451    16451        1  512B ChromeOS reserved
/dev/mmcblk2p11      64    16447    16384    8M unknown
/dev/mmcblk2p12  249856   315391    65536   32M EFI System

Partition table entries are not in disk order.
-----------------------End--------------------------

Laptop boots to recovery mode after rebooting.

SolidHal, 2018-09-08:
Is that the fdisk -l output after trying to write to the emmc?

harshbarj, 2018-09-10:
It is. I booted off the usb drive and dd'd the image to mmcblk2. I plan on doing a restore and then checking to see if it matches.

harshbarj, 2018-09-10:
I did a chrome os recovery and then did fdisk again. A quick look seems to show only a slight change to the end of /dev/mmcblk2p1.

----------------------------start---------------------------------------
Disk /dev/mmcblk2: 14.7 GiB, 15762194432 bytes, 30785536 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 1BF04A15-56AE-944D-A820-30711704912A

Device            Start      End  Sectors  Size Type
/dev/mmcblk2p1  8704000 30781439 22077440 10.5G Microsoft basic data
/dev/mmcblk2p2    20480    53247    32768   16M ChromeOS kernel
/dev/mmcblk2p3  4509696  8703999  4194304    2G ChromeOS root fs
/dev/mmcblk2p4    53248    86015    32768   16M ChromeOS kernel
/dev/mmcblk2p5   315392  4509695  4194304    2G ChromeOS root fs
/dev/mmcblk2p6    16448    16448        1  512B ChromeOS kernel
/dev/mmcblk2p7    16449    16449        1  512B ChromeOS root fs
/dev/mmcblk2p8    86016   118783    32768   16M Microsoft basic data
/dev/mmcblk2p9    16450    16450        1  512B ChromeOS reserved
/dev/mmcblk2p10   16451    16451        1  512B ChromeOS reserved
/dev/mmcblk2p11      64    16447    16384    8M unknown
/dev/mmcblk2p12  249856   315391    65536   32M EFI System

Partition table entries are not in disk order.
----------------------------end----------------------------------------

SolidHal, 2018-09-10:
Unfortunately you can't just dd the image to the internal emmc on many models as the kernel partition cant be located before a certain sector. I'm not sure why this is, but I worked around it by providing a custom partition map. 

On the same note, I don't believe your dd command worked correctly as the partition map of the image is much simpler than the one you posted at the start of this thread (There are only two partitions). 

Please install try installing PrawnOS to the internal drive using the script at the root of the drive named `InstallToInternal.sh`

Also, I have made quite a few improvements in the last few days to make it easier to install and use PrawnOS so make sure you have the most recent commit, and then run `make clean_all`.

I have also updated the readme with a bunch of information on how to build PrawnOS as it now uses `make` instead of running the scripts manually so check all of that out. 

Please let me know if this works for you.

harshbarj, 2018-09-10:
That worked. It's booting off the emmc now. Running The Install Resources script. Thanks for your help, and your work on this! I'll make sure to test everything on the laptop and report any problems. Only thing I can't test is HDMI out as I do not have the cable.

harshbarj, 2018-09-10:
I should add, the readme says to run "sudo make image", but for me that just resulted in a bunch of errors (I'll redo it later and give you the errors). I simply ran "./buildPrawnOS.sh" and that worked. I then dd'd the image to a usb drive.

harshbarj, 2018-09-10:
Ok here is what is what I get when I try to make the file as suggested in the redme.  But as I said, I simply ran the build prawn os script and that worked fine!

-------------------------------start---------------------------------
make clean_img
make[1]: Entering directory '/home/harshbarj/Documents/Prawnos'
rm PrawnOS-*-c201-libre-*GB.img
makefile:20: recipe for target 'clean_img' failed
make[1]: Leaving directory '/home/harshbarj/Documents/Prawnos'
makefile:44: recipe for target 'image' failed
-------------------------------end------------------------------------

SolidHal, 2018-09-10:
Awesome, I'm glad you got it working. 
Whoops! Sorry about that make error, it is fixed now. 
I'm going to close this issue for now, let me know if there's any reason to reopen it.

closed 2018-09-10 by SolidHal