SolidEva/PrawnOS #37: Make script to expand to full usb size

SolidHal, 2018-09-13:
Want to be able to use install image as live image, taking advantage of the full size of the usb drive we are installed on. 
Have basic concept in `scripts/InstallScripts/InstallToUSB.sh` but after reboot usb is not bootable. 
Also tried editing partition table with fdisk manually:
fdisk /dev/sda
p
g
n
1
8192
73727
n
2
(enter)
(enter)
w

But the result doesn't boot...

Things to try: Try saying yes to remove ext4 info on part 2?
See what raspbian does to expand in sd cards

jonassmedegaard, 2018-09-22:
I would advice against using fdisk non-interactively.  Instead use a tool designed for scripting, like sfdisk or parted.

You are welcome to borrow from my script used for <https://box.redpill.dk/>: <http://source.homebase.dk/box.git/tree/skel/multistrap-hooks/completion_80_rootscript_disk.sh>

jonassmedegaard, 2018-09-26:
Uhm, I just had a look at your latest image, and it seems it still uses that special GPT disk layout common to chromebooks.  Are you sure it is even possible to use a conventional fdisk tool with the c201?

Unless something has changed in the recent year (which is what assumed since you talked about conventional fdisk initially above - but now that you say your're no expert on this I roll back again), I believe to get a bootable disk format you need to use gdisk with special partition types and names.

SolidHal, 2018-09-27:
> Uhm, I just had a look at your latest image, and it seems it still uses that special GPT disk layout common to chromebooks.

That is true. I don't plan on changing that either, as chromebooks require the special layout to boot properly. You can see how the layout is made in https://github.com/SolidHal/PrawnOS/blob/master/scripts/buildDebianFs.sh The create_image() function uses parted and cgpt to make the proper partition map. 

>  Are you sure it is even possible to use a conventional fdisk tool with the c201?
Thats a good point

I'll have to look into using cgpt  to resize (if possible) 
Otherwise I'll probably end up doing something like I do in the InstallToInternal script and format a partition map to write to it with sfdisk.

jonassmedegaard, 2018-09-27:
Quoting SolidHal (2018-09-27 02:27:47)
> That is true. I don't plan on changing that either, as chromebooks require the special layout to boot properly. You can see how the layout is made in [...]

Either I am missing something in your description above, or you are 
mistaken and should *not* use sfdisk:

Chromebook firmware require GPT formatted boot disk, in so-called hybrid 
mode.

sfdisk does *not* support GPT - but may misdetect hybrid mode GPT as DOS 
MBR and break stuff.

parted has somewhat limited GPT support.

cgpt has limited GPT specific to Chromebooks.

gdisk has full support for GPT.

If you want to try do tricks like resizing, I recommend to look at 
gdisk:  http://rodsbooks.com/gdisk/

You might also (get totally confused or) be helped by the Google page on 
disk formatting, linked at the bottom og above gdisk page: 
https://www.chromium.org/chromium-os/chromiumos-design-docs/disk-format

 - Jonas

-- 
 * Jonas Smedegaard - idealist & Internet-arkitekt
 * Tlf.: +45 40843136  Website: http://dr.jones.dk/

 [x] quote me freely  [ ] ask before reusing  [ ] keep private

SolidHal, 2018-10-04:
@jonassmedegaard I've been looking into using gdisk to build the partition map for the emmc without much luck. Using gdisk to create the correct kernel partiton and root partitions (at the correct offsets) doesn't lead to a bootable system. 

To look into this further, I flashed one of my c201s with stock chromeos and booted from a usb with lunix and gdisk. gdisk doesn't even recognize that /dev/mmcblk even HAS a gpt partition map. Cgpt, and even fdisk recognize there is a partition map on the emmc. The only thing that may be tripping up gdisk is the fact that the primary GPT header is set to IGNORED by default, forcing the use of the secondary header. 
For kicks, I even wrote a new partition map to the emmc using gdisk, which worked flawlessly and even detected that is should use the secondary gpt table. The partitions it created were available under /dev/mmcblk2p1 and p2 as expected. When I then go back in to look at the emmc with gdisk, it still does not recognize the partition mapping at all. 

So, it seems gdisk does not properly support whatever weird gpt the coreboot requires the emmc to have. Or maybe there are bugs left in the mainline kernel support for the emmc. 

I will still look into gdisk for resizing a usb boot drive. 

Thanks for all the info so far
-hal

jonassmedegaard, 2018-10-04:
Quoting SolidHal (2018-10-04 05:51:11)
> So, it seems gdisk does not properly support whatever weird gpt the 
> coreboot requires the emmc to have. Or maybe there are bugs left in 
> the mainline kernel support for the emmc.

Sorry to hear that you didn't get it working yet.

I recommend to file a bugreport - while your research and attempts is 
still fresh in your mind.

 - Jonas

-- 
 * Jonas Smedegaard - idealist & Internet-arkitekt
 * Tlf.: +45 40843136  Website: http://dr.jones.dk/

 [x] quote me freely  [ ] ask before reusing  [ ] keep private

SolidHal, 2018-10-04:
Tested with the 3.14 chromeos kernel, gdisk still doesnt detect the emmc's partition map, so not a kernel bug.

SolidHal, 2018-10-15:
gdisk doesnt detect the emmcs partition map because gdisk doesnt read/modify the backup partition map and the c201 is hardcoded to use only the back partition map. The primary partition map is in an unchangable part of the emmc that has an "IGNOREME" tag. 

Looking at the chromeos recovery process it seems like they use cgpt for everything. 

Heres the emmc partition map dumped by cgpt after a fresh recovery: 

```
       start        size    part  contents
           0           1          PMBR (Boot GUID: 7CE2DD5A-EF83-0247-A2FF-05965ADF2E22)
           1           1 IGNORED  Pri GPT header
    30785503          32          Sec GPT table
     8704000    22077440       1  Label: "STATE"
                                  Type: Linux data
                                  UUID: B663593D-F9F5-F04F-90B6-88690A87BC90
       20480       32768       2  Label: "KERN-A"
                                  Type: ChromeOS kernel
                                  UUID: B6DED3FF-55A5-A249-B023-83C4C96193FB
                                  Attr: priority=1 tries=6 successful=1
     4509696     4194304       3  Label: "ROOT-A"
                                  Type: ChromeOS rootfs
                                  UUID: EC18BDF4-5F74-2C45-AB3F-A73300C53FFA
       53248       32768       4  Label: "KERN-B"
                                  Type: ChromeOS kernel
                                  UUID: B41EC191-68F0-9045-8AAF-3389B7A438E0
                                  Attr: priority=0 tries=15 successful=0
      315392     4194304       5  Label: "ROOT-B"
                                  Type: ChromeOS rootfs
                                  UUID: 1384D932-90D2-654D-AA24-102E4425EF3F
       16448           1       6  Label: "KERN-C"
                                  Type: ChromeOS kernel
                                  UUID: 7FEBD683-66F3-E84E-8F3F-6016B7FC1A75
                                  Attr: priority=0 tries=15 successful=0
       16449           1       7  Label: "ROOT-C"
                                  Type: ChromeOS rootfs
                                  UUID: 5E929AF3-AD37-7349-8A2C-3F65ECFD3A73
       86016       32768       8  Label: "OEM"
                                  Type: Linux data
                                  UUID: 4D8FB2B6-731F-B04A-ACA7-8E94FE5E1CB9
       16450           1       9  Label: "reserved"
                                  Type: ChromeOS reserved
                                  UUID: 38B962D7-E470-0A43-865F-6560D767CB74
       16451           1      10  Label: "reserved"
                                  Type: ChromeOS reserved
                                  UUID: 544D5A16-BA7A-DA40-B019-EEAE66A26D5F
          64       16384      11  Label: "RWFW"
                                  Type: ChromeOS firmware
                                  UUID: 7F111A09-ECB4-EB4A-9DB2-B7162D8C5093
      249856       65536      12  Label: "EFI-SYSTEM"
                                  Type: EFI System Partition
                                  UUID: 48A3ADD5-E9B4-AB47-90B7-9FE1D9C9D25F
    30785535           1          Sec GPT header
                                  Sig: [EFI PART]
                                  Rev: 0x00010000
                                  Size: 92
                                  Header CRC: 0x4eb39c6f 
                                  My LBA: 30785535
                                  Alternate LBA: 1
                                  First LBA: 34
                                  Last LBA: 30785502
                                  Disk UUID: 73C380DB-31B6-B343-A674-05AF1C6C3FCC
                                  Entries LBA: 30785503
                                  Number of entries: 128
                                  Size of entry: 128
                                  Entries CRC: 0xe7d15af9 
```

SolidHal, 2018-10-15:
I found lots of success using the following commands to make the emmc partition map:
```
cgpt add -i 1 -b 20480 -s 65536 -t kernel -l KERNEL -S 1 -T 5 -P 10 /dev/mmcblk2
cgpt add -i 2 -b 86016 -s 30699486 -t data -l Root /dev/mmcblk2
```
to replace the sfdisk command. Test to see if existing partitions are an issue. 
```

SolidHal, 2018-10-24:
The issue is nether cgpt nor gdisk properly handle the backup partition, which the c201 uses as primary for convoluted hardware reasons. sfdisk gpt support has been around for a while, and works for creating the necessary partition mac on the emmc. 
The only alternative is to make the 16GB image when we make the 2gb image (like devsus) and use that for writing to the emmc. 
This has the downside of increasing the install time by making the base image larger and forcing a 16GB write to the emmc. If issues arrise, we can go back to the "devsus" method later. 

Closing this issue as the usb installer is complete and tested.

closed 2018-10-24 by SolidHal