SolidEva/PrawnOS #40: Bluetooth support

dimkr, 2018-09-19:
CSR8150 Bluetooth dongles (which cost around $1-2 on eBay and Aliexpress) work with Linux-libre! `CONFIG_BT_HCIBTUSB` and `CONFIG_BT_INTEL` result in `btusb.ko` and `btintel.ko`, which support this chip bloblessly.

I had this dongle for about a year and only today I had the time and motivation to get my Bluetooth speakers to work with the C201.

To test the dongle: run `bluetoothctl` (`scan`, `devices, ``pair` and `connect`), install `pulseaudio` with `pulseaudio-module-bluetooth` enabled and make the Bluetooth sink the default using `pactl`. Then, everything including `speaker-test` will play sound through the speakers.

dimkr, 2018-09-19:
For reference: https://github.com/dimkr/devsus/commit/0e31c14212ffe8cb809bfb67822e7d8f3bcce420

SolidHal, 2018-09-20:
Awesome. I'm sure theres some decent interest in bluetooth functionality. I'll enable those when I get home later!

SolidHal, 2018-09-21:
@dimkr I've enabled the required kernel config options. Would you mind giving it a test? I don't own the required dongle.

dimkr, 2018-09-21:
> Would you mind giving it a test?

Sure, let me know when there's a prebuilt image with these changes.

SolidHal, 2018-09-25:
@dimkr Theres a prebuilt image under "releases" with the change. Thank for testing it!

dimkr, 2018-09-28:
The dongle has been detected correctly in `dmesg`.

closed 2018-09-28 by dimkr

duvh93, 2019-06-20:
Hello! Is it CSR8150 or CSR8510? I barely manage to find the former dongles for sale on the internet

SolidHal, 2019-06-20:
@duvh93 The CSR8150 is what dimkr had tested with, so I don't know if the CSR8510 would work. Can't seem to find a definitive answer whether the linux kernel bluetooth drivers support it. You may have more luck if you search for a specific brand/model of bluetooth device in this case. 

If you end up picking one up and it doesn't work, let me know.

dimkr, 2019-06-22:
@duvh93 Probably a typo

SolidHal, 2019-06-24:
Looks like it is a typo, fixed in the README. thanks @duvh93 and @dimkr

rk-zero, 2020-01-11:
Bluetooth support is great in 1.0.0 - even my quirky ms mouse now works out of the box with correct update intervals.

SolidHal, 2020-01-11:
@rk-zero thats great to hear!