SolidEva/PrawnOS #42: Add support to USB WIFI htc_7010

albsod, 2018-09-22:
Could you please add support for the other Atheros wifi dongles that use the htc_7010 firmware? Right now the binary already gets built, but you need to do the following to install it:

`cp PrawnOS/build/open-ath9k-htc-firmware/target_firmware/htc_7010.fw /lib/firmware/ath9k_htc/htc_7010-1.4.0.fw`

Note that you need to rename the file to make it work. The device works without any issues thanks to your kernel patch while previously I lost the connection all the time.

By the way, I created the following udev rule to turn off the blinking LED which I found very annoying:
`ACTION=="add", ATTRS{idVendor}=="0cf3", ATTRS{idProduct}=="7015", ATTR{brightness}="0"`

Thanks!

SolidHal, 2018-09-22:
@albsod  The htc_7010 firmware should now be built into the filesystem. Would you mind giving it a test? 

Where/ how do you create that udev rule? 
That sounds like something that could be added to the wiki when it gets going, but having the info here for now would be great.

albsod, 2018-09-23:
@SolidHal Nice! Do I need to reinstall the OS to test? If so, I'd like to do it later because I just reinstalled.

To create the udev rule, I just put contents above in the file
`/etc/udev/rules.d/100-usbwifi.rules`

SolidHal, 2018-09-25:
@albsod No need to reinstall if you don't want to. Just boot off of usb and follow the instructions in the README near the bottom of the Install section to test your wifi dongle. You can find precompiled images under releases if you don't want to go through the build process.

albsod, 2018-10-10:
I've rebuilt and reinstalled and can confirm that the wifi dongle is working as it should.

albsod, 2019-06-29:
There has been a regression here. I've just installed PrawnOS-buster-Alpha-c201-libre-2GB-git-39dcd2c4da58e0bc8a262c9a4b5b739d12a2ba95 and discovered that the htc_7010 firmware is no longer copied into /lib/firmware/ath9k_htc/

SolidHal, 2019-06-30:
@albsod Whoops, I'm away for the next few days but I'll try and get a patch pushed and a new release uploaded.
Thanks for the report.

SolidHal, 2019-06-30:
61aa9090dbaa3c7de348f44217c9519397570690 should fix this issue.

albsod, 2019-06-30:
Thanks! I had already manually copied over the file which fixed it for me. I still experience a slight regression in that the device isn't recognized during boot but only if inserted when inside XFCE, i.e. I need to take it out and put it back in again.

SolidHal, 2019-07-12:
61aa909 seems to have fixed it, closing.

closed 2019-07-12 by SolidHal