SolidEva/PrawnOS #51: Distorted sound

albsod, 2018-10-14:
After reinstalling the system built from e063a50fc5159a92a870418aea105a1f70f9f7e6 (October 9) sound has become distorted on my machine. It worked fine on 5356c583463b5749bb36afe1d8cbfa1672780ea1 (September 20). I've looked through the pulseaudio config file but can't see anything strange.

SolidHal, 2018-10-15:
hmmm. I'll start trying to recreate this. 

possible (weird) causes:
 - enabling bluetooth support? 06c9d33a8b462bdd0542ff56c3f8943400a75261
 - Other kernel config changes? 

What sound source are you using?
Does having usb devices plugged in effect it? 
Does it change depending on if the laptop is plugged in?

SolidHal, 2018-10-20:
I tested with the current git version (8e5a0690a993b18965449e58e9c7064d0adc2366) for which I'm uploading a release for right now. 
I freshly installed PrawnOS.
I played a song through youtube on firefox-esr as my source.

I played the song through the speakers at half volume, which sounded fine. At full volume they started to sound a little tinny, but the c201's speakers are unfortunately cheap and a little tinny even on the supplied ChromeOS. 
I tested through headphones (Shure se215's) plugged into the audio jack which sounded fine. 

I then tried both the speakers and headphones with power plugged in/not plugged in and with and without multiple usb devices attached, while transferring a large file off of a flash drive. 

I was unable to replicate any distortion. 

@albsod When you get a chance, could you let me know:
 1. What sound source and what sound output are you using?
 2. Are you booted off of emmc or usb or sd card?
 3. Are you running lxqt or xfce?
 4. What other processes were you running when you experienced this issue if any? 
 5. What types of sounds/songs make it most apparent (eg. bassy electronic, softer classical, etc) 

Thanks!

albsod, 2018-11-01:
@SolidHal Sorry for the late reply. I rebuilt from a Oct 30 snapshot and reinstalled the system. 

I still get either silence or highly distorted sound when playing audio streams **in Firefox** from e.g. sr.se, dr.dk, soma.fm or BBC. Try for example https://www.bbc.co.uk/radio/player/bbc_radio_one

I first thought that it might have been a problem with the AAC codec, but AAC files are fine in VLC, so this seems to be just a firefox thing. Do you experience the same problem?

SolidHal, 2018-11-07:
@albsod ah, I can recreate the issue in firefox with bbc radio one through the speakers. I didnt try the other sites or headphones.

not sure what causes it, but chromium seems to work just fine with bbc radio one so I'm thinking this is a firefox issue

SolidHal, 2018-12-12:
@albsod seems this issue might be a firefox problem as I just tested and this happens on my desktop install as well. Heres a thread about it. https://support.mozilla.org/en-US/questions/949779

albsod, 2018-12-12:
I see, good. Then I suggest renaming the tread to "Distorted sound in Firefox" or something.

albsod, 2019-01-03:
With 60.4.0esr from Debian Buster (but on Stretch otherwise), sound is working fine again.

SolidHal, 2019-01-14:
@albsod The current release has firefox from buster, but everything else from stretch. Sound seems to work fine from any site I tested.

albsod, 2019-01-14:
@ SolidHal Yes, I agree.

SolidHal, 2019-01-14:
Awesome, closing

closed 2019-01-14 by SolidHal