SolidEva/PrawnOS #52: Add kernel upgrade function

SolidHal, 2018-10-21:
Make new script that writes a bundled, built kernel image to the desired boot disk as a way to upgrade the kernel. 
Look into in place kernel upgrades that can be done from the booted system.

SolidHal, 2019-06-25:
A rough version of this now exists with commit 6baa47914928a07b069dfe9d26b4ead50b42516a
It cannot be done on a running system, but does allow for the kernel to be updated. Without running our own apt repos, which isn't out of the question in the future, in place kernel upgrades will be difficult to offer. 

This script should suffice for now.

closed 2019-06-25 by SolidHal

filberg, 2020-04-22:
I've been recently trying to come up with some ideas to automate the kernel upgrade and make it more convenient for a non-technical user.

These are the 2 basic approaches I've been reasoning on:

1- Do The Right Thing and learn how to properly package a kernel deb to be hosted somewhere (I have some hosting available if needed) on a deb repo, and configure apt so that kernel upgrades use that specific channel.
Needless to say, this is for sure the best approach in terms of maintainability and possible "upstreamability", but it has a higher learning curve for me.  

2- Reinvent the wheel and implement a custom upgrade process. This process would be packaging the upgraded kernel image and custom installation scripts in a simple archive that is simply hosted as a bare file here on the github release tab (or any other server), and check for any upgraded archives via a apt hook, whenever the user runs apt update.
The idea is to download the aforementioned package, store it in a well-known path in the currently mounted /, and have a properly-crafted initramfs in place so that during the first reboot, after the root filesystem is decrypted and mounted, and before proceeding to boot, that well-known path is checked for the presence of the "special kernel upgrade files" and, if found, the kernel image is flashed in the kernel partition, any installation scripts are run, and the system rebooted.
This should finally result in booting the new kernel.
Needless to say, this second approach is ugly and uses totally unorthodox methods, but it's the quickest one I was able to think of.

That being said, I'd like some feedback about my proposals, and I put myself at disposal to work on whatever path is deemed to best fitting for prawnos as a project.
I really want to achieve an easier way to upgrade the kernel on this cute piece of hardware that the c201 is..

For now, thanks for all you've been doing with this project.