SolidEva/PrawnOS #53: Add instructions for flashing libreboot from within PrawnOS

SolidHal, 2018-10-22:
According to the arch wiki, stock flashrom is _NOT_ recommended for the following reasons:
 - it will not write firmware successfully to the ROM of the Chromebook unless it already been programmed externally (i.e. flashing by another device over SPI with SOIC clip
 - it's missing operations like --wp-disable, --wp-status

See: https://wiki.archlinux.org/index.php/Chrome_OS_devices/Custom_firmware#Manually_with_flashrom

The first issue is much bigger than the second, as it sounds like it straight up will not work for most people as most have not flashed their ROM externally. 

So, we need to use chromeos flashrom: https://chromium.googlesource.com/chromiumos/third_party/flashrom/

To build our own we need to know: 
1) what branch to use for the c201, one of the veyron ones or just master? 
2) what config options to be specified. 

Don't attempt flashing with any test flashrom builds unless you have a way to program the c201's ROM externally, or you may end up with a machine that won't boot!

SolidHal, 2018-10-22:
I pulled flashrom from the chromeos recovery disk, installed libfdt1, but its just segfaults :/ 

Looks like using flashrom pulled from chromeos won't work, we'll just have to wait until we can build it ourselves.

riwale, 2019-09-16:
you can pull the flashrom version by chrooting into the chromeos recovery disc, instead of only taking the binary.

SolidHal, 2019-10-15:
iirc chromeos uses a custom build of flashrom. Source is likely available though.