SolidEva/PrawnOS #56: Suspend in XFCE is broken

SolidHal, 2018-10-23:
Wake after suspend sometimes fails, even if lockscreen is disabled.

Machine will wake up, but hangs at some point after suspend exit. Sometimes enough time to run dmesg. Sometimes see failure to read sectors on mmc? 

test by using `sudo systemctl suspend` when logged in. 
- happens maybe one in 10 times

Strong indicator is wifi failing to reconnect after wake. 
Touching trackpad may improve chance of failure?

Try:
Bring in config changes/improvements
Hunt down and fix these errors:

```
[   65.176277] elan_i2c 4-0015: invalid report id data (ff)
```
may be related to:
```
[    3.467669] cros-ec-spi spi0.0: Command xfer error (err:-11)
[    3.471141] cros-ec-i2c-tunnel ff110000.spi:ec@0:i2c-tunnel: Error transferring EC i2c message -11
```

and these messages:
```
[    1.813426] rockchip-dp ff970000.dp: Link Training Clock Recovery success
[    1.816901] rockchip-dp ff970000.dp: Link Training success!
[    1.928068] rockchip-dp ff970000.dp: Timeout of video streamclk ok
[    1.928091] rockchip-dp ff970000.dp: unable to config video
```

SolidHal, 2018-10-23:
For now, disable suspend on lid close.

SolidHal, 2018-11-01:
TODO: see if this is due to the gpu 
TODO: either way, reduce number of virtual terminals

SolidHal, 2018-11-06:
Something like this may be partially related, although the error I am seeing doesn't quite match:
https://www.mail-archive.com/linux-kernel@vger.kernel.org/msg1604450.html

SolidHal, 2018-11-06:
That doesnt apply directly, the rk3288 doesnt have the clock `PCLK_VIO_GRF`.
It also seems to have reset parameters that arent pressent in the rk3399
```
resets = <&cru SRST_EDP>;
		reset-names = "dp";
```

It also depends on a phy that rk3399 doesnt have:
```
		phys = <&edp_phy>;
		phy-names = "dp";
```

Which is defined as:

```
grf: syscon@ff770000 {
		compatible = "rockchip,rk3288-grf", "syscon", "simple-mfd";
		reg = <0x0 0xff770000 0x0 0x1000>;

		edp_phy: edp-phy {
			compatible = "rockchip,rk3288-dp-phy";
			clocks = <&cru SCLK_EDP_24M>;
			clock-names = "24m";
			#phy-cells = <0>;
			status = "disabled";
		};
```

So, the base rk3288 dtsi defines the edp node as:
```

	edp: dp@ff970000 {
		compatible = "rockchip,rk3288-dp";
		reg = <0x0 0xff970000 0x0 0x4000>;
		interrupts = <GIC_SPI 98 IRQ_TYPE_LEVEL_HIGH>;
		clocks = <&cru SCLK_EDP>, <&cru PCLK_EDP_CTRL>;
		clock-names = "dp", "pclk";
		phys = <&edp_phy>;
		phy-names = "dp";
		resets = <&cru SRST_EDP>;
		reset-names = "dp";
		rockchip,grf = <&grf>;
		status = "disabled";

		ports {
			#address-cells = <1>;
			#size-cells = <0>;
			edp_in: port@0 {
				reg = <0>;
				#address-cells = <1>;
				#size-cells = <0>;
				edp_in_vopb: endpoint@0 {
					reg = <0>;
					remote-endpoint = <&vopb_out_edp>;
				};
				edp_in_vopl: endpoint@1 {
					reg = <1>;
					remote-endpoint = <&vopl_out_edp>;
				};
			};
		};
	};
```

Later, in the veyron-chromebook dtsi it enables the edp, and gives pinctrl these values:

```
&edp {
	status = "okay";

	pinctrl-names = "default";
	pinctrl-0 = <&edp_hpd>;

	ports {
		edp_out: port@1 {
			reg = <1>;
			#address-cells = <1>;
			#size-cells = <0>;
			edp_out_panel: endpoint {
				reg = <0>;
				remote-endpoint = <&panel_in_edp>;
			};
		};
	};
};
```

Then veyron-speedy then makes these modifications:

```

&edp {
	/delete-property/pinctrl-names;
	/delete-property/pinctrl-0;

	force-hpd;
};
```

This `force-hpd` seems interesting, I recall a patch mentioning this

SolidHal, 2018-11-06:
The pinctrl fields were removed because:

```
- pinctrl-names: Names corresponding to the chip hotplug pinctrl states.
- pinctrl-0: pin-control mode. should be <&edp_hpd>
```

And the edp is not hotpluggable. This is also also why this parameter is added:
```
"force-hpd": forcing regard panel is connected for some panel without Hotplug pin.
```

SolidHal, 2018-11-08:
if the system is set to suspend when the lid is closed, this happens:

```
[ 1573.400503] elan_i2c 4-0015: invalid report id data (ff)
[ 1575.498556] elan_i2c 4-0015: invalid report id data (ff)
[ 1577.131555] elan_i2c 4-0015: invalid report id data (ff)
[ 1590.490018] PM: suspend entry (deep)
[ 1590.494060] PM: Syncing filesystems ... done.
[ 1590.517144] Freezing user space processes ... 
[ 1610.523182] Freezing of tasks failed after 20.000 seconds (1 tasks refusing to freeze, wq_busy=0):
[ 1610.533757] pkexec          D    0  1174    585 0x00000001
[ 1610.540059] [<c09dae54>] (__schedule) from [<c09db1cc>] (schedule+0xbc/0xd8)
[ 1610.548076] [<c09db1cc>] (schedule) from [<c0380030>] (flush_old_exec+0x200/0x6a8)
[ 1610.556673] [<c0380030>] (flush_old_exec) from [<c03cc8e0>] (load_elf_binary+0x2b4/0x1018)
[ 1610.566039] [<c03cc8e0>] (load_elf_binary) from [<c037fb84>] (search_binary_handler+0x78/0x1a4)
[ 1610.575889] [<c037fb84>] (search_binary_handler) from [<c0380ea4>] (do_execveat_common+0x4d0/0x714)
[ 1610.586128] [<c0380ea4>] (do_execveat_common) from [<c038111c>] (do_execve+0x34/0x3c)
[ 1610.595001] [<c038111c>] (do_execve) from [<c0381348>] (sys_execve+0x2c/0x30)
[ 1610.603095] [<c0381348>] (sys_execve) from [<c0201000>] (ret_fast_syscall+0x0/0x4c)
[ 1610.611745] Exception stack(0xee2f1fa8 to 0xee2f1ff0)
[ 1610.617513] 1fa0:                   016dc4c0 016e0c00 016dc4c0 bee42a68 01700988 b6d38b58
[ 1610.626779] 1fc0: 016dc4c0 016e0c00 004381e8 0000000b 016f8090 016ddbf8 b54314c8 016dc600
[ 1610.636017] 1fe0: b6cbe7bd bee407b4 0042696b b6cbe716
[ 1610.641839] OOM killer enabled.
[ 1610.645540] Restarting tasks ... done.
[ 1610.659584] PM: suspend exit
[ 1610.664450] PM: suspend entry (s2idle)
[ 1610.668702] PM: Syncing filesystems ... done.
[ 1610.680516] Freezing user space processes ... (elapsed 0.002 seconds) done.
[ 1610.690334] OOM killer disabled.
[ 1610.693943] Freezing remaining freezable tasks ... (elapsed 0.001 seconds) done.
[ 1610.703481] Suspending console(s) (use no_console_suspend to debug)
[ 1682.679524] usb usb2: root hub lost power or was reset
[ 1682.690609] mmc_host mmc1: Bus speed (slot 0) = 400000Hz (slot req 400000Hz, actual 400000HZ div = 0)
[ 1682.745649] mmc1: queuing unknown CIS tuple 0x80 (2 bytes)
[ 1682.747150] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[ 1682.748649] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[ 1682.751344] mmc1: queuing unknown CIS tuple 0x80 (7 bytes)
[ 1682.784078] rockchip-dp ff970000.dp: Timeout of video streamclk ok
[ 1682.784081] rockchip-dp ff970000.dp: unable to config video
[ 1682.804596] mmc_host mmc1: Bus speed (slot 0) = 148500000Hz (slot req 150000000Hz, actual 148500000HZ div = 0)
[ 1682.951770] elan_i2c 4-0015: invalid report id data (ff)
[ 1683.017427] usb 2-1: reset high-speed USB device number 2 using ehci-platform
[ 1683.335472] dwmmc_rockchip ff0d0000.dwmmc: Successfully tuned phase to 193
[ 1683.422758] OOM killer enabled.
[ 1683.426275] Restarting tasks ... done.
[ 1683.438233] PM: suspend exit
[ 1683.438483] mmc_host mmc2: Bus speed (slot 0) = 400000Hz (slot req 400000Hz, actual 400000HZ div = 0)
[ 1683.516798] mmc_host mmc2: Bus speed (slot 0) = 148500000Hz (slot req 150000000Hz, actual 148500000HZ div = 0)
[ 1683.776912] dwmmc_rockchip ff0f0000.dwmmc: Successfully tuned phase to 175
[ 1684.106344] PM: suspend entry (deep)
[ 1684.110379] PM: Syncing filesystems ... done.
[ 1684.143508] Freezing user space processes ... (elapsed 0.002 seconds) done.
[ 1684.153366] OOM killer disabled.
[ 1684.156994] Freezing remaining freezable tasks ... (elapsed 0.001 seconds) done.
[ 1684.166495] Suspending console(s) (use no_console_suspend to debug)
[ 1684.198018] Disabling non-boot CPUs ...
[ 1684.235090] Enabling non-boot CPUs ...
[ 1684.236478] CPU1 is up
[ 1684.237769] CPU2 is up
[ 1684.239076] CPU3 is up
[ 1684.343520] usb usb2: root hub lost power or was reset
[ 1684.356210] mmc_host mmc1: Bus speed (slot 0) = 400000Hz (slot req 400000Hz, actual 400000HZ div = 0)
[ 1684.413324] mmc1: queuing unknown CIS tuple 0x80 (2 bytes)
[ 1684.414826] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[ 1684.416322] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[ 1684.419015] mmc1: queuing unknown CIS tuple 0x80 (7 bytes)
[ 1684.448123] rockchip-dp ff970000.dp: Timeout of video streamclk ok
[ 1684.448126] rockchip-dp ff970000.dp: unable to config video
[ 1684.472188] mmc_host mmc1: Bus speed (slot 0) = 148500000Hz (slot req 150000000Hz, actual 148500000HZ div = 0)
[ 1684.599471] dwmmc_rockchip ff0d0000.dwmmc: Successfully tuned phase to 186
[ 1684.623764] elan_i2c 4-0015: invalid report id data (ff)
[ 1684.681407] usb 2-1: reset high-speed USB device number 2 using ehci-platform
[ 1684.949885] OOM killer enabled.
[ 1684.953404] Restarting tasks ... done.
[ 1684.962190] PM: suspend exit
```

robinde, 2018-11-08:
As mentioned in [Miouyouyou/RockMyy#3](https://github.com/Miouyouyou/RockMyy/issues/3):
1. I'm using openbox. However, symptoms are the same suspending from a tty, before x is started
2. I haven't had a succesful one so far
3.  I couldn't get any good dmesg logs, last thing I can see from `journalctl` is
```
alarm systemd[1]: Reached target Sleep.
alarm systemd[1]: Starting Suspend...
alarm systemd-sleep[613]: Suspending system...
alarm kernel: PM: suspend entry (deep)
```
4. I'm booting from microsd, haven't tried from the emmc, although I'm happy to test

I don't use xfce, but I'm happy to install it and test your workaround for testing

georgegeddes, 2019-02-15:
I have PrawnOS installed on my internal storage with my /home on the external microsd card and when I get this suspend issue, it always seems to be triggered by interacting with the internal storage. For example, I can wake from suspend and continue my Emacs session, editing files in my home directory. I can open a terminal and type `ls` and see my home directory, but if I run `ls /`, the CPU usage goes up to 100% and the ls output never appears. I can't figure out how to reproduce this now, but it happened to me several times yesterday. It might be related to leaving the system suspended for a long time, but I'm not sure. 

I don't know how to fix it, but maybe this information will help?

SolidHal, 2019-02-19:
@georgegeddes  Thanks for the information, let me know if you figure out how you can reproduce it. I believe I fixed the issue in develop on the newer kernel, but not wireless hasn't been stable so I haven't been able to push anything.

Whelle, 2019-02-22:
I can reproduce it in 100% cases.
I've got my own OS with kernel 4.20.10 and patches from PrawnOS.
C201 with 2GB RAM.
Steps to reproduce:
1) sudo -i (Login as root)
2) sleep 1; echo -n "mem" > /sys/power/state; dmesg -Hw
3) Wake it up (Press any key)
4) Somehow use mmc (eg. read directories using `ls`)
5) You will see in kernel log something like this:
```
...
PM: suspend exit
dwmmc_rockchip ff0c0000.dwmmc: Busy; trying anyway
mmc_host_mmc0: Timeout sending command (cmd 0x202000 arg 0x0 status 0x80202000)
...
print_req_error: I/O error, dev mmcblk2, sector 2101466
print_req_error: I/O error, dev mmcblk2, sector 2101467
print_req_error: I/O error, dev mmcblk2, sector 2101468
print_req_error: I/O error, dev mmcblk2, sector 2101469
...
```
After that my kernel crashes even without kernel panic, it just hangs up and I have to hard reboot it.

SolidHal, 2019-02-22:
Thanks for the report @Whelle. Would you mind trying out the patches in the 4.19-dev repo, specifically this one: https://github.com/SolidHal/PrawnOS/blob/4.19-dev/resources/BuildResources/patches-tested/kernel/0001-Revert-ec-changes-to-fix-wake-on-the-c201.patch

Whelle, 2019-02-22:
Patched. It doesn't help. I still get same errors and crash.

SolidHal, 2019-06-22:
Testing with suspend disabled in the kernel, if the xfce power management is set to only turn off the display and never lock the screen, the machine functions as expected when the lid is opened and closed. It is only when the xfce screen lock is enabled (light locker by default) that the black screen issue occurs. 

This is --Not-- related to errors like: 
```
[ 1063.008166] ------------[ cut here ]------------
[ 1063.013372] WARNING: CPU: 1 PID: 599 at drivers/gpu/drm/rockchip/rockchip_drm_vop.c:836 vop_crtc_enable_vblank+0x34/0x12c
[ 1063.025628] Modules linked in: ipv6 autofs4
[ 1063.030334] CPU: 1 PID: 599 Comm: xfwm4 Not tainted 4.19.15-gnu #1
[ 1063.037246] Hardware name: Rockchip (Device Tree)
[ 1063.042535] [<c0211758>] (unwind_backtrace) from [<c020c5bc>] (show_stack+0x20/0x24)
[ 1063.051215] [<c020c5bc>] (show_stack) from [<c09e4f54>] (dump_stack+0x88/0xa8)
[ 1063.059313] [<c09e4f54>] (dump_stack) from [<c0223b90>] (__warn+0xdc/0x10c)
[ 1063.067117] [<c0223b90>] (__warn) from [<c0223cac>] (warn_slowpath_null+0x4c/0x58)
[ 1063.075598] [<c0223cac>] (warn_slowpath_null) from [<c06295a4>] (vop_crtc_enable_vblank+0x34/0x12c)
[ 1063.085734] [<c06295a4>] (vop_crtc_enable_vblank) from [<c061f9f0>] (drm_vblank_enable+0x110/0x12c)
[ 1063.095868] [<c061f9f0>] (drm_vblank_enable) from [<c061faa0>] (drm_vblank_get+0x94/0xe8)
[ 1063.105028] [<c061faa0>] (drm_vblank_get) from [<c0620acc>] (drm_wait_vblank_ioctl+0x150/0x608)
[ 1063.114774] [<c0620acc>] (drm_wait_vblank_ioctl) from [<c0603620>] (drm_ioctl_kernel+0x8c/0xc4)
[ 1063.124517] [<c0603620>] (drm_ioctl_kernel) from [<c0603958>] (drm_ioctl+0x2d4/0x344)
[ 1063.133288] [<c0603958>] (drm_ioctl) from [<c0390a8c>] (vfs_ioctl+0x30/0x44)
[ 1063.141185] [<c0390a8c>] (vfs_ioctl) from [<c03916ec>] (do_vfs_ioctl+0x5b0/0x720)
[ 1063.149566] [<c03916ec>] (do_vfs_ioctl) from [<c03918a0>] (ksys_ioctl+0x44/0x6c)
[ 1063.157849] [<c03918a0>] (ksys_ioctl) from [<c03918e0>] (sys_ioctl+0x18/0x1c)
[ 1063.165842] [<c03918e0>] (sys_ioctl) from [<c0201000>] (ret_fast_syscall+0x0/0x4c)
[ 1063.174309] Exception stack(0xebe0dfa8 to 0xebe0dff0)
[ 1063.179966] dfa0:                   01cad620 01caca98 00000006 c010643a be9b071c 20000001
[ 1063.189111] dfc0: 01cad620 01caca98 c010643a 00000036 01caca98 b6f69968 01c4e128 00a00ff2
[ 1063.198261] dfe0: 004b3aa8 be9b06d4 00477ecd b6687e38
[ 1063.203912] ---[ end trace 954d202faa5ff59f ]
```

As I tested with the chromeos-4.19 kernel and PrawnOS filesystem that their team has been working on for the past few months and this issue doesn't occur but the above warning still get spammed to the console. 

Testing with suspend enabled, and screen locking disabled I suspect I'll get the same results. There is somethings specifically wrong with how turning on the display interacts with the tty switch from light locker, and no kernel messages are catching it as of yet.

SolidHal, 2019-06-24:
The light locker bug has existed for a while https://github.com/the-cavalry/light-locker/issues/114, and seems to actually be a issue with lightdm-gtk-greeter as if slick-greeter is used instead this issue doesn't occur

SolidHal, 2019-09-15:
Suspend is no longer broken in my tests, will adjust the power settings to take advantage of this.

riwale, 2019-09-15:
suspend works well, there is just one caveat:

if you close the lid and tell xfce to go to suspend, the screen will be locked, even though it's selected not to do so.
if you click suspend, THEN close the lid, the book is unlocked when you open up again.

SolidHal, 2019-09-15:
Yeah, the settings get a little weird. You have to sync up power settings and screensaver settings. 

Problem is, current version of xfce4-screensaver crashes when the lid is closed, suspend gets activated by lid close, and the lock screen is set to activate. 

I hope to have some reasonable default settings soon.

SolidHal, 2019-10-29:
Closing, suspend and screen locking are fully functional.

closed 2019-10-29 by SolidHal