SolidEva/PrawnOS #68: Install to internal reboot and boot to chromeos repair

Anthony-Sensors, 2019-01-09:
When I launch the script it start writing, then reboot(goes too fast to see cli output). Afterward chromeOS repair boot instead of PrawnOS.
I have libreboot installed and powerwashed the device just before running the script. release 0.5.

SolidHal, 2019-01-09:
Are you installing to the internal emmc or external media?

If target is the internal emmc:
Sounds like the image isn't getting written to the emmc correctly. 

After booting to the install usb edit the install script: 

`nano /InstallToInternal.sh`

and comment out or remove the reboot command at the bottom 

```
    e2fsck -p -f /dev/mmcblk2p2
    echo Rebooting... Please remove the usb drive once shutdown is complete
    #reboot
fi

echo Exiting
```

now run the install script

```
cd /
./InstallToInternal.sh
```

and post the output here. Easiest way is probably to upload a picture of the screen.

Anthony-Sensors, 2019-01-10:
#output stock
#cgpt show /dev/mmcblk2 
#output on cli:
WARNING: Primary GPT header is being ignored

#cgt show /dev/mmcblk2
#output to file:
       start        size    part  contents
           0           1          PMBR (Boot GUID: XXXX)
           1           1 IGNORED  Pri GPT header
    30777311          32          Sec GPT table
     8671232    22073344       1  Label: "STATE"
                                  Type: Linux data
                                  UUID: XXXX
       20480       32768       2  Label: "KERN-A"
                                  Type: ChromeOS kernel
                                  UUID: XXXX
                                  Attr: priority=1 tries=0 successful=1
     4476928     4194304       3  Label: "ROOT-A"
                                  Type: ChromeOS rootfs
                                  UUID: XXXX
       53248       32768       4  Label: "KERN-B"
                                  Type: ChromeOS kernel
                                  UUID: XXXX
                                  Attr: priority=2 tries=0 successful=1
      282624     4194304       5  Label: "ROOT-B"
                                  Type: ChromeOS rootfs
                                  UUID: XXXX
       16448           1       6  Label: "KERN-C"
                                  Type: ChromeOS kernel
                                  UUID: XXXX
                                  Attr: priority=0 tries=15 successful=0
       16449           1       7  Label: "ROOT-C"
                                  Type: ChromeOS rootfs
                                  UUID: XXXX
       86016       32768       8  Label: "OEM"
                                  Type: Linux data
                                  UUID: XXXX
       16450           1       9  Label: "reserved"
                                  Type: ChromeOS reserved
                                  UUID: XXXX
       16451           1      10  Label: "reserved"
                                  Type: ChromeOS reserved
                                  UUID: XXXX
          64       16384      11  Label: "RWFW"
                                  Type: ChromeOS firmware
                                  UUID: XXXX
      249856       32768      12  Label: "EFI-SYSTEM"
                                  Type: EFI System Partition
                                  UUID: XXXX
    30777343           1          Sec GPT header

#fdisk -l
#output on cli:
the primary GPT table is corrupt, but the backup appears OK, so that will be used.
GPT PMBR size mismatch (4095999 != 8028157) will be corrected by w(rite)

#fdisk -l
#output to file:
Disk /dev/mmcblk2: 14.7 GiB, 15758000128 bytes, 30777344 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: XXXX

Device            Start      End  Sectors  Size Type
/dev/mmcblk2p1  8671232 30744575 22073344 10.5G Microsoft basic data
/dev/mmcblk2p2    20480    53247    32768   16M ChromeOS kernel
/dev/mmcblk2p3  4476928  8671231  4194304    2G ChromeOS root fs
[... 163 lines cut ...]
                                  Type: ChromeOS rootfs
                                  UUID: XXXX
       16448           1       6  Label: "KERN-C"
                                  Type: ChromeOS kernel
                                  UUID: XXXX
                                  Attr: priority=0 tries=15 successful=0
       16449           1       7  Label: "ROOT-C"
                                  Type: ChromeOS rootfs
                                  UUID: XXXX
       86016       32768       8  Label: "OEM"
                                  Type: Linux data
                                  UUID: XXXX
       16450           1       9  Label: "reserved"
                                  Type: ChromeOS reserved
                                  UUID: XXXX
       16451           1      10  Label: "reserved"
                                  Type: ChromeOS reserved
                                  UUID: XXXX
          64       16384      11  Label: "RWFW"
                                  Type: ChromeOS firmware
                                  UUID: XXXX
      249856       32768      12  Label: "EFI-SYSTEM"
                                  Type: EFI System Partition
                                  UUID: XXXX
    30777343           1          Sec GPT header

SolidHal, 2019-01-10:
Thanks for all of the info. 

>  Disk /dev/mmcblk2: 14.7 GiB, 15758000128 bytes, 30777344 sectors

Ah! You have a different emmc! I knew there were at least two different ones out there but haven't seen yours yet. 

Since the c201s are picky about the positioning of partitions I had to hard code in the partition map. The file used to make the partitions is located in /InstallResources/mmc.partmap. You can open it with nano. 

set `last-lba` to `30777310`
this is so that the secondary header is not overwritten by the os. This is important because the c201 is hardcoded to ignore the primary header. 

then set the `size of /dev/mmcblk2p2` to `30691294`

run the installToInternal script again and let me know if that worked. If it did, I'll add some logic to detect the different emmcs and swap partmaps. 

I

Anthony-Sensors, 2019-01-10:
It successfully installed to emmc but InstallPackages.sh stopped with multiple error:
Cannot write: No space left on device.

SolidHal, 2019-01-10:
Are you sure you were booted from the emmc and not the usb when you ran InstallPackages.sh? 

It is very unlikely that you ran out of space on the emmc as even after installing all of the packages only 2.5gb are used.

Anthony-Sensors, 2019-01-10:
Yes I shutdown, remove the usb and launch the script.

Anthony-Sensors, 2019-01-10:
This will ERASE ALL DATA ON THE INTERNAL STORAGE (EMMC) and reboot when finished, do you want to continue? [Y/n]
Writing partition map
The primary GPT table is corrupt, but the backup appears OK, so that will be used.
Checking that no-one is using this disk right now ... OK

Disk /dev/mmcblk2: 14.7 GiB, 15758000128 bytes, 30777344 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: XXXX

Old situation:

Device         Start      End  Sectors  Size Type
/dev/mmcblk2p1 20480    86015    65536   32M ChromeOS kernel
/dev/mmcblk2p2 86016 30777309 30691294 14.6G Microsoft basic data

>>> Script header accepted.
>>> Script header accepted.
>>> Script header accepted.
>>> Script header accepted.
>>> Script header accepted.
>>> Script header accepted.
>>> Created a new GPT disklabel (GUID: XXXX).
/dev/mmcblk2p1: Created a new partition 1 of type 'ChromeOS kernel' and of size 32 MiB.
/dev/mmcblk2p2: Created a new partition 2 of type 'Microsoft basic data' and of size 14.6 GiB.
Partition #2 contains a ext4 signature.
/dev/mmcblk2p3: Done.

New situation:

Device         Start      End  Sectors  Size Type
/dev/mmcblk2p1 20480    86015    65536   32M ChromeOS kernel
/dev/mmcblk2p2 86016 30777309 30691294 14.6G Microsoft basic data

The partition table has been altered.
Calling ioctl() to re-read partition table.
/dev/mmcblk2: close device failed: Input/output error
Writing kernel partition
65536+0 records in
65536+0 records out
33554432 bytes (34 MB, 32 MiB) copied, 3.43145 s, 9.8 MB/s
Writing Filesystem, this will take about 4 minutes...
39+1 records in
39+1 records out
2059386368 bytes (2.1 GB, 1.9 GiB) copied, 172.039 s, 12.0 MB/s
Expanding Filesystem
/dev/mmcblk2p2: Inode 49568, i_blocks is 32, should be 30.  FIXED.
/dev/mmcblk2p2: Entry 'systemd-private-c48391dbbcb4486296c5c3309e1e0f9c-systemd-timesyncd.service-yh7K9E' in /var/tmp (49337) is a link to directory /var/tmp/systemd-private-20c322428b424eefb3833965a93c5d69-systemd-timesyncd.service-Coxm0t (49165).

/dev/mmcblk2p2: UNEXPECTED INCONSISTENCY; RUN fsck MANUALLY.
	(i.e., without -a or -p options)
resize2fs 1.43.4 (31-Jan-2017)
Please run 'e2fsck -f /dev/mmcblk2p2' first.

Rebooting... Please remove the usb drive once shutdown is complete
Exiting

Anthony-Sensors, 2019-01-10:
#./InstallRessources
+ DIR=/InstallResources
+ true
+ read -p 'install (X)fce4 or (L)xqt, if unsure choose (X)fce4: ' XL
install (X)fce4 or (L)xqt, if unsure choose (X)fce4: + case $XL in
+ DE=xfce
+ break
+ locale-gen
Generating locales (this might take a while)...
Generation complete.
+ apt install -y xorg acpi-support lightdm tasksel dpkg librsvg2-common xorg xserver-xorg-input-libinput alsa-utils anacron avahi-daemon eject iw libnss-mdns xdg-utils mousepad vlc dconf-tools sudo dtrx emacs25

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Reading package lists...
Building dependency tree...
Reading state information...
alsa-utils is already the newest version (1.1.3-1).
dpkg is already the newest version (1.18.25).
iw is already the newest version (4.9-0.1).
tasksel is already the newest version (3.39).
The following additional packages will be installed:
  acpi-support-base acpid adwaita-icon-theme at-spi2-core bind9-host
  cabextract console-setup console-setup-linux dconf-cli dconf-editor
  dconf-gsettings-backend dconf-service debugedit desktop-base
  emacs25-bin-common emacs25-common emacs25-el emacsen-common ethtool file
  fontconfig fontconfig-config fonts-dejavu-core fonts-droid-fallback
  fonts-freefont-ttf fonts-noto-mono gconf-service gconf2-common
  geoip-database ghostscript glib-networking glib-networking-common
  glib-networking-services gnome-accessibility-themes gnome-themes-standard
  gnome-themes-standard-data gsettings-desktop-schemas gsfonts
  gtk-update-icon-cache gtk2-engines-pixbuf hdparm hicolor-icon-theme
  imagemagick-6-common kbd keyboard-configuration liba52-0.7.4 libaa1 libaacs0
  libarchive13 libass5 libasyncns0 libatk-bridge2.0-0 libatk1.0-0
  libatk1.0-data libatspi2.0-0 libauthen-sasl-perl libavahi-client3
  libavahi-common-data libavahi-common3 libavahi-core7 libavc1394-0
  libavcodec57 libavformat57 libavutil55 libbasicusageenvironment1 libbdplus0
  libbind9-140 libbluray1 libcaca0 libcairo-gobject2 libcairo2 libcddb2
  libchromaprint1 libcolord2 libcroco3 libcups2 libcupsfilters1 libcupsimage2
  libdaemon0 libdatrie1 libdbus-glib-1-2 libdc1394-22 libdca0 libdconf1
  libdns162 libdouble-conversion1 libdrm-amdgpu1 libdrm-freedreno1
  libdrm-nouveau2 libdrm-radeon1 libdrm2 libdvbpsi10 libdvdnav4 libdvdread4
  libebml4v5 libegl1-mesa libencode-locale-perl libepoxy0 libfaad2
  libfftw3-double3 libfile-basedir-perl libfile-desktopentry-perl
  libfile-listing-perl libfile-mimeinfo-perl libflac8 libfont-afm-perl
  libfontconfig1 libfontenc1 libfreetype6 libfribidi0 libgail-common libgail18
  libgbm1 libgconf-2-4 libgd3 libgdk-pixbuf2.0-0 libgdk-pixbuf2.0-common
  libgeoip1 libgif7 libgl1-mesa-dri libgl1-mesa-glx libglapi-mesa
  libgles2-mesa libglew2.0 libglu1-mesa libgme0 libgpm2 libgraphite2-3
  libgroupsock8 libgs9 libgs9-common libgsm1 libgtk-3-0 libgtk-3-bin
  libgtk-3-common libgtk2.0-0 libgtk2.0-bin libgtk2.0-common
  libgtksourceview2.0-0 libgtksourceview2.0-common libharfbuzz0b
  libhtml-form-perl libhtml-format-perl libhtml-parser-perl
  libhtml-tagset-perl libhtml-tree-perl libhttp-cookies-perl
  libhttp-daemon-perl libhttp-date-perl libhttp-message-perl
  libhttp-negotiate-perl libice6 libijs-0.35 libio-html-perl
  libio-socket-ssl-perl libipc-system-simple-perl libisc160 libisccc140
  libisccfg140 libjbig0 libjbig2dec0 libjpeg62-turbo libjson-glib-1.0-0
  libjson-glib-1.0-common libkate1 liblcms2-2 libldb1 liblightdm-gobject-1-0
  liblirc-client0 liblivemedia57 libllvm3.9 liblockfile-bin liblockfile1
  liblqr-1-0 libltdl7 liblua5.2-0 liblwp-mediatypes-perl
  liblwp-protocol-https-perl liblwres141 liblzo2-2 libm17n-0 libmad0
  libmagic-mgc libmagic1 libmagickcore-6.q16-3 libmagickwand-6.q16-3
  libmailtools-perl libmatroska6v5 libmicrodns0 libmp3lame0 libmpcdec6
  libmpeg2-4 libmpg123-0 libmspack0 libmtp-common libmtp-runtime libmtp9
  libnet-dbus-perl libnet-http-perl libnet-smtp-ssl-perl libnet-ssleay-perl
  libnfs8 libnotify4 libnspr4 libnss3 libogg0 libopenjp2-7 libopenmpt-modplug1
  libopenmpt0 libopus0 libotf0 libpam-systemd libpango-1.0-0
  libpangocairo-1.0-0 libpangoft2-1.0-0 libpaper-utils libpaper1 libpciaccess0
  libpcre16-3 libpixman-1-0 libpng16-16 libpolkit-agent-1-0
[... 2857 lines cut ...]
Preparing to unpack .../056-libglibmm-2.4-1v5_2.50.0-1_armhf.deb ...
Unpacking libglibmm-2.4-1v5:armhf (2.50.0-1) ...
dpkg: error processing archive /tmp/apt-dpkg-install-TRGDhp/056-libglibmm-2.4-1v5_2.50.0-1_armhf.deb (--unpack):
 cannot copy extracted data for './usr/lib/arm-linux-gnueabihf/libgiomm-2.4.so.1.3.0' to '/usr/lib/arm-linux-gnueabihf/libgiomm-2.4.so.1.3.0.dpkg-new': failed to write (No space left on device)
dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
Selecting previously unselected package libatkmm-1.6-1v5:armhf.
Preparing to unpack .../057-libatkmm-1.6-1v5_2.24.2-2_armhf.deb ...
Unpacking libatkmm-1.6-1v5:armhf (2.24.2-2) ...
dpkg: error processing archive /tmp/apt-dpkg-install-TRGDhp/057-libatkmm-1.6-1v5_2.24.2-2_armhf.deb (--unpack):
 cannot copy extracted data for './usr/lib/arm-linux-gnueabihf/libatkmm-1.6.so.1.1.0' to '/usr/lib/arm-linux-gnueabihf/libatkmm-1.6.so.1.1.0.dpkg-new': failed to write (No space left on device)
Selecting previously unselected package libburn4:armhf.
Preparing to unpack .../058-libburn4_1.4.6-1_armhf.deb ...
Unpacking libburn4:armhf (1.4.6-1) ...
dpkg: error processing archive /tmp/apt-dpkg-install-TRGDhp/058-libburn4_1.4.6-1_armhf.deb (--unpack):
[... cut ...]
dpkg: error processing archive /tmp/apt-dpkg-install-TRGDhp/059-libcairomm-1.0-1v5_1.12.0-1+b1_armhf.deb (--unpack):
 cannot copy extracted data for './usr/lib/arm-linux-gnueabihf/libcairomm-1.0.so.1.4.0' to '/usr/lib/arm-linux-gnueabihf/libcairomm-1.0.so.1.4.0.dpkg-new': failed to write (No space left on device)
tar: ./symbols: Cannot write: No space left on device
tar: Exiting with failure status due to previous errors
dpkg-deb: error: subprocess tar returned error exit status 2
dpkg: error processing archive /tmp/apt-dpkg-install-TRGDhp/060-libcanberra0_0.3

SolidHal, 2019-01-10:
Seems e2fsck still didn't like something, so the filesystem doesn't get expanded by resize2fs

> /dev/mmcblk2p2: UNEXPECTED INCONSISTENCY; RUN fsck MANUALLY.
> (i.e., without -a or -p options)
> resize2fs 1.43.4 (31-Jan-2017)
> Please run 'e2fsck -f /dev/mmcblk2p2' first.

And thats the reason I moved away from using resize2fs, and instead now make a fresh filesystem and rsync over the OS.

I just posted a new release which has the installation script changes, which removes the need for resize2fs so give that a try. Make sure you modify the mmc.partmap again.

Anthony-Sensors, 2019-01-11:
There was an error umount /dev/mmcblk2: not mount
running fsck 
/dev/mmcblk2p2 is mounted 
e2fsck: Cannot continue, aborting. 

So I: "umount /dev/mmcblk2p2" instead and did: "e2fsck -p -f /dev/mmcblk2p2" as in the script

Anthony-Sensors, 2019-01-11:
Install was successful, with no other error other then the mounting error described above. Just booted to xfce, going to test further. :+1:

SolidHal, 2019-01-11:
The next release will include the support for your emmc, so no need to edit the mmc.partmap. Closing this issue. Feel free to reopen if need be.

closed 2019-01-11 by SolidHal