SolidEva/PrawnOS #73: Current kernel is not an LTS release.

Anthony-Sensors, 2019-01-11:
Switching to Linux-Libre LTS Kernel like 4.19 would offer longer support.

SolidHal, 2019-01-14:
Very true, however 4.19 adds multiple bugs including completely breaking sound for some reason, and fixes no outstanding bugs like the suspend issue. It is certainly a good goal to build off of a LTS, however I'd like to get one "fully functional" release before jumping to a new kernel. 

Additionally, unlike the 4.17.2, 4.17.19 will always be available from linux-libre since it was the last 4.17 release. So, there won't be the issue of the source disappearing like last time.

All of that said, I do have a 4.19 dev branch so pull requests are always welcome there.

SolidHal, 2019-03-06:
Closing this, as switching to a LTS kernel will be tracked separately.

closed 2019-03-06 by SolidHal