SolidEva/PrawnOS #78: keyboard does not behave properly on certain layout

Anthony-Sensors, 2019-02-17:
English (US) keyboard seem to work as intended, but Keyboard layout French Canada (Canada multilingual)  backspace key does not work. Going to test Spanish (eliminate dead keys) as well.  Also I would like to change search key to caps lock.
"sudo dpkg-reconfigure keyboard-configuration" immediately return to prompt. 
What is the best way to reconfigure keyboard layout I understand

Anthony-Sensors, 2019-02-17:
[image]

SolidHal, 2019-02-19:
Good question! Are you able to choose the proper keyboard layout after running ./InstallPackages? 
I'm not very familiar with how keyboard layouts are handled so I'll have to spend some time investigating.

SolidHal, 2019-02-21:
@Anthony-Sensors does running `dpkg-reconfigure console-setup` let you choose the proper layout and fix your issue?

Anthony-Sensors, 2019-02-21:
Thank you for the command. 
 I was using utf8 encoding which should have work as it support french english and spanish. I selected [ISO 8859-1](https://en.wikipedia.org/wiki/ISO/IEC_8859-1) (8bit latin text) I can do the back space and all key seem to work for now. But there is no way to write with the accent and special character from french or spanish ( à 	â 	ç 	æ 	é 	è 	ê 	ë 	ï 	î 	ô 	œ 	ù 	û 	ü 	ÿ
À 	Â 	Ç 	Æ 	É 	È 	Ê 	Ë 	Ï 	Î 	Ô 	Œ 	Ù 	Û 	Ü 	Ÿ and á é í ñ ó ú ü ¿ ¡) Does that have anything to do with locale. I'm using trisquel with mate on my other thinkpad laptop and have none of those issue out of the box. My OS are always in US - english but I also need to type in french and spanish from time to time. I can change the layout in XFCE: applications -> Setting -> keyboard. then keyboard model is generic 105-key(intl) PC( Should I select Asus laptop?) then I click on +Add and add french(canada) multilingual and spanish, as I use all 3 of them. But for some reason I cannot type the accent.

Anthony-Sensors, 2019-02-21:
I added locale and everything seems to work now.

SolidHal, 2019-02-22:
Reopening to address in the installer to make this an easier process, and at least to write up the fix in the README in the mean time.

Anthony-Sensors, 2019-02-23:
Update: Now it's US-en that has keys issue... backspace and and arrow key return wrong keys.
Spanish and Canadian keyboard works fine.

Anthony-Sensors, 2019-02-23:
Also now I get some warning when I plug usb key: " FAT-fs (sda1): utf8 is not a recommanded IO charset for FAT filesystems, filesystem will be case sensitive!"

SolidHal, 2019-02-27:
@Anthony-Sensors  did the keyboard layouts used to work fine on the 4.17.2 images like release version 5 and older? If it did, I probably broke it when I streamlined the installation process in the newer releases.

Anthony-Sensors, 2019-02-28:
I haven't test the keyboard with the 4.17.2 image. I also haven't test if the same issue appear with lxqt as I'm adding keyboard layout through xfce gui.

SolidHal, 2019-06-24:
If anyone has a system with a non enUS keyboard layout and can test this, or suggest a test that I can do on a enUS layout I would appreciate it.

SolidHal, 2019-09-15:
Does this issue still persist in current versions? Many config changes have occured since this was last tested.

Anthony-Sensors, 2019-09-15:
I'm sorry I was away for so long I won't have the time to test the new release yet I will get back to you in the next couple of days. In the meantime you can try testing it with your us keyboard by adding language in XFCE  and check if all the keys work try adding spanish and canadian french and see if all the key in keyboard works when you switch keyboard language. 

I'll test later if the issue is resolved.

Anthony-Sensors, 2020-01-12:
This issue is fixed. Tested with release 1.0.

closed 2020-01-12 by Anthony-Sensors

SolidHal, 2020-01-12:
@Anthony-Sensors Great to hear, thanks for testing it out!