SolidEva/PrawnOS #80: Uncomplicated firewall crash on start

Anthony-Sensors, 2019-02-17:
sudo efw enable output:

ERROR: problem running ufw-init
modprobe: FATAL: Module nf_conntrack_ftp not found in directory /lib/modules/4.17.19-gnu
modprobe: FATAL: Module nf_nat_ftp not found in directory /lib/modules/4.17.19-gnu
modprobe: FATAL: Module nf_conntrack_netbios_ns not found in directory /lib/modules/4.17.19-gnu
iptables-restore: line 77 failed
iptables-restore: line 30 failed
iptables-restore: line 11 failed

Problem loading ipv6 (skipping)
Problem running '/etc/ufw/before.rules'
Problem running '/etc/ufw/after.rules'
Problem running '/etc/ufw/user.rules'

SolidHal, 2019-02-19:
Looks like we'd need to enable `CONFIG_NF_CONNTRACK_FTP`, `CONFIG_NF_NAT_FTP`, `CONFIG_NF_CONNTRACK_NETBIOS_NS` in the kernel config. 

`CONFIG_NF_CONNTRACK_FTP` is marked experimental, but should be fine to enable.

SolidHal, 2019-06-06:
Like the Iptables issue, I'm not very familiar with efw. If I provide you a kernel with those modules enabled, could you test it?

SolidHal, 2019-09-15:
All of the kernel modules are now enabled in the current released, closing.

closed 2019-09-15 by SolidHal