SolidEva/PrawnOS #81: Iptables and ip6tables multiples issues

Anthony-Sensors, 2019-02-18:
Modprobe return FATAL: ip6tables not found in direcotry /lib/modules/4.17.19-gnu when trying to add a rule to ip6tables. 

I'm having issue using certain command with iptables such as logging which makes it hard to troubleshoot.
I'm also having slow down issue with apt when using iptables following rules:
 `*filter

#Loopback needed in linux
-A INPUT -i lo -j ACCEPT
-A OUTPUT -o lo -j ACCEPT

# web http udp
-A INPUT -i wlan0 -p tcp -m state --state ESTABLISHED,RELATED --sport 80 -j ACCEPT
-A INPUT -i wlan0 -p tcp -m state --state ESTABLISHED,RELATED --sport 443 -j ACCEPT 

-A OUTPUT -o wlan0 -p tcp -m tcp --dport 80 -j ACCEPT
-A OUTPUT -o wlan0 -p tcp -m tcp --dport 443 -j ACCEPT

#Allow router DNS
-A INPUT -i wlan0 -s 192.168.10.1 -p udp --sport 53 -m state --state ESTABLISHED,RELATED -j ACCEPT
-A OUTPUT -o wlan0 -d 192.168.10.1 -p udp --dport 53 -m udp -j ACCEPT

#allow time fetch by system
-A INPUT -i wlan0 -p udp -m state --state ESTABLISHED,RELATED --dport 123 -j ACCEPT
-A OUTPUT -o wlan0 -p udp -m udp --sport 123 -j ACCEPT

#printing only needed if wifi printer
#-A INPUT -p udp -m udp --dport 631 -j ACCEPT
#-A INPUT -p tcp -m tcp --dport 631 -j ACCEPT
#-A OUTPUT -p udp -m udp --sport 631 -j ACCEPT
#-A OUTPUT -p tcp -m tcp --sport 631 -j ACCEPT

# email
# IMAP
-A INPUT -i wlan0 -p tcp -m state --state ESTABLISHED,RELATED --sport 993 -j ACCEPT
-A OUTPUT -o wlan0 -p tcp -m tcp --dport 993 -j ACCEPT
# POP3
-A INPUT -i wlan0 -p tcp -m state --state ESTABLISHED,RELATED --sport 995 -j ACCEPT
-A OUTPUT -o wlan0 -p tcp -m tcp --dport 995 -j ACCEPT
# SMTP
-A INPUT -i wlan0 -p tcp -m state --state ESTABLISHED,RELATED --sport 465 -j ACCEPT
-A OUTPUT -o wlan0 -p tcp -m tcp --dport 465 -j ACCEPT

# SSH need both input and output? need to test if input is necessary disable for now
# Input
#-A INPUT -i wlan0 -p tcp -m state --state NEW,ESTABLISHED --dport 22 -j ACCEPT
#-A OUTPUT -o wlan0 -p tcp -m state --state ESTABLISHED --sport 22 -j ACCEPT
# Output
-A OUTPUT -o wlan0 -p tcp -m state --state NEW,ESTABLISHED --dport 22 -j ACCEPT
-A INPUT -i wlan0 -p tcp -m state --state ESTABLISHED --sport 22 -j ACCEPT

#DHCP
-A INPUT -i wlan0 -p udp -m state --state ESTABLISHED,RELATED --sport 67:68 -j ACCEPT
-A OUTPUT -o wlan0 -p udp -m udp --dport 67:68 -j ACCEPT

#Drop everything else
-A INPUT -j DROP
-A FORWARD -j DROP
-A OUTPUT -j DROP
`

SolidHal, 2019-02-19:
I disabled ipv6 in the kernel as I believed it was unused, so if you are comfortable I would try re-enabling it and see if that fixes this problem.

Anthony-Sensors, 2019-02-21:
Yes but none the less I having slowdown(hang for a while then proceed) with apt with this set-up and cannot do logging, also for some reason "REJECT" arguments did not worked but "DROP" did.
Although maybe I did something wrong as I'm not very experienced working with iptables directly.

Anthony-Sensors, 2019-02-23:
command "sudo iptables -N LOGGING" works
command "sudo iptables -A INPUT -j LOGGING" works
command "sudo iptables -A LOGGING -m limit --limit 2/min -j LOG --log-prefix "IPTables-Dropped: " --log-level 4" return "iptables: No chain/target/match by that name.

SolidHal, 2019-06-06:
I re-enabled ipv6 in recent builds, but I think this will require `IP6_NF_IPTABLES` to be enabled as well to use ip6tables. 

I'm not familiar with iptables, nor with what you are trying to do with them so I could use your help testing any fixes @Anthony-Sensors 
I can provide a kernel with `IP6_NF_IPTABLES` enabled if that would help.

SolidHal, 2019-09-15:
I think this is resolved now with the current kernel config, feel free to reopen if that is not the case.

closed 2019-09-15 by SolidHal