SolidEva/PrawnOS #84: Freeze when computer idle for too long

Anthony-Sensors, 2019-02-23:
I disable wifi, closed the backlight of the screen to maximum and left the computer idle unplugged for a whole night. When I got back to the computer in the morning prawn os was not responding when I tried to open the backlight.

Anthony-Sensors, 2019-02-23:
[LastFewKernelLog.txt](https://github.com/SolidHal/PrawnOS/files/2897028/LastFewKernelLog.txt)
Here is the kernel log just before I decide to forced shutdown the PrawnOS.

SolidHal, 2019-02-25:
@Anthony-Sensors Did you unplug you wifi dongle or just disable wifi?

Anthony-Sensors, 2019-02-25:
Only disable.

SolidHal, 2019-05-23:
I would be curious to know if there was any filesystem corruption that could have caused this, and if the recent fixes in 7ad3f7462b0097439182a7c36998c7162d5540eb fix this. I'll try to recreate.

SolidHal, 2019-06-06:
I haven't been able to recreate this on the current release, version 9. Likely do to better filesystem corruption management. Closing.

closed 2019-06-06 by SolidHal