SolidEva/PrawnOS #87: Add a debian buster based release.

Anthony-Sensors, 2019-03-08:
The full-freeze is schedule for 12 of march. 
Should /etc/apt/preferences and /etc/apt/sources.list be changed to buster?
"deb http://deb.debian.org/debian/ buster main"

SolidHal, 2019-05-23:
I would like to do a proper stable release on stretch before migrating to buster. Since the freeze is still recent, and no full buster release has happened I think we can hold off on this for now. I'll make an issue listing the things I would like to achieve for a "stable" release.

JeremyRand, 2019-05-29:
@Anthony-Sensors could you try testing https://github.com/SolidHal/PrawnOS/pull/96 ?  Would be good to get feedback on it.

SolidHal, 2019-06-06:
> I would like to do a proper stable release on stretch before migrating to buster. Since the freeze is still recent, and no full buster release has happened I think we can hold off on this for now. I'll make an issue listing the things I would like to achieve for a "stable" release.

I have since rethought my stance on this, in testing buster seems quite stable. Since we already get a couple packages from buster, much of the encryption testing has been on buster, and more packages are available in general on buster switching to buster makes sense to me.

SolidHal, 2019-06-24:
With release 10, buster is now used instead of stretch. Closing.

closed 2019-06-24 by SolidHal