SolidEva/PrawnOS #92: Rockchip crypto driver has some fixes in Linux 5.1 / 5.2

JeremyRand, 2019-05-12:
Regarding https://github.com/SolidHal/PrawnOS/pull/91 , the Rockchip crypto driver saw 2 fixes in Linux 5.1:

* https://github.com/torvalds/linux/commit/4359669a087633132203c52d67dd8c31e09e7b2e
* https://github.com/torvalds/linux/commit/c1c214adcb56d36433480c8fedf772498e7e539c

There's also a fix in master branch that I'm guessing is likely to end up in Linux 5.2:

* https://github.com/torvalds/linux/commit/f0cfd57b43fec65761ca61d3892b983a71515f23

Might be useful to test PrawnOS with Linux-Libre 5.1 and Linux-Libre 5.2 to see if we can re-enable `CONFIG_CRYPTO_DEV_ROCKCHIP` without triggering filesystem corruption in LUKS.

SolidHal, 2019-05-22:
testing may be easier if we backport the fixes to 4.17 or  4.19, since new kernel releases often break armhf things for no good reason due to the smaller user base. 
I'll add this to my todo list, as I think I've finished my work with the usb/wifi issues. If you happen to get a chance to test it out first though, let me know!

ifbizo, 2019-05-24:
> I'll add this to my todo list, as I think I've finished my work with the usb/wifi issues.
Do you have link or description about usb/wifi issue resolution? I've been trying to debug kernel crashes as well. I'll be happy to test! I haven't done the camera solder change yet and have an AR9271.

SolidHal, 2019-06-20:
Once there's a 5.2 linux libre release, I'll begin testing on it. For now, It looks like the chromeos team began working on 4.19 support for all of the veyron chromebooks ~4 months ago. Looks like the backported those patches without too much work https://chromium.googlesource.com/chromiumos/third_party/kernel/+log/refs/heads/chromeos-4.19/drivers/crypto/rockchip

SolidHal, 2019-09-15:
Now that we build the 5.2 kernel this can be closed

closed 2019-09-15 by SolidHal