SolidEva/PrawnOS #93: rndis_host kernel module for USB tethering

tslilc, 2019-05-13:
This would be useful to have by default, as it would allow users to make use of their cellphones to connect to wireless networks should they not have an appropriate USB wireless dongle available.

SolidHal, 2019-05-22:
Have you tested a build with it enabled, and did it work? If not, I can build you an image to test with as I don't have a phone that can do this.

tslilc, 2019-05-27:
@SolidHal thanks, i haven’t tested because i’m presently travelling and the only computer i have access to is the c201 — i don’t fancy building the kernel on it. If you wouldn’t mind building such a kernel + module then i’d be happy to test it, thanks!

Given that i haven’t had opportunity to test this, it may also require rndis_wlan and usbnet modules, i don’t know.

SolidHal, 2019-05-31:
> If you wouldn’t mind building such a kernel + module then i’d be happy to test it, thanks!

I can do that, I'll let you know when I have one for you.

SolidHal, 2019-06-05:
Heres a kernel with the rndis drivers built in, so no modules needed. www.solidhal.net/Shared/vmlinux-rndis.kpart

rndis_wlan, rndis_host, and usbnet drivers are all enabled and co-dependent.

tslilc, 2019-06-17:
Thanks for this kernel, i was able to verify that the usb tethered device showed up (under `ip link`) but `dhclient` was unable to obtain a lease. This could be due to the device or the computer, i currently don't have a way to test, and so i'll have to get back to you in about two weeks -- sorry!

The testing was somewhat complicated by the fact that the kernel expected modules built differently somehow -- the path it wanted was `/lib/modules/4.17.19-gnurndis-test` or something but wouldn't load the modules compiled for/against the standard kernel even when forced. It my mind anyway it's not impossible that `dhclient` just didn't work at all on the test kernel, but as i couldn't load the `ath` module i wasn't able to verify this.

SolidHal, 2019-06-22:
> Thanks for this kernel, i was able to verify that the usb tethered device showed up (under `ip link`) but `dhclient` was unable to obtain a lease. This could be due to the device or the computer, i currently don't have a way to test, and so i'll have to get back to you in about two weeks -- sorry!
> 
Huh, weird that dhclient couldn't get a lease. No rush! Let me know what you find when you can. 

> The testing was somewhat complicated by the fact that the kernel expected modules built differently somehow -- the path it wanted was /lib/modules/4.17.19-gnurndis-test or something but wouldn't load the modules compiled for/against the standard kernel even when forced. 

That is very strange. The kernel shouldn't have expected any modules in the filsystem, as I chose in the config to bake them into the kernel image.

tslilc, 2019-07-24:
I have managed to test this (with my own kernel build) and adding those three modules worked perfectly! I hope to see this added to PrawnOS soon.

SolidHal, 2019-08-02:
Sorry for the delay, have been traveling. The required modules are now enabled.

commit 72559985 (2019-08-02): Add support for usb tethering
closes #93
closed 2019-08-02 by SolidHal