SolidEva/PrawnOS #95: sporadic touchpad issues

tehbra1n, 2019-05-24:
I'm on a fresh install of 0.9, but this occurs on previous versions as well.    The touchpad will routinely become unresponsive, the mouse will move without any input, or the mouse will move slowly compared to the input from the touchpad.

This may or may not be related to the USB WiFi dongles I'm using but there seems to be some correlation.  I have a Thinkpenguin TPE-N150USB Wireless N USB Adapter which uses the AR9271 chipset.    Plugging this in will variously cause the touchpad issue, or cause the laptop to stop responding altogether.

I have a 2nd, generic USB WiFi adapter using the AR9271 chipset that seems to work reliably.  I'm not sure what the difference is between the two.

When the touchpad is unresponsive, it usually disappears from the output of xinput.

Syslog is being spammed with `elan_i2c 4-0015: invalid report id data (ff)` but otherwise I'm not seeing anything interesting.

I'm happy to grab any debugging info that might help.  Otherwise PrawnOS rocks!

SolidHal, 2019-05-31:
Interesting that the two adapters cause different issues. Could you share some information on the generic adapter you have? 

I wonder if the i2c issues I reported in #39 or the ones in #85 could be of any use for debugging this. 

This will be next on my plate for debugging. 

>  Otherwise PrawnOS rocks!

Thanks hahaha!

tehbra1n, 2019-05-31:
@SolidHal  I'm not exactly sure what info will be helpful, so I grabbed output of xinput, lsusb, lsusb -v, and dmesg.

[nousb.txt](https://github.com/SolidHal/PrawnOS/files/3242758/nousb.txt)

This file was from a cold boot with no WiFi dongle connected.  I didn't have any touchpad support.

[nousb2.txt](https://github.com/SolidHal/PrawnOS/files/3242762/nousb2.txt)

This is after a reboot, now the touchpad worked.

[generic.txt](https://github.com/SolidHal/PrawnOS/files/3242763/generic.txt)

This is with the generic AR9271 USB adapter plugged in.  Touchpad was sporadic and took on a life of its own a few times.

If you let me know any specific steps to take or info to grab, I'm happy to do so.

SolidHal, 2019-06-24:
Sorry! I had a reply typed to this a while ago but forgot to hit "comment" before switching tabs....

This sounds like a touchpad hardware issue. I have 3 of c201s and haven't experienced anything like the touchpad doing whatever it wanted in any version of PrawnOS or other. From my experience in laptop repair, it sound like the touchpad may have been liquid damaged, or is just broken. If you are lucky, it may just be a loose cable. 
Its not too hard to open these up and test out the loose cable theory.

Step 7 of this https://www.ifixit.com/Guide/Asus+Chromebook+C201+Keyboard+Replacement/60774 guide shows the white flex cable that connects the touchpad to the motherboard. Flip up the small black pieces of plastic on the cable connectors and you'll be able to slide the white cable out by the blue plastic "handle" and slide it back in. Be gentle with it, they are fragile and the pins can de-laminate. 

Of course, to better confirm this is a a hardware issue you could try the archlinux or devsus images for the laptop as well and see if this happens.

I hope this helps!

tehbra1n, 2019-06-27:
@SolidHal  it looks like you were correct.  The ribbon cable basically fell out of the connector as I pulled apart the keyboard from the chassis.  Its probably been sketchy since when I first flashed libreboot.  Thanks for the suggestion!

SolidHal, 2019-06-27:
@tehbra1n I'm glad it worked! No problem!

closed 2019-06-27 by SolidHal