SolidEva/PrawnOS #98: Kernel 4.19 support

SolidHal, 2019-06-03:
Issue: wifi hangs after very short while on kernel 4.19
This is likely the cause: https://www.spinics.net/lists/linux-usb/msg180875.html
watch for patch to get backported into 4.19

ifbizo, 2019-06-08:
Thanks for posting this! It resolves my issues with AR9271. I've been trying to debug this issue for a while now and was totally stuck.
How did you find this -- do you actively follow that list?

SolidHal, 2019-06-09:
No problem! Yeah, I've been debugging USB/ath9k issues for the better part of a year so I've been subscribed to the list. 

If you could, can you share what version of 4.19 you tested this with? Did you also have the "revert ep3 and ep4" patch as well, or just this patch?

ifbizo, 2019-06-09:
> If you could, can you share what version of 4.19 you tested this with? Did you also have the "revert ep3 and ep4" patch as well, or just this patch?

It was actually 4.17.19 that was still giving me trouble. Yes, I did also have the revert ep3/ep4 patch.
I am going to try ~5.2 again. I had good success before with everything except the wifi, but kept going back to 4.17.19 to isolate all the moving parts.

SolidHal, 2019-06-10:
> It was actually 4.17.19 that was still giving me trouble.

Interesting. What brand/make/model ar9271 adapter have you been testing with?

Just to confirm, you used this https://www.spinics.net/lists/linux-usb/msg180875.html patch on 4.17.19? 
That wouldn't apply properly as the commit it is reverting, https://github.com/torvalds/linux/commit/ed194d1367698a0872a2b75bbe06b3932ce9df3a, was not ever applied to 4.17.19.

Are you sure you weren't testing with a 4.19.x or 5.x kernel?

SolidHal, 2019-06-11:
I tested 4.19.49 libre with https://github.com/torvalds/linux/commit/ed194d1367698a0872a2b75bbe06b3932ce9df3a applied and found it to fix the freezing/crashing wireless issues I have experienced on 4.19 before.

I ran wireless and usage stress tests across a handful of new installs, including two overnight downloads, and haven't experienced a crash yet. 

Some tweaking of the sleep/power settings may be required still.

ifbizo, 2019-06-14:
I spoke too soon above, I am still having crashes.

I did a clean build using the latest PrawnOS as of today https://github.com/SolidHal/PrawnOS/commit/5999c4377671e07ea153a2c64b8d52dfbbae180a, so 4.19.49 with no other custom patches or config changes. It appeared to be working and passed my usual tests of basic ssh and downloading a 200 MB file. But after a "break" of 5 mins (still connected to AP), it crashed when doing a simple ssh connection. ath9k and irq was in backtrace, but since I didn't modify the config I didn't get a full dump. It's very inconsistent -- the past few reboot attempts I can't even fully connect to the network without it crashing immediately.

My AR9271 is this one https://wikidevi.com/wiki/ThinkPenguin_TPE-N150USB.

SolidHal, 2019-06-24:
@ifbizo, Was this with xfce or lxde network manager, or did you use a command line tool to setup the connection? 

Could you try with the most recent release and report in #77? I made a bunch of changes to usb power management in an attempt to isolate this issue, but I can't recreate it so testing is... difficult hahah. I put my test process in that issue, so maybe I'm not doing something you are?
If it crashes again, could you supply a pic of the screen/dump?

Thanks for your help!

ifbizo, 2019-07-02:
@SolidHal 
I was using command line after configuring [interfaces(5)](https://manpages.debian.org/stretch/ifupdown/interfaces.5.en.html).

Unfortunately I won't have access to my C201 for some time so I'm useless for now. The picture of the traceback in https://github.com/SolidHal/PrawnOS/issues/77#issuecomment-504776134 looks very similar to what I experienced (irq*, tasklet*, ath9k*, skb_release*, kfree_skb_list).

I thank you for your help in tracking down this issue without being able to test! It was difficult enough for me to debug and make changes even when I can reproduce it!

SolidHal, 2019-07-12:
Going to move this conversation to #102 since I'm going to move on from kernel 4.19 to 5.2.

closed 2019-07-12 by SolidHal