SolidEva/PrawnOS #99: Panfrost support

bobbybee, 2019-06-08:
The kernel driver for Panfrost was merged into kernel version 5.2. This kernel driver is stable (in heavy daily use on a RK3399 device) and compatible with the Mali T760 in the RK3288 (featured in the C201).

In userspace, Panfrost has been in mesa but disabled. I'm not sure when that reaches users. Regardless, the T760 in PrawnOS devices is supported; we have continous integration testing for both user and kernelspace running frequently, including on a RK3288 board.

For PrawnOS, the kernel needs to be upgraded to 5.2 (it's likely too difficult to backport the kernel driver), with Panfrost enabled in the `.config`. Mesa should also be packaged appropriately, but a compatible kernel is a prerequisite.

Benefits of Panfrost to PrawnOS include: accelerated desktops (Weston and sway work near-flawlessly; bringup for GNOME is WIP), accelerated video playback (mpv and Kodi work nearly flawlessly), and 3D games (SuperTuxKart, Neverball, and reportedly a Quake3 port?).

I expect Panfrost to be ready for adventurous RK3288 end-users in a few months, so opening this issue for PrawnOS to prepare accordingly.

_Disclaimer: I am the lead developer of Panfrost. I'm writing this statement on my free time; statements expressed are not those of my employer._

SolidHal, 2019-06-10:
Hey @bobbybee! I'm excited to hear from you again, and really happy to see Panfrost has come so far. I was watching it for quite a while but lost track a few months ago, so this is really a pleasant surprise.

I'm working on getting to kernel 4.19 lts, which I may have done today if everything goes smoothly, and after that I'll figure out what is needed for 5.2 to function reliably. 

> In userspace, Panfrost has been in mesa but disabled. I'm not sure when that reaches users. 
Are there specific build flags for Mesa to enable Panfrost support, or something else? If so, I'll package Mesa for PrawnOS until the Debian repos get updated. 

Thanks again for your and your teams work towards making rk3xxx devices fully supported by libre software! Accelerated graphics will be an awesome addition.

bobbybee, 2019-06-10:
> I'm working on getting to kernel 4.19 lts, which I may have done today
> if everything goes smoothly, and after that I'll figure out what is
> needed for 5.2 to function reliably. 

Nice!

> Are there specific build flags for Mesa to enable Panfrost support, or
> something else? If so, I'll package Mesa for PrawnOS until the Debian
> repos get updated. 

Bleeding edge (which I run for my Monday-to-Friday daily driver, so
regressions occur but rarely last more than a day, additional two day
penalty if I broke something on a Friday): tree is
https://gitlab.freedesktop.org/mesa/mesa

Use the meson build system.

	$ mkdir build && cd build $ meson . ..
	-Dgallium-drivers=panfrost,kmsro -Dvulkan-drivers=
	-Ddri-drivers=

So the option you care about is gallium-drivers; clear the others to
avoid building a bunch of drivers you don't need.

---

Again, purely for experimental users right now. The stuff that works
(Weston/sway, Kodi, GL1.1-class games) works, the stuff that doesn't
doesn't, no gaurantee that app XYZ will work and if you have to ask, it
doesn't. Situation should be much better in coming months/weeks (just
keep pulling, hehe), but the above build process should stay the same.

riwale, 2019-09-16:
panfrost works quite well on a new install, still have issues with a book after kernelupgrade.sh.

Chipset documentation shows 4k support through hdmi, but i wasn't able to get more than fullhd working before panfrost integration - would this work now?

I'll test in a store if no one can confirm, still looking for a suitable second screen.

bkeys-RFA, 2019-09-17:
@riwale How did you get panfrost working? I tried installing this package
https://packages.debian.org/experimental/libgles2-mesa
Because @SolidHal mentioned in the release notes that we need a packaged libgles2-mesa for Panfrost to work. Installing that package didn't get panfrost working, not sure why. I have read some articles on compiling mesa from source but haven't tried them out yet. Does anyone have any advice on getting mesa 19.2 even if it is just a hack?

riwale, 2019-09-17:
yeah, you have to explicitly compile mesa with panfrost support, as shown in the panfrost building howto: https://panfrost.freedesktop.org/building-panfrost-mesa.html

Don't just run meson, but:

$ cd ~/mesa
$ mkdir build
$ cd build
$ meson .. . -Ddri-drivers= -Dvulkan-drivers= -Dgallium-drivers=panfrost,kmsro -Dlibunwind=false
$ ninja install

glxinfo -B will show, that you only have VMWare OpenGL Support, so it runs OpenGL, but slow.

If build correctly, output is

OpenGL vendor string: panfrost
OpenGL renderer string: panfrost
OpenGL version string: 2.1 Mesa 19.3.0-devel (git-f812cbfd88)
OpenGL shading language version string: 1.20

OpenGL ES profile version string: OpenGL ES 2.0 Mesa 19.3.0-devel (git-f812cbfd88)
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 1.0.16

riwale, 2019-09-17:
compiling mesa from source is easy inside prawnOS.

su to become root,
run "apt build-dep mesa" to save the hassle to get all needed packages one by one
clone the mesa source: "git clone https://github.com/mesa3d/mesa"
then see above and that"s it.

I recommend a fresh install of prawnOS though, i still have some trouble on an upgraded version with mpv quitting after 2-3 seconds and other bugs, but supertuxracer f.e. runs fine so far.

bkeys-RFA, 2019-09-17:
@riwale I followed your instructions and it's still using the VMWare OpenGL, which packages do I need to uninstall to get it to use this custom version of mesa? Or do I need a clean image with no graphics installed?
My only question with the clean image solution is that how would I prevent apt from installing a system package anyways?

riwale, 2019-09-17:
A clean PrawnOS wont have any mesa packages installed. if you installed any, look into your bash history and remove them. if you installed from wrongly vmware compiled source, remove the version with ninja uninstall, reconfigure meson, ninja clean, ninja install again.

Recommend a clean prawnOS install though, zero problems on a second machine, mpv and vlc all play nicely there.

bkeys-RFA, 2019-09-17:
After you ran the install on image, it didn't install any system packages for mesa? What commands did you run after you installed the custom mesa to get a desktop environment?
I'm trying to run lxqt as my preferred desktop environment

riwale, 2019-09-17:
no, i just run InstallPackages.sh to get xfce, nothing else needed, the get mesa source, compile and install as shown above. Nothing more needed, took the length of one beer on my second c201p

riwale, 2019-09-17:
I typed "reboot" into the console after the install though to get an DE ;)

bkeys-RFA, 2019-09-17:
Hmm, but if you run InstallPackages.sh to get xfce then it's going to install the system mesa which we dont want?
Regardless I will follow your steps and see if it works, will report back in about an hour and a half, I got class :-)

bkeys-RFA, 2019-09-17:
@riwale I installed xfce and followed your instructions on building mesa, but when I run `glxinfo -B` here is the output
```
bkeys@PrawnOS-Alpha:~$ glxgears
MESA-LOADER: failed to open swrast (search paths /usr/local/lib/arm-linux-gnueabihf/dri)
libGL error: failed to load driver: swrast
X Error of failed request:  GLXBadContext
  Major opcode of failed request:  151 (GLX)
  Minor opcode of failed request:  6 (X_GLXIsDirect)
  Serial number of failed request:  36
  Current serial number in output stream:  35
```

It fails to open a window or anything when I run glxgears. The only notable deviation would be that I used the system mesa-utils package

riwale, 2019-09-17:
did you reinstall from scratch?

bkeys-RFA, 2019-09-17:
Yes

riwale, 2019-09-17:
please post output from glxinfo -B, i used the same mesautils package, so thats fine

bkeys-RFA, 2019-09-17:
Yes Another notable thing is that I installed xfce, and it rebooted automatically into the DE and after the DE started up I compiled mesa from scratch. Did you have to comment out the part of the script that reboots?
```
bkeys@PrawnOS-Alpha:~$ glxinfo -B
name of display: :0.0
MESA-LOADER: failed to open swrast (search paths /usr/local/lib/arm-linux-gnueabihf/dri)
libGL error: failed to load driver: swrast
X Error of failed request:  GLXBadContext
  Major opcode of failed request:  151 (GLX)
  Minor opcode of failed request:  6 (X_GLXIsDirect)
  Serial number of failed request:  50
  Current serial number in output stream:  49
```

riwale, 2019-09-17:
please post your history and whether you compiled as root

bkeys-RFA, 2019-09-17:
I had it on a usb drive cause I didn't want to rebuild it, but I messed up the simlinks so I had to recompile.
```
cd ..
git clone https://gitlab.freedesktop.org/mesa/mesa -b master
git clone https://gitlab.freedesktop.org/mesa/mesa -b master mesa0
rm -rf mesa
mv mesa0/ mesa
cd mesa/
ls
mkdir build
cd build
sudo apt-get build-deps mesa
sudo apt-get build-dep mesa
sudo apt-get build-dep mesa -y
sudo apt-get install thunderbird -y
cd /media/bkeys/3A15-ACE7/
ls
cd mesa/
ls
ninja install 
ls
cd build
ninja install
cd
sudo ln -s /media/bkeys/3A15-ACE7/mesa/ .
ls mesa/
rm -rf mesa/
mkdir mesa
sudo ln -s /media/bkeys/3A15-ACE7/mesa/ mesa/
cd mesa/
ls
cd mesa/
cd ..
ls
rm -rf mesa/
ls /media/bkeys/3A15-ACE7/mesa/
cd
cd mesa/
ls
cd build
meson .. . -Ddri-drivers= -Dvulkan-drivers= -Dgallium-drivers=panfrost,kmsro -Dlibunwind=false
ninja install
sudo reboot
sudo apt-get install ninja -y
```

riwale, 2019-09-17:
don't get me wrong, but that looks like a huge fuckup and trying to redo things instead from starting over.

bkeys-RFA, 2019-09-17:
Hmm, I did start over? This was from a clean image (after I ran the InstallPackages.sh script for xfce), the only thing I tried reusing from the old install was a pre-compiled mesa

riwale, 2019-09-17:
if you didn"t compile as root the driver won"t be installed system wide.

Please proceed exactly as advised in my previous post.

Start with a fresh install, git clone the repository and do what the panfrost build howto tells you to do.

bkeys-RFA, 2019-09-17:
Just so I have it clear I do it all in this order:

1. Install to Internal using a fresh image
2. Compile and install mesa from scratch as root
3. Run the InstallPackages.sh script to get xfce
4. Profit!

riwale, 2019-09-17:
i have a third c201p that needs upgrading anyway, so i"ll set this one up from scratch and walk you through. Just make sure you have the latest release on a stick and internet.

bkeys-RFA, 2019-09-17:
Alright I recompiled the whole mesa as the root user and rebooted and now my glxinfo -B matches yours and I'm having panfrost. Feel free to write down your steps though for other people reading this issue

riwale, 2019-09-17:
That's great, making everything ready to consume is not the point though, i guess you learned enough.

Uninstall reddit.

riwale, 2019-09-17:
If you want to contribute, please check whether we can now use a 4K display using panfrost. You will need a minihdmi cable capable of at least hdmi 1.2 - with software rendering only fullhd is possible, but the chip is capable of 4k, so output on a 4k display should be managable.

bkeys-RFA, 2019-09-17:
I'm in the shopping for a new HDMI monitor actually, I will make sure whatever I end up buying supports this stuff.

riwale, 2019-09-17:
I can test tomorrow for 4k support, enjoy hardware acceleration for now.

gregordinary, 2019-09-20:
Following the build instructions from @riwale was able to successfully build mesa with Panfrost support. I might need some clarification here, but what benefit do we get currently with Panfrost in XFCE or LXQT as neither is using Wayland? My current understanding is that Panfrost [only supports Wayland](https://panfrost.freedesktop.org/building-panfrost-mesa.html).

Using a monitor & HDMI cable that both support 4k resolutions, I gave 4k support a go but was limited to 1920x1080. I tested this under XFCE on a default install. May not be incredibly relevant, but I also tested this again on gnome3 (wayland) using Debian Sid & Kernel 5.3 with the same results (though I did notice a nice performance gain on gnome after building & enabling mesa w/ panfrost, so that was nice).

Took output from glxinfo, inxi, and xrandr. 

Here it is on the default installation:
```
gregordinary@PrawnOS-Alpha:~$ glxinfo -B
name of display: :0.0
display: :0  screen: 0
direct rendering: Yes
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: panfrost (0xffffffff)
    Device: panfrost (0xffffffff)
    Version: 19.3.0
    Accelerated: yes
    Video memory: 4012MB
    Unified memory: yes
    Preferred profile: compat (0x2)
    Max core profile version: 0.0
    Max compat profile version: 2.1
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 2.0
OpenGL vendor string: panfrost
OpenGL renderer string: panfrost
OpenGL version string: 2.1 Mesa 19.3.0-devel (git-88b8922f57)
OpenGL shading language version string: 1.20

OpenGL ES profile version string: OpenGL ES 2.0 Mesa 19.3.0-devel (git-88b8922f57)
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 1.0.16

gregordinary@PrawnOS-Alpha:~$ inxi -Fxz
System:
  Host: PrawnOS-Alpha Kernel: 5.2.9-gnu armv7l bits: 32 compiler: gcc 
  v: 7.3.1 Desktop: Xfce 4.12.4 Distro: Debian GNU/Linux 10 (buster) 
Machine:
  Type: ARM Device System: Google Speedy details: Rockchip rev: N/A 
  serial: N/A 
Battery:
  ID-1: sbs-20-000b charge: 36.5 Wh condition: 37.7/38.0 Wh (99%) 
  model: AS19XXb3KA C201-50 status: Full 
CPU:
  Topology: Quad Core model: ARMv7 v7l variant: cortex-a12 bits: 32 
  type: MCP arch: v7l rev: 1 
  features: Use -f option to see features bogomips: 0 
  Speed: 1800 MHz min/max: 126/1800 MHz Core speeds (MHz): 1: 1800 2: 1800 
  3: 1800 4: 1800 
Graphics:
  Device-1: display-subsystem driver: rockchip_drm v: N/A bus ID: N/A 
  Device-2: rk3288-dw-hdmi driver: dwhdmi_rockchip v: N/A bus ID: N/A 
  Device-3: rk3288-mali driver: panfrost v: N/A bus ID: N/A 
  Device-4: regulator-fixed driver: reg_fixed_voltage v: N/A bus ID: N/A 
  Display: x11 server: X.Org 1.20.4 driver: modesetting unloaded: fbdev 
  resolution: 1366x768~60Hz, 1920x1080~60Hz 
  OpenGL: renderer: panfrost v: 2.1 Mesa 19.3.0-devel (git-88b8922f57) 
  direct render: Yes 
Audio:
  Device-1: rk3288-dw-hdmi driver: dwhdmi_rockchip bus ID: N/A 
  Device-2: rockchip-audio-max98090 driver: rockchip_snd_max98090 
  bus ID: N/A 
  Device-3: regulator-fixed driver: reg_fixed_voltage bus ID: N/A 
  Sound Server: ALSA v: k5.2.9-gnu 
Network:
  Device-1: ASIX AX88179 Gigabit Ethernet type: USB driver: ax88179_178a 
  bus ID: 2-1:2 
  IF: eth0 state: up speed: 1000 Mbps duplex: full mac: <filter> 
  IF-ID-1: hwsim0 state: down mac: <filter> 
  IF-ID-2: sit0 state: down mac: <filter> 
  IF-ID-3: wlan0 state: down mac: <filter> 
  IF-ID-4: wlan1 state: down mac: <filter> 
Drives:
  Local Storage: total: 14.68 GiB used: 3.23 GiB (22.0%) 
  ID-1: /dev/mmcblk2 model: HAG2e size: 14.68 GiB 
Partition:
  ID-1: / size: 14.34 GiB used: 3.23 GiB (22.5%) fs: ext4 
  dev: /dev/mmcblk2p2 
Sensors:
  System Temperatures: cpu: 27.0 C mobo: N/A 
  Fan Speeds (RPM): N/A 
Info:
  Processes: 148 Uptime: 6m Memory: 3.92 GiB used: 306.5 MiB (7.6%) 
  Init: systemd runlevel: 5 Compilers: gcc: 8.3.0 Shell: bash v: 5.0.3 
  inxi: 3.0.32 

gregordinary@PrawnOS-Alpha:~$ xrandr -q
Screen 0: minimum 320 x 200, current 3286 x 1080, maximum 4096 x 4096
eDP-1 connected 1366x768+0+0 (normal left inverted right x axis y axis) 0mm x 0mm
   1366x768      60.00*+
HDMI-1 connected 1920x1080+1366+0 (normal left inverted right x axis y axis) 600mm x 340mm
   1920x1080     60.00*   60.00    30.00    24.00  
   1920x1080i    60.00  
   1600x900      60.00  
   1280x1024     60.02  
   1280x720      60.00    60.00  
   1024x768      60.00  
   800x600       60.32  
   720x480       59.94    59.94  
gregordinary@PrawnOS-Alpha:~$ 
```
Let me know if you'd like any other output, logs, etc.

bobbybee, 2019-09-20:
Panfrost does support X11 (those are old instructions). Looks like
everything is working... cheers! :-)

gregordinary, 2019-09-20:
Awesome; thanks for the clarification!

riwale, 2019-09-20:
@gregordinary  Thanks so much for testing on 4k. I'm interested in getting a 2560x1440 display to work with that. Can you post the output of xrandr -q and maybe try adding another resolution using the answer from https://unix.stackexchange.com/questions/227876/how-to-set-custom-resolution-using-xrandr-when-the-resolution-is-not-available-i

maybe bandwith is limited, but for a second coding/output screen one could force a 2560x1440 output at 30hz or so. Just interested, because datasheet shows capability for 4k and hdmi 1.2 output:

"Single Physical Layer PHY with support for HDMI 1.4 and 2.0 operation"

"Up to 1080p at 120 Hz and 4k x 2k at 60 Hz HDTV display resolutions and up to
QXGA graphic display resolutions"

gregordinary, 2019-09-20:
@riwale Happy to help! 

So I tested out 2560x1440 with no luck. I got the new resolution/refresh rate added successfully. Showed up in XFCE Display Settings too, however trying to switch to the new mode would fallback to 1920x1080. I tried with the built-in screen disabled as well. 

```
gregordinary@PrawnOS-Alpha:~$ xrandr --newmode "2560x1440_30.00"  146.27  2560 2680 2944 3328  1440 1441 1444 1465  -HSync +Vsync
gregordinary@PrawnOS-Alpha:~$ xrandr --addmode HDMI-1 "2560x1440_30.00"
gregordinary@PrawnOS-Alpha:~$ xrandr --output HDMI-1 --mode "2560x1440_30.00"
xrandr: Configure crtc 1 failed
```
I also did an EDID decode to make sure the proper information was being sent over from the external display: 

```
gregordinary@PrawnOS-Alpha:~$ xrandr --props
Screen 0: minimum 320 x 200, current 3286 x 1080, maximum 4096 x 4096
eDP-1 connected 1366x768+0+312 (normal left inverted right x axis y axis) 0mm x 0mm
	link-status: Good 
		supported: Good, Bad
	CONNECTOR_ID: 41 
		supported: 41
	non-desktop: 0 
		range: (0, 1)
   1366x768      60.00*+
HDMI-1 connected 1920x1080+1366+0 (normal left inverted right x axis y axis) 600mm x 340mm
	EDID: 
		00ffffffffffff001e6d085b73740200
		071b0103803c2278ea3035a7554ea326
		0f50542108007140818081c0a9c0d1c0
		81000101010108e80030f2705a80b058
		8a0058542100001e04740030f2705a80
		b0588a0058542100001a000000fd0038
		3d1e873c000a202020202020000000fc
		004c4720556c7472612048440a200122
		020330714d902220050403020161605d
		5e5f230907076d030c001000b83c2000
		6001020367d85dc401788003e30f0006
		023a801871382d40582c450058542100
		001a565e00a0a0a02950302035005854
		2100001a000000000000000000000000
		00000000000000000000000000000000
		000000000000000000000000000000aa
	link-status: Good 
		supported: Good, Bad
	CONNECTOR_ID: 43 
		supported: 43
	non-desktop: 0 
		range: (0, 1)
   1920x1080     60.00*   60.00    30.00    24.00  
   1920x1080i    60.00  
   1600x900      60.00  
   1280x1024     60.02  
   1280x720      60.00    60.00  
   1024x768      60.00  
   800x600       60.32  
   720x480       59.94    59.94  
   2560x1440_30.00  30.00  

gregordinary@PrawnOS-Alpha:~$ xrandr --props | edid-decode
EDID version: 1.3
Manufacturer: GSM Model 5b08 Serial Number 160883
Made in week 7 of 2017
Digital display
Maximum image size: 60 cm x 34 cm
Gamma: 2.20
DPMS levels: Standby Suspend Off
RGB color display
First detailed timing is preferred timing
Display x,y Chromaticity:
  Red:   0.6523, 0.3349
  Green: 0.3046, 0.6367
[... 75 lines cut ...]
               1080 1084 1089 1125 vborder 0
               +hsync -vsync 
               VertFreq: 60 Hz, HorFreq: 67500 Hz
Detailed mode: Clock 241.500 MHz, 600 mm x 340 mm
               2560 2608 2640 2720 hborder 0
               1440 1443 1448 1481 vborder 0
               +hsync -vsync 
               VertFreq: 59 Hz, HorFreq: 88786 Hz
Checksum: 0xaa (valid)

```

I have to start my day job but I'll circle back to this later. In the meantime let me know if you'd like me to test anything else out specifically. 

Side note, when I tested yesterday with kernel 5.3/Debian Sid/Gnome/Wayland the max resolution showed 8192x8192. Trying not to clutter these posts up with non-default tests, still found that interesting. 

```
gregordinary@PrawnOS-Alpha:~$ xrandr -q
Screen 0: minimum 320 x 200, current 3286 x 1080, maximum 8192 x 8192
XWAYLAND0 connected 1366x768+0+0 (normal left inverted right x axis y axis) 256mm x 144mm
   1366x768      59.80*+
XWAYLAND2 connected 1920x1080+1366+0 (normal left inverted right x axis y axis) 600mm x 340mm
   1920x1080     59.96*+
gregordinary@PrawnOS-Alpha:~$ 
```

gregordinary, 2019-09-20:
Messed around a bit more. 
The "maximum" value listed under xrandr -q is the combined total supported output on all screens. I tried setting a 4k resolution and it didn't allow me to attempt that unless I disabled the laptop display. 

```
gregordinary@PrawnOS-Alpha:~$ xrandr --newmode "4k" 339.57  3840 4080 4496 5152  2160 2161 2164 2197  -HSync +Vsync
gregordinary@PrawnOS-Alpha:~$ xrandr --addmode HDMI-1 "4k"
gregordinary@PrawnOS-Alpha:~$ xrandr --output HDMI-1 --mode "4k"
xrandr: screen cannot be larger than 4096x4096 (desired size 5206x2160)
```

I agree 4k should be supported. I double-checked the [rk3288 datasheet](https://github.com/rockchip-linux/docs/raw/master/SoC%20platform%20related/RK3288/Rockchip%20RK3288%20Datasheet%20V2.4-20180530.pdf) you mentioned. 

A lot of the data from xrandr and other utilities are probably pulling from the same place, still I ran the following: 

```
gregordinary@PrawnOS-Alpha:~$ find /sys/devices -name "edid"
/sys/devices/platform/display-subsystem/drm/card2/card2-eDP-1/edid
/sys/devices/platform/display-subsystem/drm/card2/card2-HDMI-A-1/edid
/sys/devices/platform/vkms/drm/card1/card1-Virtual-1/edid
gregordinary@PrawnOS-Alpha:~$ cat /sys/class/drm/card2-HDMI-A-1/modes
1920x1080
1920x1080
1920x1080i
1920x1080
1920x1080
1600x900
1280x1024
1280x720
1280x720
1024x768
800x600
720x480
720x480

```

A bit at a loss on what to try next. Right now I'm searching for other rk3288-based platforms (like Asus Tinkerboard) to see if people have had luck getting 4k resolutions working. 

Also in my searching I [found the following](https://lkml.org/lkml/2019/5/12/224): 

> Oh, I might have forgot to mention, we accidentally found that on RK3288, the decoder is in fact an [Hantro G1](https://web.archive.org/web/20190322103810/http://www.verisilicon.com/IPPortfolio_14_58_2_HantroG1.html) (the full version). 

Kernel 5.3 is bringing hardware-decoding for MPEG2 & H.264 to the 3288 with more to come (enabled with hantro_video, and hantro_video_rockchip) and it looks like there are [GStreamer plugins](https://github.com/linux4sam/gst1-hantro-g1) for the G1 already.

gregordinary, 2019-09-20:
Did some more searching on 4k support. There are two referenced Visual Output Processors (VOP): 
- vop_little
- vop_big

vop_little supports 2k and vop_big can support 4k. HDMI can talk to both vops and *should* be able to use either. Although this is supposed to be automatic, there are patches forcing HDMI to use vop_big and eDP to use vop_little. One of these patches is in the PrawnOS repo under patches-untested. https://github.com/SolidHal/PrawnOS/blob/master/resources/BuildResources/patches-untested/0020-RK3288-HDMI-clock-hacks-combined.patch

Could not get that patch to apply, may have been incorporated already. 

Few other examples: 

1. [rk3288-based Firefly board needed firmware compiled to use vop_big to support 4k.](http://bbs.t-firefly.com/forum.php?mod=viewthread&tid=1820)
2. [Similar config made here (then rolled back)](https://www.mail-archive.com/linux-kernel@vger.kernel.org/msg1314986.html)
3. [Rockchip-linux kernel did something similar too.](https://github.com/rockchip-linux/kernel/commit/287df96af0434218323d4219a762c354cf5d514f)

Not sure if this is the answer, but seemed relevant.

tehbra1n, 2019-10-21:
I'm interested in testing out Panfrost so I went through the install process but seem to still be getting the VMware driver.  The steps I followed, from a fresh install of 0.18-alpha, were:

```
$ apt build-dep mesa
$ apt install python3-setuptools
$ apt install mesa-utils
$ git clone https://gitlab.freedesktop.org/mesa/mesa -b master
$ cd mesa
$ su
# mkdir build
# cd build
# meson .. . -Ddri-drivers= -Dvulkan-drivers= -Dgallium-drivers=panfrost,kmsro 
# ninja install
# reboot
$ glxinfo -B

```
glxinfo reports the VMware driver still.  I'm not really sure what I'm missing here.  The instructions here: [https://panfrost.freedesktop.org/building-panfrost-mesa.html](https://panfrost.freedesktop.org/building-panfrost-mesa.html) include this last step:

`# chmod 0666 /dev/dri/*
`

Which I also tried with no change.

tehbra1n, 2019-10-21:
> I'm interested in testing out Panfrost so I went through the install process but seem to still be getting the VMware driver. The steps I followed, from a fresh install of 0.18-alpha, were:
>
> ```
> $ [...]

Now after doing:

```
# apt update
# apt upgrade
```

I show the panfrost driver running.  It picked up 7 packages to upgrade.  So, it seems like either a fully updated system is needed or I'm still missing something FWIW.

bobbybee, 2019-10-21:
Panfrost is in Debian buster defaults so these steps shouldn't be
necessary anymore, just the apt update && apt upgrade should do the
trick.

tehbra1n, 2019-10-21:
> Panfrost is in Debian buster defaults so these steps shouldn't be necessary anymore, just the apt update && apt upgrade should do the trick.
> […](#)

That might explain the behavior I observed with the update/upgrade enabling Panfrost.  However I just did another fresh install and then just an update/upgrade but am still not using Panfrost.  Buster seems to have Mesa 18.3.6 while Panfrost requires 19.1.

bobbybee, 2019-10-21:
Ack! I meant bullseye, not buster. My bad!

tehbra1n, 2019-10-21:
I'm still not entirely sure what is up.  Output from dmesg:

```
[    1.383786] panfrost ffa30000.gpu: clock rate = 400000000
[    1.389842] panfrost ffa30000.gpu: failed to get regulator: -517
[    1.396563] panfrost ffa30000.gpu: regulator init failed -517
[    3.682079] panfrost ffa30000.gpu: clock rate = 400000000
[    3.688264] panfrost ffa30000.gpu: mali-t760 id 0x750 major 0x0 minor 0x0 status 0x1
[    3.697000] panfrost ffa30000.gpu: features: 00000000,100277bf, issues: 00000000,24040400
[    3.706220] panfrost ffa30000.gpu: Features: L2:0x07120206 Shader:0x00000000 Tiler:0x00000809 Mem:0x1 MMU:0x00002830 AS:0xff JS:0x7
[    3.706223] panfrost ffa30000.gpu: shader_present=0xf l2_present=0x1
[    3.707319] [drm] Initialized panfrost 1.0.0 20180908 for ffa30000.gpu on minor 3
```

bobbybee, 2019-10-21:
Seems like it's probing okay to me..?

rk-zero, 2019-10-21:
when you install all dependencies for mesa it also installs llvm, apt package llvm-7-dev

when choosing the correct driver / dri / so whatever x chooses llvm first and you are stuck with vmware.

remove the package with apt remove llvm-7-dev, reboot, x chooses panfrost.

I have no idea how to make it choose panfrost by default. This workaround fixed it til recent updates, i had a fresh install today which just restarts x all the time.

But until today - that worked, so if you take a look into your x log files, you'll see it chooses the vmware driver first and ignore panfrost.

So ... i still have no clue, but that is the closest path i found to fix this.

rk-zero, 2019-10-21:
# apt build-dep mesa

i think this is a shotgun approach, for compiling just for panfrost you won't need everything.

rk-zero, 2019-10-21:
[    66.427] (II) Loading sub module "glamoregl"
[    66.427] (II) LoadModule: "glamoregl"
[    66.427] (II) Loading /usr/lib/xorg/modules/libglamoregl.so
[    66.427] (II) Module glamoregl: vendor="X.Org Foundation"
[    66.427]    compiled for 1.20.4, module version = 1.0.1
[    66.427]    ABI class: X.Org ANSI C Emulation, version 0.4

from a working panfrost install

SolidHal, 2019-10-22:
I'll add a section to `InstallPrawnOS.sh` to choose to install mesa and remove llvm. Not ready to make it default yet without some more testing.

tehbra1n, 2019-10-22:
> when you install all dependencies for mesa it also installs llvm, apt package llvm-7-dev
> 
> when choosing the correct driver / dri / so whatever x chooses llvm first and you are stuck with vmware.
> 
> remove the package with apt remove llvm-7-dev, reboot, x chooses panfrost.
> 
> I have no idea how to make it choose panfrost by default. This workaround fixed it til recent updates, i had a fresh install today which just restarts x all the time when i remove llvm
> 
> But until today - that worked, so if you take a look into your x log files, you'll see it chooses the vmware driver first and ignore panfrost.
> 
> So ... i still have no clue, but that is the closest path i found to fix this.

So I removed llvm-7-dev and everything works great now with Panfrost.  The behavior you mentioned with X restarting after login after installing Panfrost is something I noticed on recent PrawnOS alphas but never dug into.  On 0.18-alpha I did not run into this bug.  It may be a result of this fix #131

rk-zero, 2019-10-23:
I try not to spam this thread any longer, but made panfrost work on latest alpha 19.

Basic idea is to omit build-dep and install all packages needed for the smaller panfrost compile of mesa.

Taking hints from https://linux-sunxi.org/Mali_Open_Source_Driver and trying one by one working code is:

`sudo apt -y install git libx11-dev meson pkg-config python3-setuptools python3-mako zlib1g-dev libexpat1-dev libdrm-dev bison flex libwayland-dev wayland-protocols libwayland-egl-backend-dev libxext-dev libxdamage-dev libx11-xcb-dev libxcb-glx0-dev libxcb-dri2-0-dev libxcb-dri3-dev libxcb-present-dev libxshmfence-dev libxxf86vm-dev libxrandr-dev gettext
sudo apt -y install mesa-utils

cd /tmp
git clone https://gitlab.freedesktop.org/mesa/mesa -b master
cd mesa
mkdir build
cd build
meson .. . -Ddri-drivers= -Dvulkan-drivers= -Dgallium-drivers=panfrost,kmsro -Dlibunwind=false
sudo ninja install`

albsod, 2019-10-29:
@rk-zero: I've tried your scripts on a fresh install, before running InstallPackages.sh. To make it work, I first had to edit it and remove sudo, since there was only the root user.

Panfrost support is great because it allows playing films and games such as Hedgewars smoothly, but for some reason Xfce is **less** responsive compared with the default install. Also, with Panfrost, there is some flickering and missing icons in the dock. I also tried installing the full GNOME DE. It doesn't have any flickering or missing icons issues but it's quite laggy compared with an older x86_64 machine with Intel graphics.

rk-zero, 2019-10-29:
Hey @albsod 

Disable desktop composition under Settings --> Windows Manager Tweaks, last tab "Compositor"

I take it from the panfrost video at ccc that there is right now no way to render the whole desktop, just single windows. Youtube will work fine afterwards, maybe look into h264ify if you have problems. Still, some videos lag audio with firefox and chrome wont recognize panfrost yet.

I don't use the dock and uninstalled plank alltogether, but maybe it'll work after disabling composition.

If you like to further development, i use Kernel 5.3.1. with hantro and will try to make sway or similiar work. For now, i guess we should add a line to disable composition in the panfrost compile script, will look into that.

https://forum.armbian.com/topic/6506-tutorial-3d-video-acceleration-and-opencl-in-rk3288-boards-with-new-44-default-kernel/ is worth a read. My fork is pretty broken right now, but 5.3.1. (kind of) builds and has support.

You are always welcome to catch up with me on libreboot irc.

albsod, 2019-10-29:
Thanks @rk-zero! I think it works slightly better without compositing. It didn't help with plank though so I uninstalled the package.

This is the first time I've heard of the Hantro VPU driver. I don't have time to rebuild and test Linux versions right now, but I'd be happy to try any prebuilt kernel for PrawnOS.

SolidHal, 2019-10-29:
I tested panfrost with the mesa packages in bullseye, ran into similar graphical glitches and even had the system freeze when the lockscreen was enabled. Haven't had a chance to debug these fully yet, but will likely update the kernel and test again with mesa built from master instead of the debian package.

SolidHal, 2019-12-03:
Panfrost with a mostly stable mesa can now be built and installed using `InstallResources/CompilePanfrost.sh`

bobbybee, 2019-12-03:
\o/

albsod, 2019-12-03:
> Panfrost with a mostly stable mesa can now be built and installed using `InstallResources/CompilePanfrost.sh`

This is really cool. I have a question though: if `CompilePanfrost.sh` should be run after `InstallPackages.sh` (because it relies on **sudo** being set up), doesn't it conflict with the graphics configuration applied by the latter?

rk-zero, 2019-12-03:
@albsod it won't cause problems run before or after. If you use it with su - beforehand you can remove sudo from the script.

@bobbybee This is awesome work and a great inspiration to look further into C and hardware coding. If you like feedback, openarena still has problems for me, in more complex scenes only parts of the screen gets updated (little hard to aim):

`[26247.559991] panfrost ffa30000.gpu: js fault, js=0, status=INSTR_TYPE_MISMATCH, head=0x11e26b00, tail=0x11e26b00
[26247.571571] panfrost ffa30000.gpu: gpu sched timeout, js=0, status=0x52, head=0x11e26b00, tail=0x11e26b00, sched_job=71aa40ae
`

SolidHal, 2019-12-03:
It doesn’t conflict, the installation process seems to tell the system to use Mesa/pan frost somehow. Not sure how that process works, but I can confirm that after running CompilePanfost.sh the system is using Mesa/panfrost

albsod, 2019-12-03:
@SolidHal: OK, great. With the previous version of the script, I could only make it work if I 1)  removed sudo from CompilePanfost.sh, 2) ran it as root and then 3) ran InstallPackages.sh. If I executed the scripts in the reverse order, then XFCE got messed up.

rk-zero, 2020-01-12:
I just realized you can keep desktop composition on and youtube still works fine even at 1080p - i guess this is resolved.

Still have crashes with some document viewers and libreoffice. I disabled hardware acceleration in libreoffice prior to compiling panfrost, still crashes and returns to login screen. Any ideas? Not getting much from the logs, but will gladly post some if someone can tell me which files/logs are needed.

[19 more replies]

closed 2020-10-29 by SolidHal