SolidEva/PrawnOS #100: Getting corssystem/mosys to work

cspuke, 2019-06-13:
Probably related to #5 

I'm trying to change nvram boot options from PrawnOS on libreboot.
I installed crossystem from the official stretch repositories but i can't get it to work.
It sseems like it depends on mosys, but wether i get mosys precompiled from the official recovery image or i build it myself it still won't work.

# mosys
cros_ec_command_dev: Transfer failed: -1
Platform not supported

Running with strace:
stat64("/dev/port", {st_mode=S_IFCHR|0640, st_rdev=makedev(1, 4), ...}) = 0
open("//dev/cros_ec", O_RDWR)           = 4
ioctl(4, _IOC(_IOC_READ|_IOC_WRITE, 0x3a, 0x00, 0x1c), 0xbe948518) = -1 ENOTTY (Inappropriate ioctl for device)
write(2, "cros_ec_command_dev: Transfer fa"..., 41cros_ec_command_dev: Transfer failed: -1
) = 41
close(4)                                = 0
write(2, "Platform not supported\n", 23Platform not supported
) = 23
semop(32768, [{0, 1, SEM_UNDO}], 1)     = 0
exit_group(-1)                          = ?
+++ exited with 255 +++

So it looks like it uses a ioctl that the kernel do not support?
I tried recompiling with

CONFIG_CMD_CROS_EC=y
CONFIG_CROS_EC=y
CONFIG_CROS_EC_I2C=y

But still no luck...
Kernel is is 4.17.19 mainline

SolidHal, 2019-06-22:
does mosys work on the chromeos-3.14 kernel with a PrawnOS?

cpuke, 2019-06-23:
Can you give me a hint on how to try switching kernel without reinstalling? Can i just get the kernel from the official recovery image and switch it?

SolidHal, 2019-06-24:
You could copy the kernel from https://archive.org/details/devuan-jessie-c201-libre-2GB which is a devsus release from when it used the chromeos-3.14 kernel. 

Assuming you have PrawnOS installed to the internal EMMC I think you could do it by:
- Writing that devsus image to a flash drive.
- Write PrawnOS to another flash drive
- Boot the laptop to the PrawnOS flash drive
- Plug in the devsus flash drive
- confirm what /dev/sd* the devsus drive is mapped to using something like `fdisk -l`
-Note that the kernel is the first partition of the drive, so its is /dev/sd*p1
- copy the kernel from the devsus drive to the internal by running something like `dd if=/dev/sd*p1 of=/dev/mmcblk2p1`

The kernel partitions of the devsus image and PrawnOS should be the same size, 65536 sectors
You can confirm that as well using `fdisk -l`

I hope that helps! 

PS: Out of curiosity, what can/are you trying to change about the nvram boot options for libreboot on the c201? I haven't looked too closely at that stuff.

cpuke, 2019-06-26:
Thanks for your help, i'll give it a try when i have some spare time :)

> PS: Out of curiosity, what can/are you trying to change about the nvram boot options for libreboot on the c201? I haven't looked too closely at that stuff.

I wanted to harden the system a bit so i was looking to disable usb booting (and i'm currently looking into the feasibility of reenabling vboot in libreboot with self rolled keys)

SolidHal, 2019-06-26:
No problem! I did recently make a script to upgrade the kernel of PrawnOS from a usb or sd card, you can find that here https://github.com/SolidHal/PrawnOS/blob/master/scripts/InstallScripts/UpgradeKernel.sh 
It expands on the instructions above by also copying modules in the filesystem, so you can give that a try if those instructions aren't working out. 

> I wanted to harden the system a bit so i was looking to disable usb booting (and i'm currently looking into the feasibility of reenabling vboot in libreboot with self rolled keys)

self rolled vboot would be pretty cool

riwale, 2019-09-05:
You can get crossystem to work on a currrent prawnOS-release without rebooting/reinstalling.

Grab a chrome-os device and copy the whole root-partition onto an usb drive in developer mode.

in PrawnOS, simply chroot into it, mount --bind proc, sys etc. beforehand.

crossystem --all shows some errors, but you can change devboot switches without a problem.

goin2mars, 2020-02-06:
> ...
> If interested, i can upload the directory structure from a pinned 67 chromeos release

Hey could you please do this? I really don't want to buy a third one of these, yet. I'm guessing I disabled external booting by turning off developer mode and now I've also gone and broken the OS on the thing so here I sit, wishing I could have some possible way to run crossystem because apparently local compilation fails quite readily also.

rk-zero, 2020-02-06:
Yeah, no problem.

find the zip here: [(https://dec.k-ze.ro/chrome/chromeos_10452.96.0_veyron-speedy_recovery_stable-channel_speedy-mp-v3.rootstripped.zip)]

unzip the file to /mnt/sd/chromeosroot f.e. and cd into it.

To add to that, for mounting proc, the correct command order is: 

mount -t proc proc proc/
mount --rbind /sys sys/
mount --rbind /dev dev/
mount --rbind /run run/

chroot .

Tried it with latest prawnOS, crossystem still works!

Have fun :)

@SolidHal can you please fix the typo in the issue title? May be found easier in google then

Maccraft123, 2020-02-22:
> I'm trying to change nvram boot options from PrawnOS on libreboot.
Fetch coreboot repo and compile nvramtool.

SolidHal, 2020-03-31:
Added script to build mosys, libfmap from source which make crossystem fully functional. See commit fc1deb4796f71454e944a6c422e8613c1303b506

Find the crossystem section on the readme for more info.

SolidHal, 2020-03-31:
Also, I'm moving the build and installation of mosys/crossystem into buildFilesystem.sh but that script will provide what you need on existing builds.

closed 2020-03-31 by SolidHal

rk-zero, 2020-03-31:
That's great work, thanks much!

SolidHal, 2020-04-01:
As of d520bd5754209b481287c54095d1896bbf5a0ca5 crosystem and mosys are now built into the PrawnOS filesystem! I am leaving the separate build script in scripts/InstallScripts for now for those on older build and for other distributions to leverage.