SolidEva/PrawnOS #102: Loading the 9271 firmware freezes the system

albsod, 2019-06-30:
With the new 4.19 kernel, my c201 freezes immediately upon inserting a TET-N150 WiFi dongle that uses the 9271 firmware. On the other hand, my TET-N300 donge that uses the 7010 firmware doesn't trigger any such issue.

tehbra1n, 2019-07-01:
I experience the same issue with [this adapter](https://wikidevi.com/wiki/ThinkPenguin_TPE-N150USB).  @SolidHal had some conversations wrt to it and Parabola [here.](https://labs.parabola.nu/issues/2261)

That thread mentions a patch at the end that seemed to resolve the issue but I am unsure if it has been incorporated into PrawnOS.

Weirdly, I have another Atheros 9271 based dongle, identical[ to this one of unknown origin](https://www.ebay.com/itm/Atheros-AR9271L-802-11n-150Mbps-Wireless-USB-WiFi-Adapter-for-Linux-Kali-Linux/302918103655?hash=item4687537267:g:Kj0AAOSwLNVc1qlR) but that works just fine.  The only reason I'd prefer the TPE-N150USB is the form factor, especially considering the size of the C201.

patrickr81, 2019-07-01:
Same issue, kernel panic on insertion, using TPE-N150USB. It doesn't happen every time I plug it in, but fairly regularly.

I'm also seeing some intermittent drops in wifi connectivity (perhaps unrelated) and kernel panics while the adapter is connected as described in issue #77.

SolidHal, 2019-07-02:
@albsod, @tehbra1n, @patrickr81 What release or git version are you guys using? 
Releases 10 and 11 both have fixes that may have fixed this issue. Some people have reported success. 

Release 11 does have a regression regarding the 7010 #42, but commit 61aa9090dbaa3c7de348f44217c9519397570690 should have fixed that. Unfortunately the fix is not included in a release yet as I'm away right now so I can't do a proper test.

tehbra1n, 2019-07-02:
> @albsod, @tehbra1n, @patrickr81 What release or git version are you guys using?
> Releases 10 and 11 both have fixes that may have fixed this issue. Some people have reported success.
> 

I am on Release 11, installed fresh from the image on the Releases page.  I could pull from git and build an image and try if it would help.

patrickr81, 2019-07-02:
From source, building on stretch, commit 61aa909. I'm using the cli to bring up the network.

(Looking at my dmesg, it looks like the ath9k_htc driver is failing to register initially if the dongle is plugged in at boot. It registers when removed and reinserted. Probably not connected, but I mention it just in case.)

albsod, 2019-07-02:
@SolidHal I'm using the prebuilt v0.11-alpha image with XFCE on internal storage.

patrickr81, 2019-07-02:
[image]

Hope this helps. I’m gonna install the release 11 now and see if the problem persists.
 @SolidHal

SolidHal, 2019-07-09:
@tehbra1n Interesting that you're seeing different results with different ath9271 devices. 

I'm building the open ar9271 and ar7010 firmwares into the kernel to fix #103, might impact this as well as it'll remove any dependencies on the filesystem. I'll have a release pushed for that when I have a free moment. 

### For anyone else experiencing this issue: 
#### Please add your: 
PrawnOS Version
Make/model of your ath9271 device
Pictures of any kernel panic logs you can provide

tehbra1n, 2019-07-16:
Issue continues on prebuilt v12 and a current pull from git building the image in a VM running stretch.

albsod, 2019-07-16:
It remains for me as well. I used the kernel upgrade script instead of a complete reinstallation this time. By the way, I've noticed that the freeze takes place after authentication and connection to a network. Before that, the dongle discovers available networks without producing a kernel pantic.

tehbra1n, 2019-07-17:
> It remains for me as well. I used the kernel upgrade script instead of a complete reinstallation this time. By the way, I've noticed that the freeze takes place after authentication and connection to a network. Before that, the dongle discovers available networks without producing a kernel panic.
> 
> Version tested: prebuilt v11 + prebuilt v12 kernel.

I've noticed a correlation with the panic happening only after auth/connecting to a network as well but I haven't done enough tests to be sure there is causation there.   It sure seems like it though.

RobertoJBeltran, 2019-07-21:
I've had this issue with the TPE-N150USB in both the prebuilt v11 and v12 images. As suggested in the parabola thread linked to by @tehbra1n I got a USB hub and it seems to have solved the issue. I'll report back after I attempt a full day of work.

RobertoJBeltran, 2019-07-25:
It seems I just had a stroke of good luck. I managed to install LibreOffice, give two slide presentations, install LibreJS and then after surfing the web a bit I got a freeze and it's been icy since then.

Can anyone recommend any 7010 or 9271 dongles that are known to work in addition to the one @tehbra1n mentioned? I miss taking the little thing around town. It doesn't need to have a small form factor.

SolidHal, 2019-08-02:
Well this bug is irritating hahaha.
Sorry for the delay, I've been traveling for a while. 

@tehbra1n thanks for the info.
I'd like to find a reliable way to recreate this issue, since some people have differing luck encountering/not-encountering this. (myself included)

@RobertoJBeltran What I can say for sure fixes this issue is to solder in the adapter to where the webcam connects like I document here: https://github.com/SolidHal/AsusC201-usb-wifi-from-webcam
I did this to my main c201 as have a couple other people and never encounter this issue. 

I'll continue trying to debug this issue on my second c201.

SolidHal, 2019-08-22:
Hey everyone, please test out the most recent release https://github.com/SolidHal/PrawnOS/releases/tag/v0.13-alpha I developed a new workaround and tossed out the old one that seemed to cause random memory corruption and  kernel panics.

I tested the new workaround on both a 2GB and 4GB model for 24 hours by repeatedly downloading a 4GB file and both ran without error, so hopefully you all have the same experience. 

If you have any issues, make sure to leave a comment with which port you were using (closest to the screen or farthest from the screen) and any error logs/screenshots you can catch.

albsod, 2019-08-22:
Hi Solidhal: I used the great kernel-upgrade script and now the 9271 WiFi dongle seems to be working fine on my machine. Many thanks!

tehbra1n, 2019-08-22:
I compiled from git and installed and with limited testing so far there have been no hangs.  I'll follow up when I get more time behind the wheel.

georgegeddes, 2019-09-07:
I have the same two wifi dongles as @tehbra1n and both seem to be working perfectly since I installed the kernel from v0.13-alpha. I will update if anything changes. Thank you @Solidhal!

SolidHal, 2019-09-12:
Seems like this is resolved, thanks for your help everyone!

closed 2019-09-12 by SolidHal

riwale, 2019-09-15:
wifi dongles 9271 work perfectly in latest release, no need to unplug, no issues with power drains. Sometimes, when i booted up with wifi connected, i couldn't plug in another device that drains some power, seems to be resolved.