SolidEva/PrawnOS #106: Kernel 5.2 support

SolidHal, 2019-07-12:
Now that linux-libre 5.2 is out, test it and find out if we can enable panfrost!

gregordinary, 2019-07-15:
I edited the makefile and changed KVER to 5.2. It fetched 5.2 no problem, however I did run into a few issues. 

After running `sudo make image` the first time through it stops, I'm guessing because a patch being applied was already applied upstream or something:

```
Hunk #1 succeeded at 98 (offset -14 lines).
+ echo /home/<USER>/Downloads/PrawnOS/resources/BuildResources/patches-tested/kernel/0030-cros-ec-check-for-mkbp-events-on-resume-only-if-supported.patch
/home/<USER>/Downloads/PrawnOS/resources/BuildResources/patches-tested/kernel/0030-cros-ec-check-for-mkbp-events-on-resume-only-if-supported.patch
+ patch -p1
patching file drivers/mfd/cros_ec.c
Reversed (or previously applied) patch detected!  Assume -R? [n] 
Apply anyway? [n] 
Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file drivers/mfd/cros_ec.c.rej
make[1]: *** [makefile:71: kernel] Error 1
make[1]: Leaving directory '/home/<USER>/Downloads/PrawnOS'
make: *** [makefile:91: image] Error 2
```

Running `sudo make image` again, it proceeds through. I'm then prompted to enable/disable new kernel features, Panfrost being one of them. That's the only additional one I enabled: 
`Panfrost (DRM support for ARM Mali Midgard/Bifrost GPUs) (DRM_PANFROST) [N/m/y/?] (NEW) y`

After that runs through, the next error I get is a duplicate property name in a dtsi file: 

```
arch/arm/boot/dts/rk3288.dtsi:576.3-25: ERROR (duplicate_property_names): /tsadc@ff280000:rockchip,grf: Duplicate property name
ERROR: Input tree has errors, aborting (use -f to force output)
make[3]: *** [scripts/Makefile.lib:294: arch/arm/boot/dts/rk3288-evb-rk808.dtb] Error 2
make[3]: *** Waiting for unfinished jobs....
arch/arm/boot/dts/rk3288.dtsi:576.3-25: ERROR (duplicate_property_names): /tsadc@ff280000:rockchip,grf: Duplicate property name
ERROR: Input tree has errors, aborting (use -f to force output)
make[3]: *** [scripts/Makefile.lib:294: arch/arm/boot/dts/rk3288-evb-act8846.dtb] Error 2
make[2]: *** [Makefile:1249: dtbs] Error 2
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/home/<USER>/Downloads/PrawnOS/build/linux-5.2'
make[1]: *** [makefile:71: kernel] Error 2
make[1]: Leaving directory '/home/<USER>/Downloads/PrawnOS'
make: *** [makefile:91: image] Error 2
```

The file is:
`PrawnOS/build/linux-5.2/arch/arm/boot/dts/rk3288.dtsi`

The relevant section is:
```
        tsadc: tsadc@ff280000 {
                compatible = "rockchip,rk3288-tsadc";
                reg = <0x0 0xff280000 0x0 0x100>;
                interrupts = <GIC_SPI 37 IRQ_TYPE_LEVEL_HIGH>;
                clocks = <&cru SCLK_TSADC>, <&cru PCLK_TSADC>;
                clock-names = "tsadc", "apb_pclk";
                resets = <&cru SRST_TSADC>;
                reset-names = "tsadc-apb";
                pinctrl-names = "init", "default", "sleep";
                pinctrl-0 = <&otp_gpio>;
                pinctrl-1 = <&otp_out>;
                pinctrl-2 = <&otp_gpio>;
                #thermal-sensor-cells = <1>;
                rockchip,grf = <&grf>;
                rockchip,hw-tshut-temp = <95000>;
                rockchip,grf = <&grf>;
                status = "disabled";
        };
```

`rockchip,grf = <&grf>; `is in there twice, so I commented one out and once again re-ran `sudo make image` enabling Panfrost along the way. However at the final stage, I encounter the following error: 

```
+ losetup -P /dev/loop5 PrawnOS-buster-Alpha-c201-libre-2GB.img-BASE
losetup: PrawnOS-buster-Alpha-c201-libre-2GB.img-BASE: failed to set up loop device: Device or resource busy
+ cleanup
```

I do not run into this issue running the script with default values. I'm running Lubuntu 19.10 with kernel 5.1 if that makes any difference here. Let me know if you'd like any additional info or if you'd like me to try a different configuration.

gdallasdye, 2019-07-15:
rockchip-dwc2-usb-partial-power-down.donotpatch
0030-cros-ec-check-for-mkbp-events-on-resume-only-if-supported.donotpatch
usb-core-remove-local-irq-save-around-complete-handler.donotpatch

There are three files I need to dummy out to get through the patching phase. Although I've read them, I'm not sure if I can advise others to do so. Still though, that gets me to the duplicate error @gregordinary encounters. This happens on both mainline and deblobbed sources. I'll try his suggestion and see if I can get any further and report back. 

I was using my prefered search engine earlier and found this command I intend to experiment with: 

make menuconfig KCONFIG_CONFIG=linux/arch/arm/configs/myconfigfile

As a fallback approach I've also had success building the panfrost-4.19 kernel referenced by @SolidHal in the readme. However I have yet to the whole mesa build with meson thing. I'm learning.

gregordinary, 2019-07-15:
@gdallasdye Thanks for the additional info on the patches. It'll give me more to dive into, test out, and learn. 

I've also had previous success building panfrost. I did so with the 5.0.7 (non-libre) branch from the panfrost gitlab repo before it got merged into mainline. I also installed mesa-git (think 19.1 at the time) for the Gallium3D drivers. Was only able to run kmscube in that setup. I make reference to that in this [reddit comment](https://old.reddit.com/r/linux/comments/bebm93/linux_on_asus_chromebook_c201/elvtukh/). Or [full thread](https://old.reddit.com/r/linux/comments/bebm93/linux_on_asus_chromebook_c201/) if you are interested. 

Now Alyssa has done a bunch of work on Panfrost at Collabora and the last update show the drivers [working with Gnome3](https://www.collabora.com/news-and-blog/blog/2019/06/26/gnome-meets-panfrost/). Looks promising for the C201 once it's building again.

tslilc, 2019-07-25:
I can confirm @gdallasdye 's observation that those three patches had been merged by 5.2 and so aren't necessary. The `0006-ARM-DTSI-rk3288-Missing-GRF-handles.patch` needs some manual work (as observed) as `rk3288.dtsi` seems to have been partially patched upstream.

With or without these patches, and with or without `CONF_DRM_PANFROST`, using the default settings for everything not set by the `4.19.53` config the kernel produced by the build process (`make kernel` still works, and it's a short jump from there to build modules and dd it onto the device) leaves my C201 at a blank screen for as long as i dare to run it. Has anyone had any luck coaxing libre-5.2 into booting?

SolidHal, 2019-08-04:
Currently debugging why 5.2 won't boot. Seems to not even get to outputting serial in my tests.

Testing with 5.1.21 to see if it is a newer problem. 4.19.63 still works as expected. 

This is my list of config options to try toggling:
```
CONFIG_IO_URING is default yes
CONFIG_VOP_BUS is default no. The help text indicates this is usually for intel mics, but I think we use things on the VOP bus?
ROCKCHIP_RGB is default no, unlikely suspect but of some note
SND_SOC_CROS_EC_CODEC is default no, but I've been selecting yes
CROS_EC_LIGHTBAR is default yes since the is a chromeos system, I've been saying no since it this isn't a pixel
CROS_EC_VBC is default yes, this is a suspect
CROS_EC_DEBUGFS is default yes, this is a suspect
CROS_EC_SYSFS is default yes, this is a suspect
RAID6_PQ_BENCHMARK is default yes, this is a suspect
PRINTK_CALLER is default no, I've been selecting yes to help with debugging
```

Finally, it may be worth trying without some of the PrawnOS patches to see if one of them are at fault.

SolidHal, 2019-08-04:
With the previously posted config options, 5.1.21 boot properly. 

With these config options, 5.2 does not boot. 

```
DRM_PANFROST is default no, left default no
NVMEM_SYSFS is default yes, left default yes
OPTIMIZE_INLINING is default no, left default no
DEBUG_MISC is default yes, left yes
```

SolidHal, 2019-08-05:
Removing all of the PrawnOS patches did not fix the issue with 5.2 either. 

So, some change was made between the last 5.1 release and the first 5.2 release that breaks booting on the c201.

bkeys, 2019-08-25:
Has there been any updates on this since the last comments?

gregordinary, 2019-08-25:
@bkeys I've tried with each minor revision to 5.2 through the 5.2.9 linux-libre release and same issue persists. Also gave last week's 5.3 RC (non-libre) kernel a go and same behavior there as well. 

There was definitely a lot of work done for rk3288 devices for kernel 5.2 (5.3 as well) noted here. https://www.spinics.net/lists/arm-kernel/msg723289.html

I'm on mobile and can't find the thread but I did see  a comment on one of the 5.2 patch threads about some setting causing devices not to boot and I think it was a patch related to power settings. Will try and find that when I get to my destination later tonight.

bkeys, 2019-08-25:
@gregordinary Alright let me know, I was thinking of trying to get PrawnOS to work on 5.2 but it's encouraging to see others have already gotten this far down the road.

SolidHal, 2019-08-25:
@gregordinary if you can find that setting be sure to let me know. Since I got the ath9k issue patched and have reports of it being reliable I'll be focusing on 5.2 support to get gpu and crypto support.

SolidHal, 2019-08-26:
Checked out that list of patches from the 5 mails on the arm-kernel mailing list. 

Tried a build with the following reverted since they related to power management:
https://github.com/torvalds/linux/commit/fbd7af041586a15f5dc15b9ec051d25aa5ebe00a
https://github.com/torvalds/linux/commit/2dd00d31d488707d0c60088aeb41b6bc2450a7b2
https://github.com/torvalds/linux/commit/bbdc00a7de24cc90315b1775fb74841373fe12f7

With no progress. 

Tried all of the dts commits from both pull request round 2 and round 1
Round 2
https://github.com/torvalds/linux/commit/864c2fee4ee93f53a8efed206c01ebce546df4e9
https://github.com/torvalds/linux/commit/280fa349757bb240c650882feee5a861150ccc2d
https://github.com/torvalds/linux/commit/2f60eb2f03b9c3d0a31592c55a88ef62b1403b5d
https://github.com/torvalds/linux/commit/356150e86d75653d1f679c6ef583144b26d0a686
https://github.com/torvalds/linux/commit/8a5deb4e316a5469f137e62eb1bbf6147ddbfd3d
https://github.com/torvalds/linux/commit/ed27ae71bf610004d04cb956d0a7687342006a1f
https://github.com/torvalds/linux/commit/ac60c5e33df4ec2b69c7e3ebbc0ccf1557e7bd5e
https://github.com/torvalds/linux/commit/4db11c378ab1e170c3a197ea3719ffe54cd06637
Round 1
https://github.com/torvalds/linux/commit/494da92d56e45c88966fab4db2bed1a2f300c5f8
https://github.com/torvalds/linux/commit/a2b2012eab25aad00a7dd587d1754115491e59eb
https://github.com/torvalds/linux/commit/864c2fee4ee93f53a8efed206c01ebce546df4e9

Aaaand got nowhere. 

Tried the first 3 power management patches along with rounds 1 and 2, which didn't help either. 

Found some clk changes, and reverted all of these on top of what I had already reverted

https://github.com/torvalds/linux/commit/5834fd75e6236605da8c439a64eaa33f3c8d02fe
https://github.com/torvalds/linux/commit/1f55660ff805224b4a532cfd5650753f0ed8dbeb
https://github.com/torvalds/linux/commit/dfe7fb21cd9e730230d55a79bc72cf2ece67cdd5
https://github.com/torvalds/linux/commit/00c0cd9e59d265b393553e9afa54fee8b10e8158
https://github.com/torvalds/linux/commit/bf297420cce5c6c6de477d27271d9f4645633dad
https://github.com/torvalds/linux/commit/57a20248ef3e429dc822f0774bc4e00136c46c83
https://github.com/torvalds/linux/commit/f4033db5b84ebe4b32c25ba2ed65ab20b628996a
https://github.com/torvalds/linux/commit/62e59c4e69b3cdbad67e3c2d49e4df4cfe1679e3
https://github.com/torvalds/linux/commit/5834fd75e6236605da8c439a64eaa33f3c8d02fe

Which also didn't work.

SolidHal, 2019-08-26:
Further testing, 5.2rc1 has this issue so it's an early change that caused this

bkeys-RFA, 2019-09-06:
Any updates on this?

SolidHal, 2019-09-06:
Still working to find the exact config option that breaks everything. 
Like searching for a needle in a haystack unfortunately.

SolidHal, 2019-09-12:
All done, most recent release has 5.2

closed 2019-09-12 by SolidHal