SolidEva/PrawnOS #111: Current kernel 5.2.9 fails to load firmware for internal wireless card

tslilc, 2019-09-11:
@SolidHal , this may be a bit premature but i saw that you merged the 5.2 testing with the master branch so i compiled it myself to see Panfrost. Thanks for the brilliant work!

Unfortunately it would seem that the _internal_ wireless doesn't work any more. `dmesg` shows

    usb 3-1: Direct firmware load for htc_9271.fw failed with error -2

Plugging in things to ports seems to work just fine. I'm not convinced there's anything else relevant in the logs, but perhaps i overlooked something. What other information would be useful?

SolidHal, 2019-09-11:
Yeah, I messed up. Missed building the firmware into the kernel. 
Testing the change now, and will push in 20 min assuming all goes well. 

Just an FYI about panfrost, the Mesa userspace driver isn’t built into prawnos since Mesa version 19.2 isn’t packaged by Debian yet, so you’ll have to build that.

I’m playing around with cross compiling Mesa during the PrawnOS build, but Debian may get it packaged before I get something working reliably.

tslilc, 2019-09-11:
Thanks! I'll play around with building mesa. And thank you again for the tireless work on this!

closed 2019-09-11 by tslilc