SolidEva/PrawnOS #117: csr8510 disable_ertm by default

riwale, 2019-09-23:
As suggested, i bought a couple of bluetooth csr8510 dongles.

They only work as expected when power management is turned off. PAN / A2DP and simple HID devices connect, then disconnect again without the module paramater set.

I have not found an option to disable it by default in the kernel, most guides recommend sysfutils to disable it on reboot.

You can add a file to /etc/modprobe.d though and disable it:

sudo echo "options bluetooth disable\_ertm=Y" > /etc/modprobe.d/bluetooth.conf

HID, A2DP and BNEP/PAN works fine after that.

SolidHal, 2019-10-09:
I'll have a new built image with this fix uploaded in a couple days, if you have a chance you can test that out. Will be Alpha 16

rk-zero, 2019-10-31:
I have tested the new build. You need to add the user to the bluetooth group also, this is unfortunately not done automatically after installing blueman / bluetoothctl.

Those packages are not really taking up much space and i guess if you advocate bluetooth dongles  csr8510 this could be done during install of xfce to make things easier.

rk-zero, 2019-10-31:
So, bluetooth works in most cases after fixing groups. But ertm is still enabled by default. To confirm:

cat /sys/module/bluetooth/parameters/disable_ertm 
N

in later releases.

I guess the modprobe way isn't working, sysfsutils seems to be needed to have it disabled across reboots.

apt install sysfsutils

echo "module/bluetooth/parameters/disable_ertm = 1" > /etc/sysfs.conf

afterwards:

cat /sys/module/bluetooth/parameters/disable_ertm 
Y

I know this is not needed by most users, but i like to cater to every use-case.

riwale, 2019-11-01:
@rk-zero both tips work, will close this issue

SolidHal, 2019-11-19:
Reopened so that the fix can be put into PrawnOS. Thanks for the testing and additional information @rk-zero. I'll add the user to the bluetooth group and fix how `disable_ertm` gets set. I think I'm going to hold off on installing any specific bluetooth software (blueman/bluetoothctl) until I can thoroughly test it myself.

SolidHal, 2019-12-03:
Applied new fix in 6bdd2abbaffddae6fc7242a62527d5f9d8c5f210, closing

closed 2019-12-03 by SolidHal