SolidEva/PrawnOS #121: Install with root encryption fails, drops to shell

tehbra1n, 2019-10-07:
Starting a clean install with alpha 15, I chose to use root encryption on the internal drive.  On the first attempt it appeared to be proceeding with encryption but rebooted, booted from the internal emmc and dropped to a busybox shell.  

I bailed and re-installed skipping encryption and the install worked fine.  I attempted a 3rd clean install, choosing encryption again.  This time it rebooted almost immediately after choosing, booted from the internal drive and dropped to the busybox shell again.  Here is a pic of the screen after the 3rd attempt.

[image]

SolidHal, 2019-10-09:
@tehbra1n thanks for bringing this to my attention. 

Did you happen to skip the "YES" prompt when cryptsetup makes the encrypted partition? If you accidentally type enter or anything else besides capital "YES" at this prompt the script caries on without making the encrypted partition, but tells the init to use encryption. 

Only other thing I can think of is if you incorrectly entered the password to unlock the partition 3 times right after setting up the password, the script will carry on and tell the init to use encryption.
I'm pushing a fix for both of those now.

tehbra1n, 2019-10-09:
@SolidHal  I'd be pretty confident that I assumed case insensitivity.  I pulled the latest updates from git and built the image.  the new method of choosing 1 or 2 worked great and the install was fine.

thanks!

closed 2019-10-10 by tehbra1n

SolidHal, 2019-10-10:
@tehbra1n I’m not saying you typed it in wrong, rather if you happened to press any key between the y/n prompt and the second prompt, the second prompt would take that buffered key as input, fail, and then do the wrong thing despite the failure. 
Really wasn’t a great system hahaha