SolidEva/PrawnOS #123: XKB Support

rk-zero, 2019-10-09:
I stole the gallium xkb layouts from here https://github.com/optio50/ChromeBook-Keyboard-xkb and applied changes through patchfiles instead of overwriting everything.

I'm sadly unable to make a branch and cherrypick the changes for a simple pull request, so please see my commit https://github.com/rk-zero/PrawnOS/commit/6cdba6aeac9a823fc03bbabbe7e8de49c8d25b6b

It adds all chromebook layouts as system keyboard layouts so you don't have to use xmodmap and other workarounds.

in lxde, xmodmap just won't work, with this patch the layout is correct in all DEs, tested xfce and lxde so far.

SolidHal, 2019-10-30:
commits a099220fc6caf5ac302211a32598ed3ecfd8aad9 and 716306ea991c712e65b15e406ec9e244e7875b14 implement xkb. Everything tests functional, closing.

closed 2019-10-30 by SolidHal

rk-zero, 2020-04-03:
i updated the xkb files to include a (somewhat) german layout mapping for gb and us layouts.
i noticed the files in prawn durrently miss support for the c100p volume rocker (VLR+/- instead of VOL) and the "chromebook" file in keycodes - if your arrow keys dont work - that may be the issue.

ill test my patches further, then try my hands at a PR again.

SolidHal, 2020-04-03:
Arrow keys seem perfectly funcitonal on my devices. Looking forward to your PR with c100p support!