SolidEva/PrawnOS #130: Emmc discussion

SolidHal, 2019-10-18:
This is a great idea, still the eMMC has so much issues and is only double the speed of the SD, even though 5-6 times faster than USB and lots more iops.

Can we talk about issues from the eMMC in another issue/channel? I saw you signed off some kernel patches considering this and i'm hitting roadblocks trying to actually zero out the device for correct partitioning.

nvm, i'll figure it out.

Setting up a lvm and cache writes to the emmc makes sense, but the sd reader is not usb bottlenecked, you can get 50-60MBs out of it which seems fine for everyday use.

_Originally posted by @rk-zero in https://github.com/SolidHal/PrawnOS/issues/43#issuecomment-542304425_

Created this issue to brain dump about the emmc and its.. weird partitioning issues

SolidHal, 2019-10-18:
@rk-zero:

The emmc in the c201 has some dead blocks, located at the beginning of the device. That is why partitioning tools complain about not being able detect the primary partition map. 

This requires a custom partition map, starting the kernel partition after a gap. Figuring out the correct partition map was a large part of early PrawnOS development, as it turns out not all of the c201s seem to have the same amount of dead blocks. This is why some c201s have no issue with a "naive" method of writing to the first blocks. I found the mapping that worked on all the c201s I have, which can be found in `resources/InstallResources/mmc.partmap`. Later, I found some c201s have a smaller emmc, which is why `mmc_type2.partmap` exists. 

One of the kernel patches you mention is to make the kernel not care about using the backup partition map instead of the primary one. Most of the other emmc patches are for stability. I would consider the emmc with PrawnOS to be stable, as I haven't encountered any issues since adding the patches. 

Finally, just because we can make the kernel not complain about a broken primary partition map doesn't mean we don't encounter some side effects. For example, tools will always complain about not being able to write to certain blocks, causing useless input/output errors. This is annoying, but doesn't effect functionality.

SolidHal, 2019-10-18:
I'd like to do some formal measurements of emmc vs sd vs usb read/write speeds. If the sd has comparable speeds to the emmc, I would likely switch to suggesting the usage of an sd card as the main boot device. 
This would remove the storage size issue this machine has, and have the added benefit of being able to easily remove and/or destroy all of the data if the user needs to.

SolidHal, 2019-10-18:
Finally, @rk-zero, I tried to respond to your irc messages but it seems like you were offline at the time. If you end up picking up those ar9271 chips let me know how installation goes!

duvh93, 2019-10-18:
Has anyone attempted to change the soldered eMMC?

I have [checked](https://www.nckforum.com/showthread.php?t=543) that, for the part number mounted on my motherboards (2GB RAM models), there are substitutions up to 64GB.

rk-zero, 2019-10-18:
Thanks for opening this discussion.

I have booted chromeos and tried gcpt there. No errors, but after reboot or even hard reset the emmc returns to first headers ignored. Tried keeping the emmc busy on another part while rebuilding gpt, same issue, same blocks.

I guess that is just the way it is for now.

i benchmarked the internal emmc and a high class Samsung 512GB SD card to max out the card reader.

hdparm -t shows around 120MB/sec for emmc and 60MB/sec for sd card.

But for everyday use random read/write is far more interesting.

fio --randrepeat=1 --ioengine=libaio --direct=1 --gtod_reduce=1 --name=test --filename=test --bs=4k --iodepth=64 --size=4G --readwrite=randread

shows about 15-16MB/sec and 3800 (SD) to 4000(eMMC). It didn't matter wether the emmc was encrypted or not.

So the difference is actually neglectable.

ssd-cache with lvm and/or encryption enabled may show a difference, will test that for fun if bored, but maybe the overhead is not worth it.

I will just use the internal emmc for test-purposes/testboots of fresher kernels and switch to the sd as main boot in all my devices for now to care about other topics. User won't really see a difference.

SolidHal, 2019-10-18:
> Has anyone attempted to change the soldered eMMC?
> 
> I have [checked](https://www.nckforum.com/showthread.php?t=543) that, for the part number mounted on my motherboards (2GB RAM models), there are substitutions up to 64GB.

I've discussed the idea with a few of the libreboot developers. The emmc is likely epoxied down. This doesn't make replacement/upgrade impossible but much more difficult.

SolidHal, 2019-10-18:
> Thanks for opening this discussion.
> 
> I have booted chromeos and tried cgpt there. No errors, but after reboot or even hard reset the emmc returns to first headers ignored. Tried keeping the emmc busy on another part while rebuilding gpt, same issue, same blocks.
> 
> I guess that is just the way it is for now.
> 
> i benchmarked the internal emmc and a high class Samsung 512GB SD card to max out the card reader.
> 
> hdparm -t shows around 120MB/sec for emmc and 60MB/sec for sd card.
> 
> But for everyday use random read/write is far more interesting.
> 
> fio --randrepeat=1 --ioengine=libaio --direct=1 --gtod_reduce=1 --name=test --filename=test --bs=4k --iodepth=64 --size=4G --readwrite=randread
> 
> sorry, got it all wrong!
> 
> SD card unencrypted: around 15MB/sec random read, 3700 iops avg.
> eMMC unencrypted 2GB ram model: around 16MB/sec random read, 3800 iops avg.
> eMMC unencrypted 4GB ram model: **around 40MB/sec random read, 9500 iops avg.**
> eMMC encrypted 4GB ram model: around 16MB/sec random read, 3800 iops avg.
> 
> Direct writes (no file caching): fio --rw=write --name=test --size=4GB --direct=1
> 
> 2GB ram model emmc: 5MB/sec
> 4GB ram model emmc: 15MB/sec
> 
> Direct reads (no file caching) fio --rw=read --name=test --size=4GB --direct=1 --filename=test
> 
> 2GB ram models emmc unencrypted: 8MB/sec (man ... the pain ...)
> 4GB ram models emmc unencrypted: 40MB/sec
> SD card unencrypted: 16MB/sec
> 
> So no real difference sequecial read / random read.

Thanks for doing some testing. I'm a little confused by some of your results though. The difference of 2GB vs 4GB of ram shouldn't impact the speed of reads/writes that much. I have a 2GB model and 4GB model so I'll try and recreate your results. 
Maybe the two models you have contain different emmcs? Even then though....

SolidHal, 2019-10-19:
Note: When using cgpt to create new partitions or a new partition map on the c201 you _must_ run partprobe afterwords to map the /dev/(devices). fdisk,sfdisk,etc all do this for you but cgpt doesn't

rk-zero, 2019-10-20:
There is only partx (-u) available on chromeos which made no difference. I will read up on the issue before attempting more "fixes".

@duvh93 I have a spare c201p open, there is no epoxy around the emmc as in smartphones, but i have next to zero soldering experience and only one broken device i would attempt this on.

SolidHal, 2019-10-20:
@rk-zero @duvh93 

> I have a spare c201p open, there is no epoxy around the emmc as in smartphones, but i have next to zero soldering experience and only one broken device i would attempt this on.
I have been told that it could be epoxied underneath, meaning the only way to tell would be to peel of the emmc on a board you don't care about. If it is epoxied, removal may prove too destructive to the pads and traces under the BGA solder balls making replacement difficult.

SolidHal, 2019-10-21:
I opened #133 for emmc and sd benchmarking

JeremyRand, 2019-10-23:
> If the sd has comparable speeds to the emmc, I would likely switch to suggesting the usage of an sd card as the main boot device.

@SolidHal Note that the internal eMMC probably has some security advantages (e.g. regarding evil maid attacks), so the recommendations should probably tell the user to take that into account, especially if the user doesn't need more storage than the internal eMMC has.

SolidHal, 2019-10-24:
@JeremyRand

Sorry for the wall of text, I’ve been meaning to discuss the security of the c201 in a little depth so I ramble off topic a bit. I apologize if the formatting is off, I’ll fix it next chance I have at a computer. 

The sd card is arguably more secure than the emmc vs an evil-maid attack on the boot device because it is removable. 
The threats I see from an evil-maid attack are:
- Data theft: the attacker copies or outright steals your storage device. 

- Modification of the kernel image:
The attacker builds a custom malicious PrawnOS kernel image and flashes it onto your c201. You could imagine the malicious kernel could do many things, like wait for you to decrypt your root and then steal your password/data. Really, unless we verify the kernel partition in the bootloader/bios this will always be possible. Luckily, we can eventually add support to do this on the c201 with libreboot and depthcharge. Even with bootloader kernel verification, an attacker could modify the bios image with long enough physical access. But, that is kinda out of scope for what we are talking about. 

Both of these attacks require access to the storage device you use to boot. 

Data theft is relatively simple if you use the emmc or an sd card to boot. 
Simply pop out the sd card and copy it, or  the attacker can boot the laptop with their own usb device and copy over everything on the emmc or and SD card. Encryption of course helps mitigate this threat, but you can imagine a well determined and funded attacker could copy your encrypted data and then crack the encryption at their leisure.

Modification of the kernel image is harder, since it requires knowledge of the kernel to make a malicious one. Once the attacker has a malicious kernel, execution of the attack is similar to data theft. 

Since the main threat requires access to the device you use to boot, either the internal emmc or the sd card, the obvious way to prevent such attacks is to keep your boot is to prevent physical access. 

It’s much easier to always keep a micro-sd on your person or locked up in a safe, rather than the entire laptop. 

While not really related to the topic of evil-maid attacks on storage devices, the sd card is also easier to wipe/dispose of than the emmc if the need arises. 

Of course, you could imagine many other evil-maid style attacks targeting other components of the device that would let t an attacker gain access to your data. Examples could be modifying the wireless device or connecting a device to the data liners connecting the cpu, emmc, and sd card. 

A proper organized discussion of potential threats to the security of PrawnOS and the c201 and what hopes we have of mitigating them is something that would be beneficial in the future.

Since PrawnOS has a focus on privacy, it makes sense a desire for security follows. While we have removed obvious threats like the Intel Management Engine, closed source WiFi firmware, and closed source software from boot to userspace unfortunately none of that guarantees security, only increases trustworthiness and verifiability.

Note: everything I’ve said about sd cards is of course also true for usb drives.

SolidHal, 2020-02-26:
see #133 for sd vs emmc read write performance

SolidHal, 2020-02-26:
closing. Feel free to reopen for additional discussion

closed 2020-02-26 by SolidHal

rk-zero, 2020-04-04:
I sent one c201p and some c100p to maintainers of libreboot. The errors in the first sectors are due to locking of the emmc in coreboot, so changing the emmc wont have any effect.
If a better libreboot comes along those errors will be gone, so further tries to fix it in userspace wont really make sense.

SolidHal, 2020-04-04:
I figured. I imagine coreboot did it for a reason though. Likely a hack to get around issues with the emmc hardware or how the arm cpu expects to interface with the emmc. Thanks for sending them machines! I'm working on getting my hands on a c100p myself.