SolidEva/PrawnOS #133: EMMC vs SD read/write performance tests

SolidHal, 2019-10-21:
Creating this to document testing read/write on the internal emmc and sd card with `fio`, `dd` and `hdparm`

SolidHal, 2019-10-21:
DD read/write test on the emmc
```
prawn@PrawnOS-Alpha:~$ sudo ./benchmarkBootDevice.py 
[sudo] password for prawn: 
running dd benchmarks, this may take a while.....
################## dd write 2GB bs=50M avg results #######################
write = 43.379999999999995, read = 859.2, cached read = 823.2

################## dd write 2GB bs=1M avg results #########################
write = 44.99999999999999, read = 910.0, cached read = 859.8

################## dd write 2GB bs=512 bytes avg results ##################
write = 43.56, read = 195.0, cached read = 193.6

################## dd write 4GB bs=50M avg results #######################
write = 45.14, read = 167.2, cached read = 156.6

################## dd write 4GB bs=1M avg results #########################
write = 44.42, read = 148.0, cached read = 139.4

################## dd write 4GB bs=512 bytes avg results ##################
write = 43.81999999999999, read = 136.8, cached read = 133.6

```

rk-zero, 2019-10-23:
I installed all c201p with a fresh alpha 19. Can you share the script you used? I get 3800 IOPS now on the 2GB model and 4200 IOPS on a 4GB. dd write from /dev/zero is 41,5MB/sec for the 2GB model and 45,2MB/sec for the 4GB.

So, not that big of a difference after all.

SolidHal, 2019-10-23:
Yeah, im almost done with a full benchmarking script that uses dd and fio and handles sequential, random, read, write and read/write (“realistic mode”).

It outputs in CSV format so we can import the data an make some nice graphs. 

It’ll be in the PrawnOS repo. I’ll post here once I finish testing it.

SolidHal, 2019-11-19:
I have full benchmarks for the emmc, just trying to get them for an sd card. Have had two runs killed by read/write errors. Trying a new sd card.

SolidHal, 2019-12-08:
After a loooong wait, the benchmark results for the SD card and emmc on the c201 are complete. 
To benchmark, I used a script I wrote that runs fio and dd with many different block sizes and file sizes, running each test multiple times and taking an average to reduce error.
The script is available here https://github.com/SolidHal/Boot-Device-Benchmarking

Setup: asus c201 4GB model running PrawnOS with luks encrypted root partition
the sd card I used is a sandisk extreme U3 32GB microsd, which has a minimum sequential write speed of 30MB/second

Using fio, across the board for all reads/writes the internal emmc performs almost 2 times as well as the SD card and sometimes 3 to 4 times better. 
The most interesting fio test is the read/write random workloads. This mimics average usage by doing 90% reads and 10% writes randomly across the storage device. The internal emmc did 2 times as well as the sd card for each block size and file size in this test. 

The dd results are closer and inconclusive, with the sd card and emmc performing just about the same in each category.
Because dd does only sequential reads/writes, is single threaded, and is not designed for benchmarking storage devices like fio the dd results are flawed.

The full results can be found here: https://github.com/SolidHal/Boot-Device-Benchmarking/tree/master/c201_results

Conclusion: the internal emmc is much better suited as a boot and main storage device. If we need additional storage space, It would make sense to support using and encrypting an sd card as extra user storage.

SolidHal, 2020-02-26:
Closing, conclusion has been reached.

closed 2020-02-26 by SolidHal