SolidEva/PrawnOS #135: running CompilePanfrost.sh on alpha ver 20 doesn't result in Panfrost driver being used

tehbra1n, 2019-10-31:
On a fresh install of alpha ver 20 on the internal emmc, w/encryption enabled and Xfce installed, after running CompilePanfrost.sh and rebooting, I still show that mesa is using the VMWare LLVM driver.

CompilePanfrost.sh output seems to show the process as I would expect to see (clone the mesa repo, run meson, run ninja install) and looks similar as to when I do a Panfrost install by hand following instructions in #99.  Confusingly, running the commands from #99 by hand on a fresh alpha ver 20 install also results in the VMWare LLVM driver being used.  glxinfo -B shows that the mesa version is 18.3.6.  Manually cloning and installing the mesa repository with -b master seems to give version 20.  Cloning with -b "19.2" or "19.3" doesn't result in any improvement.

I don't know enough about how the system selects the driver to do much more myself.  Here is the relevant bits from dmesg:

```
[    1.384068] panfrost ffa30000.gpu: clock rate = 400000000
[    1.390126] panfrost ffa30000.gpu: failed to get regulator: -517
[    1.396847] panfrost ffa30000.gpu: regulator init failed -517
[    3.649972] panfrost ffa30000.gpu: clock rate = 400000000
[    3.662034] panfrost ffa30000.gpu: mali-t760 id 0x750 major 0x0 minor 0x0 status 0x1
[    3.675991] panfrost ffa30000.gpu: features: 00000000,100277bf, issues: 00000000,24040400
[    3.675996] panfrost ffa30000.gpu: Features: L2:0x07120206 Shader:0x00000000 Tiler:0x00000809 Mem:0x1 MMU:0x00002830 AS:0xff JS:0x7
[    3.675999] panfrost ffa30000.gpu: shader_present=0xf l2_present=0x1
[    3.677046] [drm] Initialized panfrost 1.0.0 20180908 for ffa30000.gpu on minor 3
```

SolidHal, 2019-11-19:
I'm not really sure how the system selects a driver, but can confirm that when I run the script it doesn't replace llvm. I'm poking around to see if I can figure it out, and I asked on the panfrost irc.

SolidHal, 2019-12-03:
I have an updated version of CompilePanfrost.sh that I have been testing, however I'm running into stability issues with everything I've tested so far. Will update when I find a stable setup.

SolidHal, 2019-12-03:
The newest version of CompilePanfrost.sh will install a stable version of mesa now, and will result in the system using panfrost. There is still likely a memory leak in mesa along with other issues both with mesa and panfrost, but the system will run for a while without any issues. Closing

closed 2019-12-03 by SolidHal