SolidEva/PrawnOS #137: curl ssl/tls CA certificate issue

SolidHal, 2019-12-20:
On a fresh install of an old build all ssl requests I would try to make using curl would fail
with the error talking about being unable to verify the CA root certificate. I'm not sure why this would happen, I suspect its an issue with the older build? 
The fix is to reload the symlinks to the ssl certificates by running 
```
update-ca-certificates --fresh
```
I'll test if this still happens on more recent builds.

ffff135, 2019-12-24:
Same issue on the latest release, "v0.20-alpha". Neither curl nor youtube-dl would work without the options --insecure or --no-check-certificate respectively. Reloading the symlinks fixed it.

SolidHal, 2019-12-25:
I’m running a new build, I’ll see if it still happens there.

SolidHal, 2019-12-29:
Still happens on a new build, seems to be a but with debootstrap? I'm adding a line to the image building process to update the certificates during deboot

commit 7bf6c06f (2019-12-30): reload the root CA certificates after package installation. Fixes #137
closed 2019-12-30 by SolidHal