SolidEva/PrawnOS #141: uncomplicated firewall does not work.

Anthony-Sensors, 2020-01-12:
sudo ufw enable
ERROR: problem running ufw-init
modprobe: FATAL: Module nf_conntrack_netbios_ns not found in directory /lib/modules/5.2.21-gnu
iptables-restore: line 6 failed
iptables-restore: line 22 failed
iptables-restore v1.8.2 (nf_tables): Couldn't load match `conntrack':No such file or directory

Error occurred at line: 2
Try `iptables-restore -h' or 'iptables-restore --help' for more information.
iptables-restore: line 4 failed
iptables-restore: line 5 failed
iptables-restore: line 21 failed
iptables-restore v1.8.2 (nf_tables): unknown option "--dport"
Error occurred at line: 19
Try `iptables-restore -h' or 'iptables-restore --help' for more information.
iptables-restore: line 10 failed
iptables-restore v1.8.2 (nf_tables): Couldn't load match `limit':No such file or directory

Error occurred at line: 8
Try `iptables-restore -h' or 'iptables-restore --help' for more information.
iptables-restore v1.8.2 (nf_tables): Chain 'ufw-before-input' does not exist

Problem loading ipv6 (skipping)
Problem running '/etc/ufw/before.rules'
Problem running '/etc/ufw/after.rules'
Problem running '/etc/ufw/user.rules'

SolidHal, 2020-01-19:
I'm enabling the  NF_CONNTRACK_NETBIOS_NS driver in the config. From the debug info above that should fix this problem.

SolidHal, 2020-11-30:
I *think* everything required for this is now available but don't know if I have tested properly. closing. please reopen if this issue persists.

closed 2020-11-30 by SolidHal