SolidEva/PrawnOS #143: android tools adb broken since buster

rk-zero, 2020-01-17:
I know this is more a debian issue, but i use this device as my main and only computer, so adb not working and quitting with key not found was a big issue for me.

To fix it, add stretch as a source in /etc/apt/sources.d/stretch.list

deb http://deb.debian.org/debian/ stretch main

then downgrade adb:

apt update && \
apt install adb=1:7.0.0+r33-1 android-libadb=1:7.0.0+r33-1 android-libbase=1:7.0.0+r33-1 android-liblog=1:7.0.0+r33-1 android-libcutils=1:7.0.0+r33-1

voila, adb works again.

SolidHal, 2020-01-19:
Thanks for providing instructions. I'll leave this issue open to serve as instructions until debian buster adb is functional again.

rk-zero, 2020-01-19:
To add to that, if you make a connection using the older version your adbkey will be created correctly and you can upgrade fine to the latest adb in buster.