SolidEva/PrawnOS #144: Chromebook is turning off randomly on charged battery usage. Please help

UralZima, 2020-01-19:
Hello. I have Asus C100P (veyron_minnie). I built PrawnOS Kernel and added dtb/dts for veyron_minnie, and it works. 
C201 and C100P are the same hardware.

The issue is, my chromebook is turning off on battery usage randomly. I did some tests to identify the issue. so:
1. It is turning off on youtube playback, 5 min or 30 min, it is guaranteed to turn off on battery usage. 
2. It is not battery. I wanted to check it and discharge it fully. I turn on my chromebook and did not enter luks password - it is working until full discharge. But. If the battery level <50%, after entering password it is turning off immediately after KMS (kernel mode switching). After fonts are changed in console, it turns off.
3. It is not the kernel. Archlinux did the same
4. Original chromeos works perfectly. 
5. I didn't flash libreboot, I am afraid to brick it, because no instructions for C100P, only c201... I am in dev mode
6. I installed KDE (just for fun and test, don't say anything about it) and observed such thing: battery level <60%, I disconnect the charder, kde icon still shows charging for about 2-4 min. Immediately after the icon changes to 'battery use'. it turns off.

So, it may be video-related, some kernel voltage guard related, or something like this. 
Please help me resolve this issue.

SolidHal, 2020-01-19:
Definitely a curious problem. Not something I have seen on any of my c201s. I would start by looking at what power management drivers are in use by the kernel and what userspace power management options are enabled. I believe power management is handled first through the chromebook EC and then by the kernel drivers, so it is possible the two are miscommunicating. In xfce Power manager, there is a devices tab that should list your battery and provide some information about it. Could you post a screen shot of that information?

rk-zero, 2020-01-19:
c100p is still way different from c201p. If you managed to boot prawnOS let us know about your progress. I couldn't boot the prawnOS kernel ootb.

I doubt the system is booting from the emmc.

librebooting the c100p from the c201p scripts WILL BRICK YOUR c100p.

rk-zero, 2020-01-19:
I have c100p for use laying around, if you can supply build instructions i will happily try the kernel on a non-librebooted c100p myself.

SolidHal, 2020-01-19:
If you're booting from usb, maybe usb power management is kicking in?

UralZima, 2020-01-20:
Hi everyone. Thank you for answers.
1. I am running from emmc
2. To run PrawnOS on C100P just add dtb to kernel.its. Copy and paste fdt section, and change dtb path to arch/arm/boot/dts/rk3288-veyron-minnie.dtb. Thats all needed to run PrawnOS on C100P

Everything else I'll test later and will report here. The battery is 100% ok, because I discharged it to 1% on luks prompt, and never discharged it <50% after full init, it always turned off itself. 
Archlinux kernel do the same. 

P.S. I built PrawnOS based on linus torvald's kernel (just modified wget source), but including all patches, I don't have atheros stick still and using brcmfmac. But unloading this driver from kernel also caused c100p to turn off.

rk-zero, 2020-01-20:
Thanks for providing instructions @UralZima - i compiled with the modified fdt section, it wouldn't allow 2 ftd sections, though.

I used a c100p that was working with chromeos very fine with no problems and installed to internal emmc. The expand install from SD fails though (no idea why, mmcblk0/2 are the same in /dev)

Book boots PrawnOS fine, wifi is working fine.
Battery shows 32.2 Wh, 38 on c201p.

Maybe the batterz detection is wonky? It shows 129% on my c201p fully charged.

PrawnOS on the c100p with touchscreen ... ITS SO BEAUTIFUL.

Will report back, battery rn is at 77%.

UralZima, 2020-01-21:
@rk-zero, kernel.its allow many fdt sections, just need to define the default, and name them fdt@1, fdt@2, etc. I compiled a kernel that run both c201 and c100p. Look at arch's kernel.its, it contains all arm dtb.

I am using emmc only, because usb or sd after waking up from sleep - loose the filesystem access and say i/o error. same on arch. 

I tried the cpu benchmarks with 'while true; do 7zr b; done'. I disabled sddm, boot into console and run the tests without xorg and left fully charged c100p for the whole night. Morning it was off, the battery level was at 25%. Better results, but anyway it turned off. It worked many hours with full cpu load.
Now I am running same tests with AC plugged in for hours - it works, so it is not soc overheat problem.  Any ideas? 

@rk-zero, can you please run video playback test? Open a youtube video in hd or fullhd (force), on charged battery use and see how it run until full discharge. 

P.S. Useful scripts http://kmkeen.com/c100p-tweaks/  
But in PrawnOS the c100p.tablet-mode program fails with segmentation fault, but on archlinux works (it detects the angle of chromebook and disables keyboard in tablet mode)

rk-zero, 2020-01-21:
@UralZima Thanks for the additional help, we should sureley enable kernel building for both (@SolidHal ahem).

I kept the book running over night and played some youtube videos in the evening. The book was off this morning with depleted battery. syslog shows nothing unusual until 6 in the morning, then i guess the book just shut down. I don't know where to look if it shutted down planned because of low battery, but the book started shortly, so some charge was left. Charged to 10%, plays youtube videos fine and runs 7zr benchmark in another window.

I will try the scripts and play with the book some more, maybe it will crash within the next days, but so far ... works great. Scrolling with 2 fingers doesn't work f.e. I really like his introduction about the book, i'm likewise in love with the c201p and c100p lol. It's just a great device and plus it's 100% free. So great. Love it everyday.

My flipbook unfortunately only has 2GB of RAM which means the ssd may differ too. Other than that the 30/38wh difference shows usually around 10h remaining on my c201p with usual work, but only 6 on the c100p.

rk-zero, 2020-01-21:
http://kmkeen.com/c100p-tweaks/ is REALLY helpful.

I compiled ectool from coreboot to play around with the charging current as i bought a usb boost adapter last year so i can charge the book on traveling from a 20wh usb solar power bank. This may sound stupid, but i carry that anyway and can pack a small usb charger that way (-400g in the backpack matters). It didn't charge, maybe i can get it to work now and build a couple of smaller USB-Charging cables.

I guess you should start from the PrawnOS vanilla and not use the broadcom wifi. I have a couple or AR9271 and they work fine so far. I recommend the Version 1 of the TP-Link TL-WN722N which has best compromise in size / range. It will bend a little without an extension cord on a flat surface though.

Other than that, if you are located in the EU, i still have 6 c100p lying around with minor defects if you need parts, just write me an email r@k-ze.ro. Battery works in all of them, so maybe switch that?

SolidHal, 2020-02-26:
Closing as issue seems to be resolved. Feel free to reopen for additional discussion

closed 2020-02-26 by SolidHal