SolidEva/PrawnOS #152: Add support for upgrading kernel on a live system

filberg, 2020-04-25:
I've initially commented on #52 but I'm not sure if it was the right place, since the ticket is closed, so I'm bumping the discussion here (feel free to close this one if you prefer to re-open #52):

> I've been recently trying to come up with some ideas to automate the kernel upgrade and make it more convenient for a non-technical user.
> 
> These are the 2 basic approaches I've been reasoning on:
> 
> 1- Do The Right Thing and learn how to properly package a kernel deb to be hosted somewhere (I have some hosting available if needed) on a deb repo, and configure apt so that kernel upgrades use that specific channel.
Needless to say, this is for sure the best approach in terms of maintainability and possible "upstreamability", but it has a higher learning curve for me.
>
> 2- Reinvent the wheel and implement a custom upgrade process [...]
> Needless to say, this second approach is ugly and uses totally unorthodox methods, but it's the quickest one I was able to think of.

Just for having something to begin with, I've started working on the second approach. 
[Here's a script](https://github.com/filberg/PrawnOS/blob/master/scripts/InstallScripts/FlashKernelPart.sh) that upgrades the kernel on a live system, starting from a prebuilt vmlinux.kpart image.
This is far from being ready for a PR, as it's just a draft. It still needs to manually download the image, but I'll be working on the whole "check-for-updates/download-images/check-signature/flash-image" pipeline in the next days, so I guess something usable will be available soon.

In the meantime, I'd like some feedback and comments on the work done so far, so that we can maybe discuss how to improve it or even rewrite it altogether.

Can't wait to have this "live kernel upgrades" thing happen ;)

SolidHal, 2020-04-26:
I agree that having a proper package would be nice. I actually have dreams of providing an apt repo with the kernel, install scripts, and system config but have gotten stuck on finding a place to host it. I'll always accept PRs that improve on the current offering, and having some more-automated way to update the kernel is definitely an improvement :)

gdallasdye, 2020-04-30:
I've been giving this some thought as well. My main concern about having apt manage the kernel as it is, is that the scripts for the Debian kernel expect to output the compiled kernel to a file, in a folder, with all the luxuries of a filesystem. Adding support to those scripts to detect which device to use and hopefully writing to the write one seems like it could be problematic. And as for why, I wouldn't even know where to begin.  

I've been thinking that as a workaround to depthcharge's limitations, we could build the kernel with kexec support. I'm just thinking out loud for now, but I'm wondering if partition 1 is built with that, then maybe the initramfs could kexec to a partition dump file, say in /boot/c201/kernelver/vmlinux.dump. If stored in a luks partition, the user could have additional assurances the kernel they boot is they want. It could also serve as a workaround if a user's battery drains, and they cannot boot from external media, (and if they can't or won't enable usb boot), the internal kernel could kexec to a external kernel and userspace. I'm producing a build now with that kernel config option turned on (it's under boot in menuconfig) and also doing some light reading in preparation for that. I've got some ideas for test booting against different version numbers, kernel vendors, partition vs dd dump, and kernel module loading. I'll maybe need to massage the busybox and initramfs scripts though. 

Some more thinking aloud: Can GRUB2 run in userspace? Can a minimal kernel run grub, and then grub tells linux to kexec to a separate kernel and initramfs, much like a "normal linux" can? If grub can't run in userspace, can it run as a chromeos kernel, and then load linux?

rk-zero, 2020-04-30:
[arch does it already](https://github.com/archlinuxarm/PKGBUILDs/blob/master/core/linux-armv7/linux-armv7-chromebook.install)

(thanks to swiftgeek for the wisdom)

filberg, 2020-05-01:
> [arch does it already](https://github.com/archlinuxarm/PKGBUILDs/blob/master/core/linux-armv7/linux-armv7-chromebook.install)
> 
> (thanks to swiftgeek for the wisdom)

Yes, that's where I got the initial hint from (actually, I referred to the parabola [linux-libre-chromebook](https://www.parabola.nu/packages/libre/armv7h/linux-libre-chromebook/) package, that uses the same concept).

I also dropped the idea of doing the actual flashing in initramfs, as it's overly complicated; as per the parabola/arch rationale, the kernel image can be flashed directly on a live system.

filberg, 2020-05-01:
>  but have gotten stuck on finding a place to host it. 

I can provide you with FTP hosting space on my personal website, if that can help.
After all, you would still be able to sign official kernel releases with a trusted key of yours, so it shouldn't matter if upgrades are fetched from a "non-official" url.. 

> I'll always accept PRs that improve on the current offering, and having some more-automated way to update the kernel is definitely an improvement :)

A PR is coming soon, definitely.
I'm also working on a set of scripts to automate kernel builds whenever a new linux-libre version is released.
My intention is to package a PR including all those scripts, together with improved version of [the script I originally proposed](https://github.com/filberg/PrawnOS/blob/master/scripts/InstallScripts/FlashKernelPart.sh).

Stay tuned!

SolidHal, 2020-05-07:
Staying tuned! 

I've been rolling around the idea of automated testing/building and I think I have got a reasonable approach to the automated testing portion now that we can use crosystem/mosys to change the target boot device without user intervention, so additions to automated building/kernel updates fall in line with that. 

I think I'll start with getting an apt repo up on my personal site, and get a gauge of how much traffic it'll get. Then we can look into packaging other parts of the build, signing them, and uploading them.