SolidEva/PrawnOS #153: Wake from Sleep - results in EXT4-FS inode errors, system must be power cycled

cvmiller, 2020-04-27:
Using version 1.0.3 booted from USB stick on an Asus C201P

After closing the lid, counting to 10, reopening, noted that the X display often doesn't return, and there are repeating EXT4-FS error messages. The FS is disconnected, and no app can be run from the FS (e.g. ls, or dmesg).

Only recovery seems to be power cycle the Chromebook. When booted again, dmesg reports that the broken inodes have been fixed.

SolidHal, 2020-04-28:
This is likely due to usb power saving during sleep, and would occur when booted from an SD card as well. This doesn't happen when you are booting from the internal emmc. 
There might be a way to tweak what power mode it puts the usb ports in when the kernel enters sleep, so I'll leave this open

cvmiller, 2020-04-28:
Thanks for looking into if the Kernel can support this.

SolidHal, 2020-11-30:
I take back what I said about this also happening when booted from sd. It likely would *not* happen if booted from sd.

cvmiller, 2020-11-30:
I agree. I have been booting from SD, and I don't see the FS corruption.