SolidEva/PrawnOS #155: Kernel Header files not in /root/PrawnOS/... Unable to build kernel modules

cvmiller, 2020-05-12:
Bought a RTL822x2B Wifi Dongle. Found open source driver at:
https://github.com/RinCat/RTL88x2BU-Linux-Driver

But unable to build kernel modules, make fails with:
```
make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/5.4.23-gnu/build M=/home/cvmiller/Downloads/RTL88x2BU-Linux-Driver-master  modules
make[1]: *** /lib/modules/5.4.23-gnu/build: No such file or directory.  Stop.
```
When looking at the build directory it is a sym-link to:
```
lrwxrwxrwx 1 root root     32 Mar  2 22:06 build -> /root/PrawnOS/build/linux-5.4.23
```
Which doesn't exist. Since this kernel is not part of the Debian system, apt can't find kernel headers for it.

**Request:** Kernel headers for kernel being used in PrawnOS and please include support for the RTL88x2B in future versions of PrawnOS.

tehbra1n, 2020-05-12:
This isn't my area of expertise, but I'm not certain this driver works without a proprietary firmware blob.  I believe PrawnOS is avoiding firmware blobs since there is no real way to tell what that firmware enables.  I see [this thread](https://forums.puri.sm/t/a-route-to-better-wifi-performance-with-realtek-rtl88x2bu-chips/9225) discussing this WiFi card from a few days ago on a similar project.  The thought there seems to be that there is a firmware blob but it may or may not be required.

rk-zero, 2020-05-12:
you can get the header files while compiling the kernel yourself, just take the [config](https://github.com/SolidHal/PrawnOS/blob/master/resources/BuildResources/config) supplied by PrawnOS

https://unix.stackexchange.com/questions/270123/how-to-create-usr-src-linux-headers-version-files

plus one to have this supplied for stuff like xpadneo though.

cvmiller, 2020-05-13:
I appreciate the pure open source stance that PrawnOS has taken. However, since there is a custom kernel supplied, it would be nice to include the kernel header files included, so others can compile kernel modules without having to recreate the entire kernel.

Please add the kernel header files to this git repo,

rk-zero, 2020-05-13:
> I appreciate the pure open source stance that PrawnOS has taken

That is the scope of the project. I told you how to do your user case, but contributing to a more mainstream approach is possible here: https://github.com/SolidHal/PrawnOS/issues/152

You can buy a plethora of working wireless adapters that will work without hassle under 10$. Look into TP-Link TL-WN722N v1.

SolidHal, 2020-05-18:
@cvmiller Please remember that everyone here is just working on things they want to in their limited free time :) 

I am working on getting all parts of PrawnOS packaged as debs and hosted on a PrawnOS APT repo. I have the repo up and running, so now the work is re-doing all of the build process to build debs instead of just throwing the files in the right places. 
When I package the kernel, which will allow for the easy updates #152 refers to, I will be sure to package the headers separately so that everyone can access them. 

You will likely be able to run ```git clone``` and ```make kernel``` before I get to completing that. :D 
People have said they are able to do so just fine on their c201s running PrawnOS.

cvmiller, 2020-05-19:
@SolidHal Thanks for all that you have done, and are doing.

It will be great to have the parts of PrawnOS hosted as debs.

I pulled a 5.4.23 kernel, and have made it, but have had no luck building the wireless driver (RealTek), as the linker fails with all sorts of errors. So I look forward to your deb packages.

rk-zero, 2020-08-05:
Sorry for the late answer. I reinstalled Prawn today with latest 5.4.55 kernel from @alexisgrey fork.

After make image, go to build/linux-version folder.
Run make headers_install. The header files are now located in build/linux-version/usr/include.
copy that whole directory to your local /usr/src/linux-5.4.55-gnu folder (create it).

That way you can install dkms and have all software that expects the header files in /usr/src/linux-`uname -r` find the files.

cvmiller, 2021-02-16:
Sorry to be so long before I got back to this. I am running PrawnOS 1.3.1

I see what looks like kernel headers in /usr/src/linux-5.9.12-gnu/include 

When I attempt to compile a RTL88xx Wifi dongle driver I get the following error:
```
~/build/rtl8821ce$ make
make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/5.9.12-gnu-PrawnOS/build M=/home/cvmiller/build/rtl8821ce  modules
make[1]: *** /lib/modules/5.9.12-gnu-PrawnOS/build: No such file or directory.  Stop.
make: *** [Makefile:2244: modules] Error 2
```

Any pointers on how to proceed? TIA

rk-zero, 2021-02-17:
you can simply change the lib modules build path to the usr src one (omit include).
https://github.com/RinCat/RTL88x2BU-Linux-Driver/blob/afc917d82f3c4306e057e42b2b4708edc1f81839/Makefile#L1309

its hardcoded in the makefile of your driver.

What you have there is a path expected if you compiled the kernel from sources before but we only ship the modules for easy use of dkms stuff.
If you cannot make that work try to make a complete build of PrawnOS from source and take a look at the kernel build directory, it will have the expected directory structure.

rk-zero, 2021-02-17:
still, dont get me wrong, but a compatible wifi dongle is 7$

cvmiller, 2021-02-18:
Thanks for the pointrs @rk-zero , I "fixed" the KSRC line as per your suggestion, but I get a similar error: no rule to make target modules. I'll try your second suggestion of building PrawnOS.

I tried to get an inexpensive compatible Wifi adapter. I now have two incompatible wifi adapters, and so I am back to trying to compile modules again. One of my Wifi adapters has 5 Ghz (the RTL one) and I would prefer that as the 2.4 Ghz band is quite crowded where I live.

cvmiller, 2021-02-18:
It would appear that is not possible to build PrawnOS on PrawnOS :(
```
~/build/PrawnOS$ sudo make image TARGET=$ARCH
TARGET is armhf
make clean_image
make[1]: Entering directory '/home/cvmiller/build/PrawnOS'
TARGET is armhf
rm -f /home/cvmiller/build/PrawnOS/PrawnOS-Shiba-armhf.img
make[1]: Leaving directory '/home/cvmiller/build/PrawnOS'
make filesystem
make[1]: Entering directory '/home/cvmiller/build/PrawnOS'
TARGET is armhf
BUILDING /home/cvmiller/build/PrawnOS/build/armhf/prawnos-pbuilder-armhf-base.tgz
pbuilder create --basetgz /home/cvmiller/build/PrawnOS/build/armhf/prawnos-pbuilder-armhf-base.tgz --configfile /home/cvmiller/build/PrawnOS/filesystem/resources/pbuilder/prawnos-pbuilder-armhf.rc
W: /root/.pbuilderrc does not exist
W: cgroups are not available on the host, not using them.
I: Distribution is bullseye.
I: Current time: Thu Feb 18 10:19:17 PST 2021
I: pbuilder-time-stamp: 1613672357
I: Building the build environment
I: running qemu-debootstrap
E: qemu-debootstrap does not exist, install or change DEBOOTSTRAP option
W: Aborting with an error
make[1]: *** [/home/cvmiller/build/PrawnOS/filesystem/makefile:66: /home/cvmiller/build/PrawnOS/build/armhf/prawnos-pbuilder-armhf-base.tgz] Error 1
make[1]: Leaving directory '/home/cvmiller/build/PrawnOS'
make: *** [makefile:97: image] Error 2
```
Perhaps the DKMS directory structure could be included in the _next_ release of PrawnOS?

austin987, 2021-02-18:
> E: qemu-debootstrap does not exist, install or change DEBOOTSTRAP option

Have you installed the required dependencies?

cvmiller, 2021-02-18:
> Have you installed the required dependencies?

Apparently not. But let me go back to the original problem/issue. **One can not build kernel modules with the current release (1.3.1**). The build directory structure doesn't exist. The kernel is not a Bullseye Kernel, so the user can't use apt-get to get the kernel headers from the repo.

Please consider this a Feature Request, that would allow users to natively build kernel modules on PrawnOS (like other distros). Thx.

austin987, 2021-02-19:
Sure, I think the request is valid (though it's not my call to make).

My comment was mostly to unblock you/anyone else that comes across the report. I haven't tried it recently, but I have built PrawnOS under PrawnOS in the past, so it should work (IF you install the dependencies first, as stated in the README.md)