SolidEva/PrawnOS #165: Automate testing image with qemu

austin987, 2020-05-20:
Now that https://github.com/SolidHal/PrawnOS/pull/161 is merged, builds are automated with github actions. This is a followup to #150 

However, the image itself isn't actually tested (other than building). Ideally, we'd use qemu (maybe in conjunction with `expect` https://linux.die.net/man/1/expect) to attempt to boot/install the image.

I'm familiar with qemu, but never used it for arm stuff, so will take some research. Anyway, opening the bug to keep track of it.

austin987, 2020-05-20:
Feel free to assign this to me if you'd like, I don't have permission for that.

SolidHal, 2020-05-22:
Thanks for taking this on @austin987!

austin987, 2020-05-23:
Some notes for myself or anyone interested:
https://stackoverflow.com/questions/37079990/is-any-arm-platform-emulation-in-qemu-functional-enough-to-run-arm-v7-code-in-mu
https://gist.github.com/luk6xff/9f8d2520530a823944355e59343eadc1
https://blog.3mdeb.com/2014/2014-08-07-coreboot-for-qemu-armv7-vexpress-a9-emulated-mainboard/

austin987, 2020-06-05:
https://github.com/lukechilds/dockerpi has a dockerfile that runs a armv6 image inside of qemu.

It supports armv7 as well, but apparently networking doesn't work yet :( https://github.com/lukechilds/dockerpi/pull/4. Though in a quick test, networking doesn't work for me in armv6 mode either.

Anyway, still useful.