SolidEva/PrawnOS #166: Enable CONFIG_VIDEO_HANTRO for VPU Support

gregordinary, 2020-05-20:
The RK3288 is equipped with a full Hantro G1 Video Decoder.

**Request**: Enable [CONFIG_VIDEO_HANTRO](https://cateee.net/lkddb/web-lkddb/VIDEO_HANTRO.html) in the kernel config. 

From: https://lkml.org/lkml/2019/5/12/224
> Oh, I might have forgot to mention, we accidentally found that on
> RK3288, the decoder is in fact an Hantro G1 (the full version), IMX8M
> has a small version of Hantro G1 (vp8 and h264 only).

Potential format support [based on VPU capabilities](https://web.archive.org/web/20190322103810/http://www.verisilicon.com/IPPortfolio_14_58_2_HantroG1.html):
- H.264 Baseline, Main and High Profiles, levels 1 – 5.1
- H.264 MVC Stereo High Profile
- VP8 for WebM, WebRTC and WebP support
- MPEG-4 Simple and Advanced Simple Profiles, levels 0 – 5
- Sorenson Spark and H.263 Profile 0, levels 10 – 70
- WMV9 / VC-1 Simple, Main and Advanced Profile, levels 0 - 3
- MPEG-1&2 Main Profile, levels low, med and high
- RealVideo 8/9/10
- DivX ® 3/4/5/6 support – Home Theatre Profile Qualification
- VP6,VP7
- AVS Jizhun Profile
- AVS+ (AVS1-P16 Level 6.0 and 6.2) 4:2:0 color format 
- JPEG and MJPEG support 

Collabora has been doing work on this since 5.0 with the driver landing in 5.3 and further improvements since then. 

[5.3 Notes](https://www.collabora.com/news-and-blog/news-and-events/linux-kernel-5.3.html):

> Ezequiel Garcia and Boris Brezillon lead our work on the Hantro VPU driver. An important milestone has been reached on this project: addition of MPEG2 decoding. One might think that adding MPEG-2 decoding support is not in itself a particularly interesting feature (who uses MPEG2 these days?!), but, by doing that we introduced the common building blocks that will allow us to support other codecs like H.264 or VP8.
> 
> One important thing to note here, while kernel support for MPEG-2 decoding has been merged, we currently have no userspace library officially supporting the stateless codec APIs, which is anyway marked unstable and subject to incompatible changes.

[5.4 Notes](https://www.collabora.com/news-and-blog/news-and-events/linux-kernel-5.4.html):

> On the media subsytem front, Boris Brezillon and Ezequiel Garcia continued leading the work on the Hantro VPU driver, which supports video decoding on Rockchip RK3288, RK3399 and NXP i.MX8MQ SoCs. This release introduces support for H.264 decoding on RK3288, and also VP8 decoding on RK3288 and RK3399. Popular RK3288-based platforms include ASUS Chromebook Flip and **ASUS C201 Chromebook**, so this change brings Chromebooks one step closer to running upstream, reducing the up/downstream gap.

[5.5 Notes](https://www.collabora.com/news-and-blog/news-and-events/linux-kernel-5.5.html):

> On the VPU front, Ezequiel and Boris kept improving the hantro driver by cleaning up the code base, fixing a few corner cases and reviewing improvements/fixes posted by others.

[5.6 Notes](https://www.collabora.com/news-and-blog/news-and-events/linux-kernel-5.6.html):

> Hantro Video Processing Unit driver
> 
> The Hantro VPU allows video decoding and encoding acceleration from/to various formats, such as JPEG, H.264, VP8 and VP9.
> 
> This hardware unit is present in several Rockchip SoCs, such as RK3399, RK3288, RK3328 (which are already supported by the current kernel driver), PX30 and some others. It's also present in NXP SoCs, such as i.MX8MQ whose, support was added by Philipp Zabel from Pengutronix and is expected to be included in the kernel v5.7.
> 
> The upstream driver was introduced by Collabora in the kernel v5.0 supporting only JPEG encoding acceleration.
> 
> With our contributions, the driver currently supports JPEG, VP8, H.264 and several post-processing transformations.
> 
> Along with a number of cleanups and fixes, in the v5.6 development cycle we contributed support for color conversion via a post-processor hardware block (IPP) to expose the YUY2 format in addition to NV12. We plan to continue to maintaining and improving this driver, with a focus on encoding.

5.3 noted the lack of userspace libraries. I did find the following project, [gst1-hantro-g1](https://github.com/linux4sam/gst1-hantro-g1), which is a Hantro G1 Video Decoder GStreamer plugin that may be of use.

gregordinary, 2020-05-20:
Looks like there is a LIBVA branch to support H264 decoding through the new kernel interface: 
https://github.com/pH5/libva/tree/hantro-h264
https://github.com/bootlin/libva-v4l2-request/pull/26

SolidHal, 2020-05-24:
Thanks for the detailed request! I'll get those kernel options enabled.

Maccraft123, 2020-09-05:
I'll work on this for this week.

SolidHal, 2020-10-29:
I believe this is now fully supported in the most recent release, reopen if you find this is not the case

closed 2020-10-29 by SolidHal