SolidEva/PrawnOS #169: Installing to a USB device doesn't work

austin987, 2020-05-23:
While testing some more intensive changes, I noticed that installing from a USB device (/dev/sda, installed via dd from x86_64 linux) to another USB device (/dev/sdb, via PrawnOS-1.0.3) take a while (fair enough), but then fails to boot. Pressing 'CTRL+U', beeps, then returns to the white unlocked warning boot screen.

Tried two different USB devices to be sure. For both devices, installing FROM a USB stick, within PrawnOS, to another USB drive WILL NOT boot. However installing FROM a USB stick to a MicroSD card, WILL boot. I didn't test eMMC, because I only have one device and don't want to reformat ;).

It's not obvious to me from a quick look what's wrong here :/

rk-zero, 2020-05-23:
pls post output of

`cpgt show /dev/sda` and/or `cpgt show /dev/sdb`

beep and reboot is usually issue that depthcharge does not find the kernel, maybe diff both first partitions to see if the kernel was written correctly (should be 1:1 same)

austin987, 2020-05-24:
/dev/sda = good
/dev/sdb = bad

```
laptop ~ # sha1sum /dev/sda1 /dev/sdb1
f979f3714210c5d2858f997be613eaac83c56595  /dev/sda1
57b587e1bf2d09335bdac6db18902d43dfe76449  /dev/sdb1

laptop ~ # cgpt show /dev/sda1 
WARNING: Primary GPT header is invalid
WARNING: Secondary GPT header is invalid
ERROR: GptSanityCheck() returned 2: GPT_ERROR_INVALID_HEADERS

laptop ~ # cgpt show /dev/sdb1 
WARNING: Primary GPT header is invalid
WARNING: Secondary GPT header is invalid
ERROR: GptSanityCheck() returned 2: GPT_ERROR_INVALID_HEADERS

austin@laptop ~ $ cgpt show PrawnOS-1.0.3.img 
       start        size    part  contents
           0           1          PMBR
           1           1          Pri GPT header
           2          32          Pri GPT table
        8192       65536       1  Label: "Kernel"
                                  Type: ChromeOS kernel
                                  UUID: E3D4B398-B70E-1D43-B5C3-FF21A9E0853E
                                  Attr: priority=10 tries=5 successful=1 
       73728     4022239       2  Label: "Root"
                                  Type: Basic data
                                  UUID: 32B13C2A-F37C-DA4A-A362-2B8E3ABA01DE
     4095967          32          Sec GPT table
     4095999           1          Sec GPT header

austin@laptop ~ $ sudo cgpt show /dev/sda
Password: 
       start        size    part  contents
           0           1          PMBR
           1           1          Pri GPT header
           2          32          Pri GPT table
        8192       65536       1  Label: "Kernel"
                                  Type: ChromeOS kernel
                                  UUID: E3D4B398-B70E-1D43-B5C3-FF21A9E0853E
                                  Attr: priority=10 tries=5 successful=1 
       73728     4022239       2  Label: "Root"
                                  Type: Basic data
                                  UUID: 32B13C2A-F37C-DA4A-A362-2B8E3ABA01DE
    15728607          32          Sec GPT table
    15728639           1          Sec GPT header

austin@laptop ~ $ sudo cgpt show /dev/sda
       start        size    part  contents
           0           1          PMBR
           1           1          Pri GPT header
           2          32          Pri GPT table
        8192       65536       1  Label: "Kernel"
                                  Type: ChromeOS kernel
                                  UUID: E3D4B398-B70E-1D43-B5C3-FF21A9E0853E
                                  Attr: priority=10 tries=5 successful=1 
       73728     4022239       2  Label: "Root"
                                  Type: Basic data
                                  UUID: 32B13C2A-F37C-DA4A-A362-2B8E3ABA01DE
    15728607          32          Sec GPT table
    15728639           1          Sec GPT header
```

Unfortunately, the diff is _huge_:
```
laptop /tmp # diff -aur /dev/sda1 /dev/sdb2 > kernel.diff
laptop /tmp # du -sh kernel.diff 
3.8G	kernel.diff
laptop /tmp # xz -9 kernel.diff 
laptop /tmp # du -sh kernel.diff.xz 
1.1G	kernel.diff.xz
```

The kernel itself compresses pretty well:
```
laptop /tmp # sha1sum bad-kernel 
57b587e1bf2d09335bdac6db18902d43dfe76449  bad-kernel
laptop /tmp # bzip2 bad-kernel 
laptop /tmp # du -sh bad-kernel.bz2 
4.0K	bad-kernel.bz2
```

Thanks for the cgpt tip, I'll try debugging some more on the host itself.

austin987, 2020-05-24:
[bad-kernel.gz](https://github.com/SolidHal/PrawnOS/files/4672924/bad-kernel.gz)

rk-zero, 2020-05-24:
The bad-kernel is just empty, if that was the joke, i had a good laugh.

SolidHal, 2020-05-24:
It looks to me like the usb install process isn't copying the kernel to the new use device properly. 
I'm going to test out the script and see if I can recreate your results. 
Are you choosing encryption? 
have you tried another target usb device?

austin987, 2020-05-24:
Looks like the joke is on me; looks like a hardware issue. I noticed that if a usb drive is bumped, the computer may shutdown.

I recently replaced the keyboard, suspect I may have pinched a wire or something, need to tear down/reassemble.

That said, I put the computer on a table and tested, usb -> usb is fine (sha1sum matches).

Sorry for the noise.

closed 2020-05-24 by austin987