SolidEva/PrawnOS #170: dhclient doesn't automtatically run for usb ethernet

austin987, 2020-05-23:
Haven't investigated this, may later, it's late here.

Something I noticed that wasn't documented. IMO, should be documented and/or fixed, probably both. README.md mentions running dhclient for wifi, but not eth0.

Installing a desktop requires internet (I have a PR to mention this coming), but, at least for me, it doesn't get a DHCP address with me intervening. While it makes some sense to not join just _any_ WiFi connection without user intervention, in general I think it makes sense to use eth0, if present.

gregordinary, 2020-05-23:
I agree, @austin987 . Lately I've been using a USB-ethernet adapter for my testing and have been just doing a quick edit of /etc/network/interface to add:

```
    auto eth0
    allow-hotplug eth0
    iface eth0 inet dhcp
```
I'd advocate for USB-Ethernet working by default. Though, another approach could be to work Network Config into the install script and ask the user if they want to enable and configure Ethernet and/or WiFi. This could be right at the start of the script, or perhaps after selecting a component that requires internet access.

austin987, 2020-05-24:
IMO offering to configure makes sense for wifi, though that would take a non-trivial amount of work to script/polish it. I think that should be it's own feature request though.

SolidHal, 2020-05-24:
I agree that ethernet should "just work" when plugged in. I'll make those changes @gregordinary  mentioned a part of the install process

As you mentioned @austin987 wifi config is more complicated, but tools to configure it are already included. 
(and installation of a DE should NOT require an internet connection, see #172)

SolidHal, 2020-11-30:
in recent images, usb-ethernet is auto-configuring properly. Closing

closed 2020-11-30 by SolidHal