SolidEva/PrawnOS #173: Filesystem image build fails when installing xsecurelock

SolidHal, 2020-05-24:
```
+ chroot /root/PrawnOS-Master/tmp.mE4EUX24v5 apt-get install -y -t testing -d xsecurelock
Reading package lists...
Building dependency tree...
Reading state information...
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libc6-dev : Breaks: libgcc-8-dev (< 8.4.0-2~) but 8.3.0-6 is to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.
 ```

SolidHal, 2020-05-24:
@austin987 I see this even in the builds that are passing right now 
https://pipelines.actions.githubusercontent.com/KJs3rCmQkYdsgrZCXv53kj04Pw5UmLmhjEXoChGwTycmxGMbxP/_apis/pipelines/1/runs/5/signedlogcontent/3?urlExpires=2020-05-24T21%3A34%3A32.8309007Z&urlSigningMethod=HMACV1&urlSignature=sAN7eTYXCHb2wT9lkXlp0Y%2Bs7tnaWLcfSvsril53VTo%3D

closed 2020-05-24 by SolidHal

gdallasdye, 2020-05-26:
Hi. I believe libc6-dev may be striking again. I've seen libgtk3-perl, libpeas and gir1.2 before, and they were contributing factors in my removing libc6-dev from my test builds a couple of weeks ago. I got [this build fail alert a while ago,](https://github.com/gdallasdye/PrawnOS/runs/707619514) and the failure at the end sure looks familiar. 

Would removing libc6-dev again be worth it?

gdallasdye, 2020-05-28:
Some thoughts on mosys: when building mosys, we grab in packages packages from unstable. This later breaks downloading packages later. If we move mosys build towards the end after we download packages, we run out of room to build.

I've tried moving the building of mosys into it's own chroot, to avoid mangling the package manager and avoid running out of space. It's a little outside my skillset at the moment however. It may not be a good solution or a preferred solution, but it might still be a solution. Any help or discussion would be appreciated.

Stray thought: Can a chroot create and enter another chroot? Reliability issues?

SolidHal, 2020-06-09:
@gdallasdye You are right that grabbing the build reqs from unstable breaks things. I'm working on getting mosys built/packaged in a cleaner way :)

commit 9a38a994 (2020-05-24): Fix the build, without breaking libinput gestures.
closes #173