SolidEva/PrawnOS #177: build is broken

austin987, 2020-05-28:
From https://github.com/SolidHal/PrawnOS/actions, looks like this was introduced in 93192fa5117adb15e456c856d27bac911877a435.
```
2020-05-28T02:59:30.2757555Z dpkg-checkbuilddeps: error: Unmet build dependencies: libpam-dev libx11-dev libxcomposite-dev libxext-dev libxfixes-dev libxft-dev libxmuu-dev libxrandr-dev libxss-dev pandoc x11proto-core-dev
2020-05-28T02:59:30.2792780Z W: Unmet build-dependency in source
2020-05-28T02:59:30.7181806Z make[3]: Entering directory '/PrawnOS/packages/filesystem/xsecurelock/src'
2020-05-28T02:59:30.7182092Z dh clean --with autoreconf
2020-05-28T02:59:30.8374206Z    debian/rules override_dh_auto_clean
2020-05-28T02:59:30.8392279Z make[4]: Entering directory '/PrawnOS/packages/filesystem/xsecurelock/src'
2020-05-28T02:59:30.8392467Z dh_auto_clean
2020-05-28T02:59:30.9053495Z rm version.c
2020-05-28T02:59:30.9060790Z rm: cannot remove 'version.c': No such file or directory
2020-05-28T02:59:30.9064515Z make[4]: [debian/rules:44: override_dh_auto_clean] Error 1 (ignored)
2020-05-28T02:59:30.9064971Z make[4]: Leaving directory '/PrawnOS/packages/filesystem/xsecurelock/src'
```

austin987, 2020-06-02:
FYI @gdallasdye

austin987, 2020-06-03:
Not interested in playing the blame game either, just want to see it working again :).

I CC'ed you mostly because gnome is when things _started_ failing, see https://github.com/SolidHal/PrawnOS/actions. With 9522627 (adding chromium), the image built. The next push was adding 93192fa  (adding gnome), which is when things broke.

Anyway, looks like @SolidHal committed some fixes (though things stayed broken), it's now working as of https://github.com/SolidHal/PrawnOS/actions/runs/118390139 / 9f80d90

Which I should've noticed before cc'ing you, my bad..

closed 2020-06-03 by austin987

SolidHal, 2020-06-09:
Yeah, there were and are some issues with grabbing packages across debian suites. Hope to fix it all by packaging the items PrawnOS takes from bullseye and sid